softwarecrash/Daly2MQTT #1: high voltage after some time

feudas, 2022-04-11:
The Package Volt goes to 40V after some time of use. After a reset of the ESP, the value is ok again.

softwarecrash, 2022-04-11:
Can you Provide the Debug Log?

feudas, 2022-04-11:
Where can I enable debug?
MQTTDEBUG has no function and SERIALDEBUG (serial1) produces no output on D2

softwarecrash, 2022-04-11:
Ach deutsch ist viel besser ;)

An einem ESP ist serial1 auf d4 und gnd, kann man für mit einem zweiten USB zu ttl Adapter auslesen. Eine andere Möglichkeit gibt es bzw habe ich nicht gefunden ein debug Output zu erzeugen. Zum testen kann man auch einen großen arduino mega nehmen der hat mehrere serial, da kann man dann den serial Port für das BMS zb auf die 1 oder 2 legen und direkt auf serial0 über USB den Log abziehen

feudas, 2022-04-11:
TTL Adapter hab ich doch, ich dachte serial1 liegt auf D2 nun kommt da auch was :D
Ich nutze ein Daly 8S für Lifepo4. Wie soll ich dir das Log zukommen lassen und welchen Teil?
Einfach warten wann er mal wieder auf 40V geht?

softwarecrash, 2022-04-11:
Einfach reseten, dann kommen die Infos das Webserver startet ab da mitschneiden bis die Daten falsch werden.
Als Textdatei oder so einfach per ZIP hier hochladen.
Kann auch gut sein das der Fehler jetzt weg ist mit deinem anderen Fehler, das Problem aktuell in der Treiber Schnittstelle ist das wenn einmal Daten falsch interpretiert werden sich der Treiber verhaspelt und aus dem Tritt kommt, ich suche aktuell noch nach einer Möglichkeit das abzufangen

feudas, 2022-04-11:
Ja derzeit läuft es ordentlich, kann gut sein das er vorher mit dem Nuller ein Problem hatte.
Wenn es was zum Testen gibt immer her damit. 
Im laufe des Jahres werde ich auf eine 16S umstellen.
Und danke für deine Arbeit.

softwarecrash, 2022-04-11:
Ja er hat ein frame abgeschnitten was noch mit berechnet werden sollte bei 8s, das sollte jetzt behoben sein. Evt folgt noch eine Fehlerkorrektur bei dem verhaspeln.

Auf das Angebot komme ich gern zurück, ggf auf anderen Kanälen?

Wenn du etwas zu dazu beitragen kannst wie Anschlussplan oder Bilder wäre ich dankbar.

Ja gern, es ist nicht allein meine Arbeit, der Treiber ist ein fork mit Ersteller stehe ich in Kontakt das wir ihn weiter entwickeln und meine Änderungen einfließen lassen.

Teste es ein paar Tage wenn es jetzt fehlerfrei laufen sollte bitte schließe dann die issues

feudas, 2022-04-12:
Leider treten gleich mehrere Fehler auf.
Der alte High Volt ist noch da, jetzt schafft er sogar 204 Volt.
Zusätzlich habe ich alle MQTT Ausgaben in HomaAssistant eingebunden, mir ist ziemlich schnell aufgefallen das er einfach so bestimmte Werte nicht mehr aktualisiert.
So wie ich das sehe betrifft das Voltage, Current und SOC. Wie gesagt bei denen scheint es ihm irgendwann egal zu sein was das BMS sagt. Der Rest auch die Zellenspannung usw. sind weiterhin fehlerfrei.
Ich versuche das mal in einem LOG fest zu halten.

softwarecrash, 2022-04-12:
dann verhaspelt er sich noch bei den antworten, versuche doch mal bitte hier https://github.com/softwarecrash/DALY-BMS-to-MQTT/blob/30db5bd3082cbe729d0fe27d30d971efe3b1dfcf/src/main.ino#L348 auf 5*1000 zu ändern ob es dann immer noch auftritt.

softwarecrash, 2022-04-12:
hab gerade noch eine kleinigkeit geändert, wäre super wenn du das noch mit übernehmen kannst zum testen

feudas, 2022-04-12:
Änderungen sind eingebaut.
[Bilder.zip](https://github.com/softwarecrash/DALY-BMS-to-MQTT/files/8476164/Bilder.zip)
Mal bissl was Bildliches wie es hier ausschaut.

softwarecrash, 2022-04-12:
was stecken den da für zellen dahinter?
ja hatte überlegt die mah auch in ah umzurechnen, aber vielleicht braucht dann jemand anders wieder mah, das kann man sich in der visualisierung ja basteln wie man mag.

hier wird das erste zellenpaket gerade erst gebaut, wenns funktioniert werden es wohl mehrer, ist für nen garten solar projekt

feudas, 2022-04-12:
So die Änderungen haben es leider verschlimmbessert.
Jetzt geht gleich nach dem Neustart gefühlt 1 wert raus und dann ist ende. Zeitgleich braucht das Webinterface ewig bis es aufrufbar ist. Ich habe von einem Startvorgang bis das Webinterface erreichbar ist ein log angelegt
[log.zip](https://github.com/softwarecrash/DALY-BMS-to-MQTT/files/8476382/log.zip)

Ist ein 24 Volt LifePo4 aus 8 mal 200Ah Zellen. Werde ich dieses Jahr noch auf 48Volt umrüsten. weitere 8 benötigte Zellen sind schon bestellt. Dann wechselt das BMS auf 16 Zellen ;)

softwarecrash, 2022-04-12:
danke, der fehler scheint von den balance states zu kommen, die werden auch in mehreren frames gesendet, den fehler zu finden bzw zu korrigieren ist schwierig ohne ein bms was die werte sendet, bin auch nur hobbybastler kein profi programmierer ;)
kommentiere mal bitte diese zeile aus
https://github.com/softwarecrash/DALY-BMS-to-MQTT/blob/0ea24e8445eae181e25f8dbd6c7fe222d6c31d66/src/daly-bms-uart.cpp#L54
ob das problem immer noch da ist.

softwarecrash, 2022-04-12:
hab noch nen kleinen überlauf abgefangen. evt liegt es daran.
hab es entsprechend geupdated

feudas, 2022-04-12:
Es scheint nun zu laufen, oder es hält länger durch :D
Allerdings schwangt der Current Wert extrem. Ich entnehme recht Stabiel 9,5A bis 10A.
Bekomme aber schwankende Werte von 1,8A bis 15A angezeigt.  
Der Rest entspricht dem was auch andere Geräte melden, ist also ok.

softwarecrash, 2022-04-13:
Na wenigstens etwas.
Ich schau mir das nachher nochmal mit meinem BMS an, irgendwo werden fehlerhaft empfangene Daten trotzdem verarbeitet

feudas, 2022-04-13:
Ich nutze noch auf dem PC die Monitoring Software die nutzt zeitgleich den CAN/485 Bus. Die schließe ich aber nur zur Kontrolle an, weil ich das über USB Kabel habe. Nutze ich normal nur um Einstellungen Am BMS zu machen. Voltgrenzen anpassen und den aktuellen SOC stand korrigieren, falls der nach vielen rauf und runter zu ungenau wird. Dann habe ich noch einen Solarladeregler über den auch Entladen wird, der sagt mir sehr gut was aus dem Accu raus geht.
Aber auch nach den vielen Stunden läuft es jetzt stabil, bis auf die A anzeige alles ok :)

softwarecrash, 2022-04-13:
springt die ampere anzeige nur mal kurz weg oder ist das länger?
wie schaut den das log aus, tauchen da vermehrt fehler auf das er was falsches oder nichts bekommt?

softwarecrash, 2022-04-13:
sehr seltsam, da kann ich mir gerade keinen reim drauf machen, auser das die kommunikation völlig aus dem tritt ist.
ich hab eben einige bugs noch beseitig und ein wenig fehlerkorrektur eingebaut.
falls das ampereproblem immer noch besteht jetzt dann kurz bescheid geben dann muss ich die ganze kalkulation und rohdaten mal in den debug log packen

feudas, 2022-04-13:
Also was ich sehe ist sehr häufig das hier:

<DALY-BMS DEBUG> Send command: 0x98 Checksum = 0x85
<DALY-BMS DEBUG> Calculated checksum: 243, Received: 152
<DALY-BMS DEBUG> Error: Checksum failed!
<DALY-BMS DEBUG> RX Buffer: [0x80x00x00x00x00x00x00x00x00x690x1650x10x152]
<DALY-BMS DEBUG> Receive failed, Failure Flags won't be modified!
Data sent to MQTT Server

<DALY-BMS DEBUG> Send command: 0x90 Checksum = 0x7D
<DALY-BMS DEBUG> Calculated checksum: 132, Received: 1
<DALY-BMS DEBUG> Error: Checksum failed!
<DALY-BMS DEBUG> RX Buffer: [0x00x00x00x00x00x00x00x700x1650x10x1440x80x1]
<DALY-BMS DEBUG> Receive failed, V, I, & SOC values won't be modified!

<DALY-BMS DEBUG> Send command: 0x90 Checksum = 0x7D
<DALY-BMS DEBUG> Calculated checksum: 252, Received: 0
<DALY-BMS DEBUG> Error: Checksum failed!
<DALY-BMS DEBUG> RX Buffer: [0x00x1170x1170x10x800x1260x1650x10x1440x80x10x40x0]
<DALY-BMS DEBUG> Receive failed, V, I, & SOC values won't be modified!

<DALY-BMS DEBUG> Send command: 0x90 Checksum = 0x7D
<DALY-BMS DEBUG> Calculated checksum: 66, Received: 66
<DALY-BMS DEBUG> 26.00V, 0.90A, 33.60SOC

Am ende hat es wohl mal funktioniert, allerdings stimmt der Wert von 0,90A überhaupt nicht.
Also mehr Debug Ausgabe bitte :D

softwarecrash, 2022-04-13:
also das sieht stark nach verbindungsproblemem gerade aus, wie ist den deine verkabelung?

feudas, 2022-04-13:
Arg, da hätte ich auch mal drauf kommen können, wo Hardware doch schon eher meins ist.
Mein Mini Oszi zeigt mir gegen GND gemessen Negative Datenimpulse von 2,6V an.
Das durch einen widerstand noch weiter runter zu ziehen halte ich da für eher den falschen weg.
Bei meinem Daly zumindest. Ich schalte mal eine kleine Treiberplatine dazwischen habe ich eh rumliegen das zeug. da sind nur ein paar kleine Transistoren drauf. Nimmt man normal für den 3,3 auf 5V Pegelwandel. Der findet hier aber nicht stat. 
Das Daly hat auch nur 3,3v Pegel, Ich gehe davon aus das da auch nur ein kleiner admel oder so drin steckt.

feudas, 2022-04-13:
Wobei wir nicht außer acht lassen sollten, das die anderen Werte soweit in Ordnung sind ;)

softwarecrash, 2022-04-14:
Laut den Daly Datenblättern ist es UART, das Problem ist einfach das man wohl den internen pull-up von dem ESP auf rx nicht abschalten kann bzw die Hardware serial libray den aktiviert.
Im bms steckt eher ein stm oder dergleichen.
Ich werde heut Abend oder morgen Mal das ganze mit nem PNP Transistor prüfen.

Ist mir auch erst aufgefallen als keine Daten kamen und das saleae dann zeigte das rx nicht annähernd auf 0 gezogen wurde.

Das die anderen Werte stimmen kann auch daher kommen das er bei fehlerhafter Kommunikation die Werte nicht auf 0 setzt und so Recht lange plausible Werte im Speicher sind.

Das wäre noch eine Grundsatzfrage wie man damit umgehen kann wenn Fehler auftreten, letzte Werte behalten oder auf Null setzen?

feudas, 2022-04-14:
Nein, ich meine schon das der Rest ordentliche Werte hat. SOC zb. stimmt auf jeden fall, Volt ist ein Problem weil LifePo4 ja einen sehr flachen verlauf hat.
> Btw, das ufert hier langsam aus, evt auf WhatsApp/Telegramm/discord verlegen?
Ja wenn du was laufen hast, ich habe alles am Start.

softwarecrash, 2022-04-14:
Probieren wir es doch Mal mit discord das hab ich immer dabei
https://discord.gg/vRDSNSTm

softwarecrash, 2022-04-27:
Bug fixed with V0.2

closed 2022-04-28 by softwarecrash