softwarecrash/Daly2MQTT #4: Binary available ?

nodedbfo, 2022-04-24:
Hello Rocky, 
thank you for your work on the Daly BMS. 
Would it be possible to publish a binary for the Wimos D1 mini you use for your project ?
This would help me, and most likely others a lot who are not so firm with the compiler 
Thank you so much
Dirk

softwarecrash, 2022-04-24:
Hello,
yes sure, 
but now i have to find a bug in the load calculation, when its done and tested i will make a release with precompiled binarys

temporary here is a binary for testing:

nodedbfo, 2022-04-24:
Perfect thank you. 
I will test it and let you know the result. 
but it will not be before the next weekend. :-)

softwarecrash, 2022-04-26:
ok something is wrong with the bin and the start adress.
but you can use the tasmoizer, with this it works, i have tested
https://github.com/tasmota/tasmotizer

softwarecrash, 2022-04-27:
> thanks I will test ist as soon as it is available. The esp easy was a copy and paste mistake from my side in this chat only. . I did test it with your file not with the espeasy one.

its avaible with actual bugfix

nodedbfo, 2022-04-27:
In the meantime i was able to flash the 0.2.0 version 
it is working without the transistor on my device (Daly R32W-GC02, 200A/100A)
Also the online update is working. I did update to 0.3.1 (without debug)
the update was successful. 
The connection to my MQTT Server is working. 
COC is working
voltages are visible
first view looks realy good. 

by the way I did try to go directly via the mini USB connection which is on the BT-Device. This is not working The ESP is even not booting up on this one. I did hope that this would boot up and with minor modifications on the port this would be the perfect cable for the ESP. The BT Device is comming with an connector on the BMS Side and a Micro USB on the BT Side.

softwarecrash, 2022-04-27:
> In the meantime i was able to flash the 0.2.0 version it is working without the transistor on my device (Daly R32W-GC02, 200A/100A) Also the online update is working. I did update to 0.3.1 (without [...]
nice to hear that :)

some daly bms need a transistor, other need a levelshifter, or it works without.

> by the way I did try to go directly via the mini USB connection which is on the BT-Device. This is not working The ESP is even not booting up on this one. I did hope that this would boot up and with [...]

dont do that, realy DONT! the micro usb connector for the BT module is no standart pinout, it have 15V and packVoltage(for wakeup) on the pins and can destroy the esp.
i have buy a female micro usb cable and hoked it up to the esp so i can switch it easy.

softwarecrash, 2022-04-28:
Problem solved

closed 2022-04-28 by softwarecrash