softwarecrash/Daly2MQTT #7: Send all data in one Single JSON (nice to have)

PanicMan, 2022-07-23:
Hello, I have some timing problems regarding that every single value is sended as an single message to MQTT.
Is it possible, to keep things for existing users simple with an config switch, default false, to send all the values as an single JSON Object?

Would be very nice, thank you!

Something like this?
--------------------------------------------------------
{
    "Pack Voltage": 13.3,
    "Pack Current": 0.0,
    "Pack SOC": 98.10,
    "Pack Remaining mAh": 176580,
    "Pack Cycles": 5,
    "Pack Min Temperature": 26,
    "Pack High Cell": {
		"Cell": 2,
		"Voltage": 3.335
	},		
    "Pack Low Cell": {
		"Cell": 1,
		"Voltage": 3.329
	},		
    "Pack Cell Difference": 6,
    "Pack ChargeFET": true,
    "Pack DischargeFET": true,
    "Pack Status": "Stationary",
    "Pack Cells": 4,
    "Pack Heartbeat": 177,
    "Pack Balance Active": false,
    "Pack Cells Voltage": {
		"Cell 1": 3.329,
		"Cell 2": 3.335,
		"Cell 3": 3.335,
		"Cell 4": 3.331
	},		
    "Pack Cells Balance": {
		"Cell 1": false,
		"Cell 2": false,
		"Cell 3": false,
		"Cell 4": false
	},		
    "Pack Temperature Sensor No 1": 26,
    "Ampere Debug": ""
}

softwarecrash, 2022-07-23:
hello, that is possibil, but im not so perfect firm with json. let me check that the days, i will look that i build in a option for that (self compile witha extra option)

PanicMan, 2022-07-23:
Cool, thanks!
I think it will be also ok.
By the way, when you create an Object, and then serialise it, then it will create such JSON ;)

softwarecrash, 2022-07-23:
sure, the serialize is always build in for the web pages communication

softwarecrash, 2022-07-24:
i have added some things, you can enable the json string in the settings.h
baldy my bms actual not work so i cant test it

PanicMan, 2022-07-26:
Ok, looks good, thanks a lot!
But I'm not sure how to build it, can you put an switch into settings page and provide an new bin, so I can test it for you?

softwarecrash, 2022-07-26:
here the compiled bin.
actual i cant add a switch for this settings, so i rework the EEPROM managmant for more settings and other things

[firmware-json-test.zip](https://github.com/softwarecrash/DALY-BMS-to-MQTT/files/9188094/firmware-json-test.zip)

softwarecrash, 2022-07-27:
so the try is an error, i cant send more data over mqtt as a json, it works with 4-5 entrys, but more it doenst work. so i have roled back. sorry. when i have more time i will check it.

softwarecrash, 2022-08-01:
now i have a working version. a question @PanicMan its better to sort the json string like sonoff / tasmota or another struct?

softwarecrash, 2022-08-01:
New Release with Json String, only FET switches ar not working jet

PanicMan, 2022-08-02:
> now i have a working version. a question @PanicMan its better to sort the json string like sonoff / tasmota or another struct?

Good question, how it would look like? What are the benefits?

softwarecrash, 2022-08-02:
> > now i have a working version. a question @PanicMan its better to sort the json string like sonoff / tasmota or another struct?
> 
> Good question, how it would look like? What are the benefits?

the beneftis are for people that use smart homes, the tasmota or sonoff json struct will get in the sonoff adapter and will display right.
[image]
it looks it have nested objects for status and commands and other things.

PanicMan, 2022-08-02:
Does not matter for me, as I do not use such adapters...
If you want, you can do it, but as tasmota, not sonoff, maybe tasmota will some day take over the code ;-)

softwarecrash, 2022-08-02:
> Does not matter for me, as I do not use such adapters... If you want, you can do it, but as tasmota, not sonoff, maybe tasmota will some day take over the code ;-)

tasmota and sonoff uses the ecxact same string when i see right, i only use tasmota devices, so i only can rebuild the string of this devices.

ok i will try to modify it to fitt the tasmota/sonoff json string. actualy you can use the json option, but in future it will change again the structure

PanicMan, 2022-08-02:
Ok, great, thanks a lot! I will change it on my system when it is ready.
At the moment it works fine, but I have a lot of traffic at the MQTT Server
[image]

softwarecrash, 2022-08-02:
> Ok, great, thanks a lot! I will change it on my system when it is ready. At the moment it works fine, but I have a lot of traffic at the MQTT Server [image]

ah german ;)
warum bricht man sich dann nen zacken aus der krone mit englisch
der json string läuft bereits und ist in v0.3.3 integriert

PanicMan, 2022-08-03:
Haha, ja, einfach mal ein deutsches Wort rein werfen, schon weiß man Bescheid 😉 
Ich nutze OpenHab3, keine Ahnung ob es da einen Tasmota Adapter gibt, ich müsste dann per hand ca. 20 Items ändern, daher warte ich lieber auf die finale Version von dir und mache es dann. 😄

softwarecrash, 2022-08-03:
kannst du schon so nutzen, ich werde den string so lassen bis ich den rest angepasst habe was die fet schaltung angeht denke ich.
das auf den sonoff string umzufrickeln wird noch ne weile dauern denke ich.

PanicMan, 2022-08-03:
Ja, diese FET-Schaltung wäre aber schon wichtig, sonst könnte ich das nicht aus der Ferne steuern... 
Wobei, das BMS ist ja nur als letzte Instanz gedacht, den AkkuPack steuere ich über NodeRED, das BMS müsste eigentlich nie eingreifen, blöd nur wenn es das doch tut und ich dann nichts umschalten kann...

softwarecrash, 2022-08-03:
ja, die fet steuerung wird noch kommen, aber dafür muss ich das alles ein wenig umstricken, das dauert noch klein wenig. wie gesagt bin nicht sonderlich fitt in json, grad was das auseinandernehmen angeht. aber bin drann

softwarecrash, 2022-08-03:
so, die FET steuerung ist integriert

PanicMan, 2022-08-07:
Servus, habe jetzt heute die v0.3.4 drauf gemacht.
Funktioniert soweit super, auch das Umschalten der FET, nach einigem probieren, allerdings gibt es da ein Problem, ich kann bei OpenHab3 immer nur der beiden Schalter beim Umschalten senden, nicht beide gleichzeitig:
```
daly_bms/daly-bms : msg.payload : string[29]
"{"Pack":{"ChargeFET":"true"}}"
```
oder
```
daly_bms/daly-bms : msg.payload : string[32]
"{"Pack":{"DischargeFET":"true"}}"
```
Dabei wird jeweils der andere Schalter immer ausgeschaltet. Sprich ich kann nie beide einschalten.
Wenn ich mir deinen Code so anschaue, ist es klar, du setzt immer beide Schalter, ohne nachzuschauen ob der Wert dafür überhaupt gesendet wurde. 
Weiterhin kann ich zwar einen Schalter einschalten, aber nicht ausschalten, wenn ich das hier sende: 
```
daly_bms/daly-bms : msg.payload : string[33]
"{"Pack":{"DischargeFET":"false"}}"
```
passiert irgendwie gar nichts...

softwarecrash, 2022-08-07:
Das kann gut sein, ich habe in den Tests immer nur den gesendeten String als ganzes verändert, dabei stehen die Werte für beide Schalter natürlich schon drin.
Müsste ich Mal sehen wie das geht wenn man nur einen Teil bzw einzelne Abschnitte des json sendet.

softwarecrash, 2022-08-08:
Sollte jetzt mit V0.3.5 gelöst sein

softwarecrash, 2022-08-17:
solved

closed 2022-08-17 by softwarecrash