softwarecrash/Daly2MQTT #8: BMS data not available

robertdiers, 2022-08-06:
Hi,

After connecting to my local network I tried to read data from 2 different daly BMS. Sadly I cannot see any values in HTML GUI. GUI itself is working fine. Do I need to use the ADUM or what are u using to read the debug log? Sorry I do not have experience with serial dev. Just used tasmotizer to flash the debug image.

Thanks Robert 

[image]

softwarecrash, 2022-08-06:
hello,
for the debug log you need a usb-uart dongle.
can you show me your wireing? its look like the esp cant read data from the bms

robertdiers, 2022-08-06:
Hi, I used the orignal cable. Black to black, green to green, white to white, red to yellow (as i did not had red pin). I tried less -f for /dev/ttyUSB0 but there doesnt seems to be an output.
[image]

robertdiers, 2022-08-06:
[image]
[image]

robertdiers, 2022-08-06:
[image]

softwarecrash, 2022-08-06:
ah directly connected the bms to the esp. some bms dont like that, the bms have often not enugh power to pull down the rx pin from the esp, take a look at the readme, i have two versions for solve this problem

robertdiers, 2022-08-06:
Thanks - thats the info I need. Just have to buy the hardware first...

softwarecrash, 2022-08-06:
please give feedback when it works or you have other problems

robertdiers, 2022-08-06:
Yes, I will - thanks so far

PanicMan, 2022-08-07:
Look at the readme page, either the ADUM or the transistor, both worked for me 👍

robertdiers, 2022-08-07:
Yes, I already ordered the transistors. Will give feedback if that works.

robertdiers, 2022-08-11:
Hi, it works great with BC327-25 transistor. Thanks for your help! 
[image]

robertdiers, 2022-08-11:
[image]

softwarecrash, 2022-08-11:
> Hi, it works great with BC327-25 transistor. Thanks for your help! [image]

great!
so happy logging :)
please close the issue when no more problems

robertdiers, 2022-08-11:
Fixes

closed 2022-08-11 by robertdiers