softwarecrash/Daly2MQTT #9: Cell data missing

robertdiers, 2022-08-11:
Hi, 

I can read the Json information with MQTT client. Cell count is 16 ("Cells"). Sadly the detailed cell information is giving cells 1 to 13 only. Is there a possibility to configure this?

Thanks Robert

softwarecrash, 2022-08-11:
can you provide the BMS data and Json string?
can you chek it when you switch to classic mqtt, will it display all cells?

robertdiers, 2022-08-11:
JSON:
{"Pack":{"Device IP":"192.168.1.29","Voltage":53.5,"Current":0,"SOC":100,"Remaining mAh":90000,"Cycles":49,"MinTemperature":28,"MaxTemperature":28,"High CellNr":7,"High CellV":3.446,"Low CellNr":10,"Low CellV":3.319,"Cell Difference":127,"DischargeFET":true,"ChargeFET":false,"Status":"Stationary","Cells":16,"Heartbeat":29,"Balance Active":false},"CellV":{"CellV 1":3.443,"Balance 1":false,"CellV 2":3.331,"Balance 2":false,"CellV 3":3.34,"Balance 3":false,"CellV 4":3.324,"Balance 4":false,"CellV 5":3.34,"Balance 5":false,"CellV 6":3.322,"Balance 6":false,"CellV 7":3.446,"Balance 7":false,"CellV 8":3.346,"Balance 8":false,"CellV 9":3.324,"Balance 9":false,"CellV 10":3.319,"Balance 10":false,"CellV 11":3.338,"Balance 11":false,"CellV 12":3.326,"Balance 12":false,"CellV 13":3.34,"Balance 13":false}}

robertdiers, 2022-08-11:
v0.3.4 - when I turn of the JSON flag I just get two topics with Charge/Uncharge and true/false values - nothing else...

robertdiers, 2022-08-11:
ok, seems to be an issue with my mqtt dashboard, without JSON the cells are available.

robertdiers, 2022-08-11:
ok, have to reconfigre everything

robertdiers, 2022-08-11:
It is working - thanks! Will you create a release or shall I go with the zip?

softwarecrash, 2022-08-11:
the problem is simple the space, i must give the json string more space, but it is limited by the esp.
can you post the new JSOn string with all data that i can calculate it

a new release will comes up later with this change and more

robertdiers, 2022-08-11:
{"Pack":{"Device IP":"192.168.1.29","Voltage":53.7,"Current":0,"SOC":100,"Remaining mAh":90000,"Cycles":49,"MinTemperature":26,"MaxTemperature":26,"High CellNr":7,"High CellV":3.518,"Low CellNr":10,"Low CellV":3.32,"Cell Difference":198,"DischargeFET":true,"ChargeFET":false,"Status":"Stationary","Cells":16,"Heartbeat":4,"Balance Active":false},"CellV":{"CellV 1":3.508,"Balance 1":false,"CellV 2":3.334,"Balance 2":false,"CellV 3":3.343,"Balance 3":false,"CellV 4":3.326,"Balance 4":false,"CellV 5":3.346,"Balance 5":false,"CellV 6":3.323,"Balance 6":false,"CellV 7":3.518,"Balance 7":false,"CellV 8":3.354,"Balance 8":false,"CellV 9":3.326,"Balance 9":false,"CellV 10":3.32,"Balance 10":false,"CellV 11":3.345,"Balance 11":false,"CellV 12":3.328,"Balance 12":false,"CellV 13":3.344,"Balance 13":false,"CellV 14":3.354,"Balance 14":false,"CellV 15":3.35,"Balance 15":false,"CellV 16":3.338,"Balance 16":false},"CellTemp":{"Temp Sensor 1":26}}

robertdiers, 2022-08-11:
I already started to replace my business logic to work without JSON... I just rolled back. All good.

robertdiers, 2022-08-11:
Update via web gui worked.

softwarecrash, 2022-08-11:
thanks, with 16 cells are the json limit by 1024byte are reached, so it cuts the string.
i hope with 2048byte space it will work for more than 16 cells.

robertdiers, 2022-08-11:
you might simply shorten Balance to Bal or something like that...

softwarecrash, 2022-08-11:
> you might simply shorten Balance to Bal or something like that...

great idea, have you any idea what shorten to?

robertdiers, 2022-08-11:
or might simply ignore the balance field per cell - not sure if someone is interested in... I think nearly all people just want to have the voltage

softwarecrash, 2022-08-11:
> or might simply ignore the balance field per cell - not sure if someone is interested in... I think nearly all people just want to have the voltage

sure, the balance per cell is for debug or find a damaged cell.

you can close it when it worked for you

robertdiers, 2022-08-11:
I'm using the voltage over time to identify damaged ones ;-)

closed 2022-08-11 by robertdiers

softwarecrash, 2022-08-11:
> I'm using the voltage over time to identify damaged ones ;-)

sure, that will work too