softwarecrash/Daly2MQTT #10: The CellDiff display on the webpage is easier to read in volts as all voltages are in volts

thestealth131205, 2022-08-21:
Cell Diff wrong format in app an all other Datas are V only Cell Fiss ist mV

This is Confuse

`document.getElementById("cellDiff").innerHTML = (data.cellDiff/1000)+'V ';`

in webpages main.h i edit the data line to convert to Volt

And the mos fet toggle doesn't work because your data reading happens too fast and overwrites the command

in next days i edit the code and post it.

sorry for my bad english

Dennis

softwarecrash, 2022-08-21:
I have Made the diff in mV because its Most a verry smal value.

With the fet Switch, its the half right. Its to fast yes, but it Switch es, with the next read cycle it will get the changed Data Back.
Is German easyer for your?

thestealth131205, 2022-08-21:
> I have Made the diff in mV because its Most a verry smal value.
> 
> With the fet Switch, its the half right. Its to fast yes, but it Switch es, with the next read cycle it will get the changed Data Back. I have a Point on the To-Do list to implement a Blocker for Reading Data when Switch writes comes in, but Not realised Jet. Is German easyer for your?

okay but when i edit the update intervall to 5 seconds its all fine - test it.

2 seconds is for me to fast the wifi air time is with a few esps to full.

Anyway thank you very much for your work.
We seem to be building a very similar facility.
A Mpp 5kva hybrid inverter is already lying next to me to operate a large part of my house as an
with which I will also use your other project here

softwarecrash, 2022-08-21:
I Plan to rework this to websocket function with Realtime Update to one second, but its a Lot of Work.
The Big question is, need it? The Web Page was Not used regulary. Only for Setup and watch, the Most Data comes over mqtt Inc. Switching fets.

I Hope my Others Project can Help you with the inverter

thestealth131205, 2022-08-21:
i have found on webpages you have a static interval with 5 seconds this is the problem  

in main.h on line 99 -103

        }
        setInterval(fetch, 5000);
        fetch();
        });

softwarecrash, 2022-08-21:
Yes, every 5 Seconds the Javascript will fetch the Data from the ESP.
When i can rework to websocket the ESP Server will call the Website for new Data, thats better, but i need more time to do that or Someone help

softwarecrash, 2022-08-21:
Actual i have the epever Project for a friend on the desk, thats ready today. So i will Switch to the daly next time.

thestealth131205, 2022-08-21:
I will Show that i can Help.

softwarecrash, 2022-08-22:
> I will Show that i can Help.
> 
> The mqtt Data Point for mosfet has Same Problem the state ist not Safe

yes sure, all of my projects have this issue, i will create a save routine to fetch the switching and resend the correct state.
need a little bit more time to do that

thestealth131205, 2022-08-22:
Ok yes ok nice I'll see what I can do too. Yesterday I created a new subscribe dischargefetset for test purposes and then a callback with the set command of the library.

This works great so far 

I suspect that the get overrides the set command

softwarecrash, 2022-08-22:
Jep i See, let me rework the Web Page top websocket then i can make only one Trigger for web and mqtt

thestealth131205, 2022-08-22:
https://github.com/thestealth131205/DALY-BMS-to-MQTT

softwarecrash, 2022-08-26:
the flipping switches issue is fixed with V0.3.6

softwarecrash, 2022-08-27:
@thestealth131205 can it close?

thestealth131205, 2022-08-27:
Yes  can it

closed 2022-08-27 by softwarecrash