softwarecrash/Daly2MQTT #12: charge / discharge state does not turn on

robertdiers, 2022-08-31:
Hi,

any idea why charge / discharge state does not turn of when charging voltage is on? Flags get disabled when cell voltage is to low, but it looks like they don't get activated. It works with connected BT module. (Version 0.3.6)

Webpage is open all the time in smartphone, might it be a problem?

thanks and regards Robert

softwarecrash, 2022-08-31:
i think its a issue with 0.3.6
can you self compile it?

robertdiers, 2022-08-31:
If you give command to do so :-) I don't have that much C++ experience...

softwarecrash, 2022-08-31:
[firmware.zip](https://github.com/softwarecrash/DALY-BMS-to-MQTT/files/9464326/firmware.zip)
please try this

robertdiers, 2022-08-31:
Alternative would be to send on command every few minutes. (In case it does not work with latest code) 
I will try - Thanks!

robertdiers, 2022-08-31:
installed on two devices - tomorrow I can give feedback. Thanks for the fast response!

softwarecrash, 2022-08-31:
can you explain it with more details? my english is not the best. 
the bms have a featuere, that will enable discharge by enable load or reverse, i dont understand that exactly but it comes from the bms

robertdiers, 2022-08-31:
Wir können auch in Deutsch sprechen :-) Wenn eine Zellspannung außerhalb der Grenzwerte fällt (oder die Temp zu hoch geht), stellt das BMS den entsprechenden Vorgang ein - schaltet also entweder Laden oder Entladen ab. Normalerweise wird dies aber wieder aktiv, wenn der gegenteilige Vorgang gestartet wird. Alternativ kann man auch über die GUI (egal ob die HTML von dir oder das BT das Flag) noch einmal selbst setzen. Nachts zieht es meine Akkus meist leer, das Entladen schaltet also ab. 2 der 3 BMS schalten den Ladevorgang aber nicht ein, wenn am Folgetag wieder Spannung durch Solar oder Ladegerät anliegt. Beim dem dritten BMW klappt es, wenn die BT Adapter verbunden sind klappt es auch. Eventuell ist es auch ein Kabelkontakt, aber das Verhalten ist irgendwie strange...

softwarecrash, 2022-08-31:
überprüfe mal penibel ob alle einstellungen identisch sind, batterien auch?
die daly dinger haben die eigenheit wenn zu tief entladen und feherlstufe 2 erreicht dann blockieren die ganz bis man sie resetet.
hängt der esp an einer eigenen stromversorgung?

robertdiers, 2022-08-31:
Eine pragmatische Alternative wäre es den Status alle paar Minuten von außen auf aktiv zu setzen. Entweder per HTTP Aufruf oder per MQTT - den Code dazu habe ich gesehen, nur noch nicht ausprobiert. 

Der ESP wird über das BMS mitversorgt, geht auch mit dem BMS an und aus. Fehlerstufe 2 sollte ich nicht erreichen, habe die Spannung der LiFePo Zellen auf 3,00 bis 3,45 V begrenzt. Vor ein paar Wochen hatte ich noch 2,8 und 3,6 V, das lief mit den BT Geräten - aber der Spannungsausgleich unter den Zellen war miserabel.

softwarecrash, 2022-08-31:
ja, es scheint massive stromprobleme zu geben, nur die beiden datenleitungen und masse verbinden, den esp selber über eine eigene stromversorgung. evt klappt das besser.

robertdiers, 2022-08-31:
ok, da brauch ich erst einmal Kabel die lang genug sind. Probiere ich aus - soll ich das Topic hier offen lassen oder schließen wir?

softwarecrash, 2022-08-31:
lass ruihg offen

robertdiers, 2022-09-01:
BMS haben heute das Laden gestartet mit neuer Version

softwarecrash, 2022-09-01:
super, dann ist da wohl nen bug im release. werd zeitnah ein neues machen.

closed 2022-09-03 by softwarecrash