softwarecrash/Daly2MQTT #13: Cannot compile with platform io

sckndgn, 2022-09-14:
Hello,
I saw a platformio.ini file, and I assume I need to compile it with platformIO. Unfortunately, I mostly used Arduino ide for programming, and I could not find proper documentation for platformIO and couldn't manage to compile it. I googled a lot before bothering here with this without any luck. I would really appreciate it if you would give some hints about building it or compiling a bin file for ESP32 would be great. 
I'm having the below errors when I try to build it with platformIO. I'm looking for your support to resolve it. Thank you.;
[image]

softwarecrash, 2022-09-14:
esp32 ar not supported

sckndgn, 2022-09-14:
Thank you for your quick response. I will add the folder and go with an ESP8266. Do you know why ESP32 is not supported?

softwarecrash, 2022-09-14:
Esp32 is Not supported why i dont add the envroiment for that. It dont need a Big espew for that little Job.

In flatform io Import my Project and then Hit compile, PIO does the rest

sckndgn, 2022-09-14:
It worked, thanks a lot.

closed 2022-09-14 by sckndgn