softwarecrash/Daly2MQTT #18: Failed to connect to mqtt server

he77PC011, 2022-12-04:
[image]
[image]
[image]
https://github.com/emqx/MQTTX

softwarecrash, 2022-12-04:
only can see you have set different ports on esp and mqtt broker

he77PC011, 2022-12-04:
my mqtt servre m.fdj.ltd  61883
web http://m.fdj.ltd:58083/  demo demo123

softwarecrash, 2022-12-04:
set your mqtt port and user/pass korrekt

he77PC011, 2022-12-04:
local area network[interconnected network](javascript: void(0))
[image]
[image]
Failed to connect to mqtt server

softwarecrash, 2022-12-04:
never tested or used it with external domain/ip settings. go to the mqtt broker to check connecting fails.

he77PC011, 2022-12-04:
[image]
[image]
Testing with client is normal.

softwarecrash, 2022-12-04:
try it with local ip....

he77PC011, 2022-12-04:
DALY-BMS-to-MQTT.V0.4.9.bin
I've tried one thing. I can't.

he77PC011, 2022-12-04:
Main.ino failed to compile. Maybe the library is wrong.

he77PC011, 2022-12-04:
If you have time to test it
my mqtt servre m.fdj.ltd 61883
web http://m.fdj.ltd:58083/ demo demo123

softwarecrash, 2022-12-04:
> Main.ino failed to compile. Maybe the library is wrong.

no, freshly tested, no problems with the librays, so i think you use odd arduino IDE, use vscode or make sure you have the matching libray versions from the platform.io file installed.

dont understand what you are doing!
all users dont have any issue with the mqtt connection, so its a problem on your broker or port/network settings, or wrong configuration!

so i test with your data
[image]
[image]

only a bug is shown with the mqtt port, with more than 4 digits, it will fixed with the next release

dont see any problems

closed 2022-12-04 by softwarecrash