softwarecrash/Daly2MQTT #20: Example on switching the FETs via MQTT

derFrickler, 2022-12-19:
Hi There, thanks a lot for this great project. Reading data from my Daly works just fine. 

I am just struggling on how to switch the FETs via mqtt. could you provide an example of the topic and payload i need to sent?

softwarecrash, 2022-12-19:
hey,
have a look at the wiki
https://github.com/softwarecrash/DALY-BMS-to-MQTT/wiki/Editable-values-over-MQTT-and-Json

so the mqtt datapoints are listed is editable, and the same with the json string, just grab the string and edit the fet values and send it back

derFrickler, 2022-12-20:
Thanks for your answer! 
As i am running "MQTT Json Style" the single topics for setting the values do not work, as they are not subscribed (Is there a reason for not doing that? setting single values could still be useful even if reporting is set to JSON.

But although setting the values via JSON has no effect. I tried sending the full received JSON message back, just with ChargeFET set to false, but its not set. 

In general its strange that the topic for setting values is the same that is used for reporting by the device. May i ask why you did that instead using a separate set topic as most other mqtt projects do.

btw. i am running V0.4.9.

softwarecrash, 2022-12-20:
When you have the time, come to the discord Channel, thet we can Take a closer Look at the Problem, the Channel ist German  too ;)

softwarecrash, 2022-12-20:
fixed

closed 2022-12-20 by softwarecrash