softwarecrash/Daly2MQTT #22: Possibility to connect more than 1 BMS?

thomas41587, 2022-12-21:
Is there any possibility to connect multiple BMS? Maybe using softserial or similar (as there is only one hardware serial port)?
To be more specific: I do have two BMS and one active balancer. I would love to connect all three to one Wemos/ESP.

softwarecrash, 2022-12-21:
the Code only Support one BMS

thomas41587, 2022-12-22:
I was already aware that there seems to be no config option for more than one BMS. So this is rather a feature request then a bug.

softwarecrash, 2022-12-22:
The Problem is, that the known Software serial  librays dont Work with the BMS.

When im right remember the BMS give a Option to give him a number (PC Software) so more than one IS possibile by this Adress, i will Check this later but i have No Idea what going on when Connect the UART Ports from the BMS to a Common.

But you can try it with a uart-usb Adapter. In the PC Software search for the Option, numerate your BMS and Connect all to the uart-usb then the PC Software listen more decive (i Hope) when that working we can modify the Code.
I cant Test it so i have only one BMS for testing

all-solutions, 2022-12-26:
I think the address assignment will only work via rs485, as all BMSs are then connected to the rs485 bus. I still have a second BMS here, I will try it out in the next few days.

softwarecrash, 2022-12-26:
> I think the address assignment will only work via rs485, as all BMSs are then connected to the rs485 bus. I still have a second BMS here, I will try it out in the next few days.

I think it will Work with rs232 too, so in my Code i must Adress a BMS, and in the PC Software (evt in App too) give a number, so with a little luck it will Work.

Btw, have you the time to make a PCB for a esp8266-01 with Transistor? I will add a second Transistor for wake Up the BMS, Test it next weeks.

all-solutions, 2022-12-26:
> Btw, have you the time to make a PCB for a esp8266-01 with Transistor? I will add a second Transistor for wake Up the BMS, Test it next weeks.

I can do that. I'm on holiday until New Year's Eve ;-).
I think you want to connect the WakeUp transistor to GPIO2, right?
Did you find out how Daly does the wake-up? WakeUp pin against GND or +5V does not work.

softwarecrash, 2022-12-26:
> I can do that. I'm on holiday until New Year's Eve ;-).
> I think you want to connect the WakeUp transistor to GPIO2, right?
> Did you find out how Daly does the wake-up? WakeUp pin against GND or +5V [...]

I must Check First what Transistor will Work with wich gpio.

Wake Pin to gnd will wake the BMS, when connected the BMS doenst Go sleep

all-solutions, 2022-12-26:
> Wake Pin to gnd will wake the BMS, when connected the BMS doenst Go sleep

OK... I will test it again. I had tried to wake up a sleeping BMS in this way, but it didn't work (otherwise I had planned to install a push-button on the current PCB).

softwarecrash, 2022-12-26:
> > Wake Pin to gnd will wake the BMS, when connected the BMS doenst Go sleep
> 
> OK... I will test it again. I had tried to wake up a sleeping BMS in this way, but it didn't work (otherwise I had planned to install a push-button on the current PCB).

When you have the Bluetooth Thing with the Button you can measure wich Pin will goes to ground when pressing it.

all-solutions, 2022-12-26:
[image]

[image]

First (ALPHA) design ;-)

softwarecrash, 2022-12-27:
> [image]
> 
> [image]
> 
> First (ALPHA) design ;-)

ah i have a little thinking fault, when the bms sleeping it gives no power out on the uart port so it cant wake up the bms.
when the wake pin hold down the bms never goes sleep
btw the bms gives 3,3v out not 5v so no voltage converter is needed.
the pin 4 is the wake up pin when it goes to ground the bms will wake up
6(RX)-5(TX)-4(BD_WAKE)-3(15V)-2(3V3)-1(GND)

all-solutions, 2022-12-27:
> ah i have a little thinking fault, when the bms sleeping it gives no power out on the uart port so it cant wake up the bms.
> when the wake pin hold down the bms never goes sleep

Hm, then maybe "hardcoded" as a jumper for the guys who want to feed the board via BMS and via GPIO with external power supply?
> btw the bms gives 3,3v out not 5v so no voltage converter is needed.

F**k, I also have a thinking fault on the PCB for the Wemos... shit. But the Wemos is absolutly fault tolerant. BMS_3V3 ist connected to Wemos_5V, the PCB's work as BC and ADUM version nevertheless.
I saw it in the documentation of the BMS (3V3) and then (because it is a USB connection) defined it as 5V in Eagle and then continued to work with it. 
The converter would then only be necessary for the external power supply (I am currently building it into the circuit), so you would have a voltage range of 5-9V.

softwarecrash, 2022-12-27:
> Hm, then maybe "hardcoded" as a jumper for the guys who want to feed the board via BMS and via GPIO with external power supply?
>
>
> F**k, I also have a thinking fault on the PCB for the Wemos... shit. [...]

a hard 2 jumper for holding wake the bms is a option.

the wemos power in is tollerant, when feed it with 3v3 from the bms after the lm117 it comes with 3v its low but the esp will work

softwarecrash, 2023-01-15:
after reading a lot about the communication and multi bms, i think this is not a option for the actual project.

closed 2023-01-15 by softwarecrash