softwarecrash/Daly2MQTT #26: Permanentes Connection-Loss Wemos -> BMS

all-solutions, 2023-01-03:
Hi Rocky,

ich schreib mal auf Deutsch. Ist wahrscheinlich einfacher und verständlicher ;-).

Ein Bekannter (und auch dessen Kumpel) haben ein absolut komisches Phänomen mit Deinem Programm. Die veröffentlichten Versionen von Dir verlieren permanent die Verbindung zum BMS. Ich habe ihm dann Deine Version mit aktivem Debug kompiliert (damals als erste mit der 0.4.9). Lief ohne jegliche Abbrüche. Er hat seitdem kein Update mehr gemacht gehabt bis zur 0.4.13. Bei der Version das gleiche "Problem".
Du hast in der Version das Debugging in der main.ino selbst drin gelassen. Hilft nicht. Ich habe dann in der daly-bms-uart.cpp das Debugging auch noch aktiviert und den Zweien daraus eine Test-Version kompiliert. Damit läuft es, keinerlei Abbrüche.
Ich kappier das nicht, habe Deinen Code durchforstet, es gibt keinen Grund dafür, dass es ohne Debugging nicht funktionieren sollte. Habe auch Deinen Original-Code selbst neu kompiliert (vielleicht liegt es ja am Kompiler???). Nein, sobald ich in der daly-bms-uart.cpp die Debug-Zeile rauskommentiere, läuft das bei den Zweien nicht mehr.

Beide benutzen mein PCB-Layout. Den benutze ich selbst auch. Ich hab da keinerlei Probleme mit, obwohl ich einen "unterirdischen" RSSI zum Wemos habe. Es macht da auch keinen Unterschied, ob sie den Wemos via BMS versorgen oder per ext. Netzteil.

Hast Du da vielleicht eine Idee? 

Gruß
Manu

softwarecrash, 2023-01-03:
Grüße,
das ist sehr interessanter Anhaltspunkt mit dem debug.
Fakt ist das dass schreiben der debug Daten auf den zweiten serial jeweils ein paar cpu Zyklen kosten.
da fallen mir ad-hoc ein paar Szenarien ein warum das Problem auftritt
- Transistor ist zu langsam bzw kommt nicht schnell genug aus der Sättigung raus.
- das daly ist zu träge mit antworten oder die libray ist zu hektisch.
man könnte jetzt versuchen das ganze mit delaymicroseconds ein wenig auszubremsen, ich kann es bei mir leider in keinem Szenario nachvollziehen.
man könnte auch mal versuchen mit den Standard levelshiftern zu arbeiten, mittlerweile hab ich welche da (falls das Problem daher kommt)
oder man wählt den faulen weg und lässt das debug aktiv, das wäre an sich kein Problem, man muss es nur wissen. am ende frisst das debug kein Brot.
ich wäre aktuell für letztere Option, da ich es wie gesagt leider nicht reproduzieren kann.
du kannst gern ein wenig testen wenn du das debug deaktivierst und ein paar delayMicroseconds() verteilen in der daly-uart.cpp kann man eigentlich nur testen wo, aber ein Ansatz wäre da wo sonst die debug ausgaben abgearbeitet werden.

all-solutions, 2023-01-03:
Ich kann es selber auch nicht reproduzieren. Die Jungs wohnen Nähe Hannover und ich im Ruhrpott. "mal Eben" hinfahren ist da eher schlecht.
Ich sprech mal mit den Beiden, ob sie mal "Beta-Tester" spielen wollen. Dann kann ich mal ein paar Delays reinbauen. 

Achso, die arbeiten mit der ADUM-Version. OK, da ist jetzt ein Unterschied... ich nutze die Transistor-Version. Ich hab noch Platinen da, ich löte mal eine mit Transistor fertig und schick dem Bekannten die mal per Post...
Parallel bau ich mir mal eine mit ADUM und teste hier.

softwarecrash, 2023-01-03:
Das wäre Super.
ich kann auch mal probieren mit den einfachen Levelshiftern das ganze zum laufen zu bringen. ggf funktionieren die besser als transistor und adum, die gibt es für 1-3euro bei amazon und sind ziemlicher standart krempel

all-solutions, 2023-01-03:
Der eine Kollege macht schon mal als "Test-Opfer" mit. Mal sehen, ob ich es Morgen nach Feierabend schaffe zumindest seine Platine fertig zu machen und zu versenden.

Ich überlege grad auch noch in eine ganz andere Richtung... TLP's, also Optokoppler. Die ziehen Dir alles runter durch die galvanische Trennung. Müsste mal testen, ob die schnell genug sind. Hätte den Vorteil (für Leute, die den Wemos ext. versorgen), dass eine komplett galvanische Trennung vorhanden ist. Klar, dann wäre das Design meiner PCB auch hinfällig, die müsste ich komplett überarbeiten. Hab die TLP's noch massenweise hier (gab's nur im 50'er Pack als ich 2 davon brauchte).
Ich schlaf mal eine Nacht drüber...

softwarecrash, 2023-01-03:
Optokoppler wären auch eine Option, davon hätte ich auch die Standart 4pin bastel Variante da, müsste ich Mal schauen welchen Typ.
Die Transistor Variante hatte sich nur durchgesetzt da sie wirklich simpel nachzubauen ist und viele das auf Lochraster stecken (ich auch) oder gar fliegend verdrahten.

Ein traum wäre ja eine kaufbare fertig bestückte Platine auf SMD Basis, aber dafür gibt es noch zu viele Baustellen und Ideen glaube ich.

Du machst das vermutlich über jlpcp? Hast du da grob ne Richtung was eine bestückte Platine kostet wenn man angenommen Optokoppler nimmt nen Transistor für wake, pinheader für dauerwach? Evt müsste man Mal den 15v Ausgang checken, der kann auch nur max 50ma aber mit step down hat man bisl mehr Power und dazu nen esp12f oder sowas

all-solutions, 2023-01-03:
Ja, ich mach's über jlcpcb, aber bisher nur als reine Platine (auch bei anderen "Projekten"). Habe bewusst SMD rausgelassen, damit jeder die selbst bestücken kann.
Wäre aber mal interessant zu wissen, was die mit Bestückung kosten würden. 
Ich bau mal eine SMD-Version und lad die da mal hoch. Mal sehen, was die veranschlagen. Würd mich auch mal interessieren ;-). Die "nackten" Platinen liegen unter einem Euro/Stück, Wenn es preislich passt mit Komponenten, dann wäre das eine Überlegung. Wemos & Co kosten ja auch noch etwas, wenn man die dann selber bestückt. 
Dann würde ich aber eine komplett bestückte Platine samt ESP "bauen" (inkl. "Programmierport" und D0-Taster, dann aber auf RS232-Basis sonst können wir direkt einen Wemos drauf packen).

Wenn dann alle Baustellen abgearbeitet sind, wäre das echt eine Überlegung, das als Fertigplatine rauszujagen.

softwarecrash, 2023-01-03:
Ja ein esp01 reicht da auch... Wenn man das über das BMS versorgt und die Led vom ESP entfernt könnte das klappen, dann gibt es da halt keine Option zum aufwecken sondern nur nen Jumper für dauerhaft wach halten.

Mit dem esp12f wäre es nur schön flach geworden wenn man es auf SMD Basis bastelt, ok programmier Header würde wieder auftragen.

Am Ende bin ich da ein wenig zu blöd in Elektronik um das sauber zu realisieren, da müsste ja auch noch ne z Diode hin damit niemand Mal eben 5v in den USB rammt und alles grillt.

Erstmal das debug Phänomen in den Griff bekommen, ansonsten ist gpio2 auch nicht für ggf wake Funktion frei

all-solutions, 2023-01-03:
"gpio2 auch nicht für ggf wake Funktion frei"

GPIO0 mit Pullup würde aber auch noch gehen. 

Hast aber Recht, auch wenn ich es selbst als Phänomen sehe, bei zwei Leuten aktuell bei mir. Das Schlimme: ich kann es nicht selber reproduzieren, mal sehen, ob es nicht -warum auch immer- am ADUM liegt. Aber warum läuft das Vieh dann mit aktiviertem Debug in der daly-bms-uart.cpp? Das Debug in der main.ino ist ja schon aktiviert in Deinem Quellcode.

all-solutions, 2023-01-03:
GPIO0 darf nicht low sein beim Booten. Danach kannst Du mit dem alles "anstellen" was Du willst.
Low muss der zum Flashen sein im Boot-Vorgang.

Aber das wäre echt der Hammer, wenn es am ADUM liegt. Kollege wollte extra die "Luxus-Variante" mit dem ADUM haben. Das würde erklären, warum er die Probleme hat und ich mit der "billigen" Transistor-Variante nicht.

softwarecrash, 2023-01-04:
> GPIO0 darf nicht low sein beim Booten. Danach kannst Du mit dem alles "anstellen" was Du willst. Low muss der zum Flashen sein im Boot-Vorgang.
> 
> Aber das wäre echt der Hammer, wenn es am ADUM liegt. Kollege wollte extra die "Luxus-Variante" mit dem ADUM haben. Das würde erklären, warum er die Probleme hat und ich mit der "billigen" Transistor-Variante nicht.
> 
> BTW: in meiner ersten Testschaltung (ohne irgendwas, ADUM, BC) lief das auch problemlos. Die damalige Lochraster hab ich noch hier liegen.
Das ist ja das kurriose oder besser das seltsame dem ich noch nicht auf die spur gekommen bin, bei manchen funktioniert die kommunikation ohne alles, bei mir zb bei zwei ziemlich verschiedenen BMS, keine chance. senden geht problemlos, das kommt sauber bei dem BMS an, aber vom BMS zum ESP war immer das problem das der RX vom ESP auf high ist (soweit ok) aber das BMS mit seinem TX den High pegel vom ESP nicht runter ziehen konnte, bzw nicht weit genug so das die antworten im nirvana verschwanden. man könnte auch sagen der ESP ist zu kräftig oder eben andersrum das BMS zu "schwach"

> Ich bau Morgen dem Bekannten mal die Transistor-Version und schick ihm die Platine. Parallel werde ich ihm mal eine Software-Version machen mit deaktiviertem Debug in der main.ino machen. Ich bin gespannt...

ich hab unter diskusionen mal was gestartet für die platine wo man mal alles zusammentragen kann https://github.com/softwarecrash/DALY-BMS-to-MQTT/discussions/27

all-solutions, 2023-01-04:
So, der Bekannte hat sich heute selbst eine "Transistor-Version" gebaut (ADUM aus einem der PCB's rausgelötet und BC eingesetzt). Bei ihm funktioniert weder die eine noch die andere mit mit der Original-Firmware.
Habe ihm eben dann eine Version kompiliert, in der in der main.ino das Debug raus ist und in der daly-bms-uart.cpp aktiviert ist. Damit laufen bei ihm beide Platinen-Versionen.

softwarecrash, 2023-01-04:
> So, der Bekannte hat sich heute selbst eine "Transistor-Version" gebaut (ADUM aus einem der PCB's rausgelötet und BC eingesetzt). Bei ihm funktioniert weder die eine noch die andere mit mit der [...]

ok, dann würde ich als bugfix erstmal das debug in der libray aktiv lassen.
ohne logic analyzer ist das jetzt ganz schwer rauszubekommen woran das liegt

softwarecrash, 2023-01-04:
hab grad nochmal durchgeschaut, der einzige output über das debug der immer kommt ist in zeile 561-566 alle anderen debug out in der daly libray sind nur bei fehlern.
wäre mal interessant wenn man das debug deaktiviert und dort ein delay von 1-5ms einsetzt ob das de gleichen effekt hat, eventuel ist in der funktion das darunter folgende fix für den sleep bug auch mitschuld

all-solutions, 2023-01-04:
> ohne logic analyzer ist das jetzt ganz schwer rauszubekommen woran das liegt

Jepp, und erst recht so als "Fernfehlersuche".

softwarecrash, 2023-01-04:
[daly-bms-uart.cpp.txt](https://github.com/softwarecrash/DALY-BMS-to-MQTT/files/10344865/daly-bms-uart.cpp.txt)
probier es doch bitte mal damit, ggf kannst du den wert in zeile 568 auch erhöhen.

all-solutions, 2023-01-04:
Mit 500ms keine Änderung.

Mit 1000ms Zitat vom Bekannten:

> Mit ADUM auf den ersten Blick wieder negativ, aber irgendwie "anders". Erst weg, kommt dann nach kurzer Zeit aber von selber wieder und geht nach einiger Zeit aber wieder in Sleep bzw. Disconnect.
> Mit Transistor zeigt sich aber gleiches Verhalten/Fehlerbild wie bei ADUM. Es dauert insgesamt etwas länger bis der Sleep eintritt, aber der kommt auch da irgendwann. Hier kommt dann aber auch nichts von selber wieder. Es hilft ein manueller Refresh der Seite, aber der nächste Sleep ist dann binnen Sekundenbruchteilen wieder da.
> Korrektur auch hier kommt es von alleine wieder, aber nicht dauerhaft

Also den Delay noch höher setzen... ich weiß nicht.

softwarecrash, 2023-01-04:
> Mit 500ms keine Änderung.
> 
> Mit 1000ms Zitat vom Bekannten:
> 
das sind us, sprich 1000 ist 1ms. da kannst ruihg experimentieren, ich denke bis 5000 sollten da keine probleme auftreten, das wären dann gerade mal 5ms.

all-solutions, 2023-01-04:
> das sind us, sprich 1000 ist 1ms. da kannst ruihg experimentieren, ich denke bis 5000 sollten da keine probleme auftreten, das wären dann gerade mal 5ms.

Ne, das sind ms ;-)
https://arduinogetstarted.com/de/reference/arduino-delay

> als zweite quelle wäre das abfangen der reflexionen wenn das bms schläft, kommentiere unter dem delay mal bitte diese zeilen aus

Probier ich mal.

softwarecrash, 2023-01-04:
> > das sind us, sprich 1000 ist 1ms. da kannst ruihg experimentieren, ich denke bis 5000 sollten da keine probleme auftreten, das wären dann gerade mal 5ms.
> 
> Ne, das sind ms ;-) https://arduinogetstarted.com/de/reference/arduino-delay //EDIT: Befehl zurück, bin zu blöd zum Lesen (delayMicroseconds() )
> 
> > als zweite quelle wäre das abfangen der reflexionen wenn das bms schläft, kommentiere unter dem delay mal bitte diese zeilen aus
> 
> Probier ich mal.

hab das mal gaaaanz grob überschlagen, dieser eine debug blog über dem neuen delayMicroseconds braucht braucht zum senden ca 0,05 sekunden. entsprechend würde ich als grenze 6000-8000us ansetzen bis wohin sich eine merkliche verbesserung zeigen müsste

all-solutions, 2023-01-04:
Ok, ich zieh mir die grad runter.

Übrigens: so **könnte** das EPS01-Board aussehen: https://a360.co/3VLXptl
Der OK1 auf der Platine ist ein Dual-Highspeed-Opto.

softwarecrash, 2023-01-04:
> Ok, ich zieh mir die grad runter.
> 
> Übrigens: so **könnte** das ESP01-Board aussehen: https://a360.co/3VLXptl
> Der OK1 auf der Platine ist ein Dual-Highspeed-Opto.

Schaut gut aus, fehlt noch der fet/Transistor für wachhalte Signal und Taster zum manuellen wecken.
Was mich beim esp01 stört ist halt das dass ganze durch die pinheader so hoch aufträgt
Machst du das mit Fusion bzw seinen kostenlosen Nebenteilen? Da muss ich mir glaub Mal ne Lehrstunde bei dir holen ;)

all-solutions, 2023-01-04:
> Machst du das mit Fusion bzw seinen kostenlosen Nebenteilen? Da muss ich mir glaub Mal ne Lehrstunde bei dir holen ;)

Ist Eagle. Der kann dann das PCB zu Fusion 360 hochladen.

Ich hatte da auch schon mit dem ESP12 "gebastelt": https://a360.co/3Io4Qnu

softwarecrash, 2023-01-04:
> > Machst du das mit Fusion bzw seinen kostenlosen Nebenteilen? Da muss ich mir glaub Mal ne Lehrstunde bei dir holen ;)
> 
> Ist Eagle. Der kann dann das PCB zu Fusion 360 hochladen.
> 
> Ich hatte da auch schon mit dem ESP12 "gebastelt": https://a360.co/3Io4Qnu

Das sieht irgendwie schöner aus.
Für flashen würde ich da dennoch ein extra pinheader einplanen, kann ja auch gewinkelt sein oder so, und der wake Button fehlt noch ;)
Ach siehst jetzt weiß ich wieder was noch auf die Wunschliste solle, ein schaltausgang der je nach Spannung oder soc zu oder abschalten kann.
Hatte unter Diskusionen ja was angelegt da kannst gern auch bisherige Ideen und Baupläne reinschreiben

all-solutions, 2023-01-04:
> Für flashen würde ich da dennoch ein extra pinheader einplanen, kann ja auch gewinkelt sein oder so, und der wake Button fehlt noch ;)

Pinheader und Wake-Button stehen noch auf der To-Do. War erstmal ein reiner Entwurf.
> Ach siehst jetzt weiß ich wieder was noch auf die Wunschliste solle, ein schaltausgang der je nach Spannung oder soc zu oder abschalten kann.

Was soll der Schaltausgang an Leistung können sollen?
> Hatte unter Diskusionen ja was angelegt da kannst gern auch bisherige Ideen und Baupläne reinschreiben

Ja, mach ich Morgen mal, passe da vorher noch die (fehlenden) Sachen an ;-).

Delay von 10 bringt übrigens keinen Unterschied. Bevor ich jetzt zig mal neu kompiliere und dem Bekannten das schicke, jag ich den Delay mal auf 500ms hoch und schicks ihm. Wenn es damit läuft, kann ich den ja nach und nach wieder runterschrauben.

softwarecrash, 2023-01-04:
Der schaltausgang muss nur so viel können um ein kleines 5v Relais zu steuern, was eigentlich Quatsch ist den wenn es nur vom BMS gespeist wird steht kaum Leistung zur Verfügung also reichen 20ma, oder gar nur einen weiteren Optokoppler an die Stelle setzen, dann kann man danach selber mit Transistor und Relais basteln wie man will.

Ja du kannst durchaus die Zeit sehr hoch ansetzen, es kann nur passieren das es dann ein wenig hängt.
Alternativ Mal die Zeilen rausnehmen die ich erwähnt hatte. Wenn sich da garnichts tut suchen wir an der falschen Stelle. Es ist leider gerade ein stochern im trüben

softwarecrash, 2023-01-04:
Könntest du deine Bekannten fragen was sie für Modelle haben? (Steht ja immer oben drauf) oder ein Foto davon.

all-solutions, 2023-01-04:
Ich denke mal, das ist die falsche Stelle. Antwort vom Bekannten zur 500ms-Version:
> Mit ADUM keinen Besserung. Im Gegenteil. Sleep kommt schnell und das Teil bleibt weg.
> Transistor das Gleiche Spiel. Zügig im sleep, kein Ping Pong. Auch der bleibt weg.

> Könntest du deine Bekannten fragen was sie für Modelle haben? (Steht ja immer oben drauf) oder ein Foto davon.

Der hat nen 8S 250A mit RS485/CAN. Baugleich mit meinem. Daher verstehe ich es schon garnicht. Ich hab keinerlei Probleme.

softwarecrash, 2023-01-04:
> Ich denke mal, das ist die falsche Stelle. Antwort vom Bekannten zur 500ms-Version:
> > Mit ADUM keinen Besserung. Im Gegenteil. Sleep kommt schnell und das Teil bleibt weg.
> > Transistor das Gleiche Spiel. Zügig im sleep, kein Ping Pong. Auch der bleibt weg.
> 
> > Könntest du deine Bekannten fragen was sie für Modelle haben? (Steht ja immer oben drauf) oder ein Foto davon.
> 
> Der hat nen 8S 250A mit RS485/CAN. Baugleich mit meinem. Daher verstehe ich es schon garnicht. Ich hab keinerlei Probleme.
> Sein Kumpel hat den auch, jedoch ohne RS und CAN, dafür mit Lüfter.

Bei mir werkelt auch ein 8s 250A in einem Batterieblock vorgefertigt, und zum testen ein 7s 40A beide problemlos.

Dann bleibt nur die sleep Funktion (die Zeilen) rauszunehmen ob sich was ändert, delay und debug dafür weg.

Wenn das auch nicht hilft bleibt nur debug aktiv lassen was nicht schlimm wäre.

Ich wüsste Grad absolut nicht woran das noch liegen kann, evt ist bei den beiden nen anderer Chip verbaut der empfindlicher auf die Timings reagiert aber das bekommt man nur raus wenn man die Dinger auseinander nimmt oder röngt.

Auf Verdacht versuchen ein identisches zu bekommen ist mir ein wenig zu kostspielig.

all-solutions, 2023-01-04:
Ich kompiliere denen grad die Version ohne sleep-Zeilen... 
> Ich wüsste Grad absolut nicht woran das noch liegen kann, evt ist bei den beiden nen anderer Chip verbaut der empfindlicher auf die Timings reagiert aber das bekommt man nur raus wenn man die Dinger auseinander nimmt oder röngt.

Das Teil vom Bekannten hat sogar die gleiche Hardware-Version. Was aber bei den Schlitzaugen ja nichts heißen muss...
> Wenn das auch nicht hilft bleibt nur debug aktiv lassen was nicht schlimm wäre.

Ja, es stört ja nicht. Auch wenn es mich (und denke Dich auch) wurmt...

softwarecrash, 2023-01-04:
> Ich kompiliere denen grad die Version ohne sleep-Zeilen...
>
> Das Teil vom Bekannten hat sogar die gleiche Hardware-Version. Was aber bei den Schlitzaugen ja nichts heißen muss...
>
> Ja, es stört ja [...]

Hardware Version heißt im Land der aufgehenden sonne nur das es das gleiche gehäuse ist 😅

Wurmt ist noch sehr milde ausgedrückt, hab heut den halben Tag den Code hoch und runter studiert und frustriert festgestellt das bis auf einige programmier Patzer und anderer Kram der unschön ist das ganze in sich problemlos laufen sollte.... Was es ja auch macht, nur eben nicht bei jedem.... Argh!

all-solutions, 2023-01-04:
> Wurmt ist noch sehr milde ausgedrückt, hab heut den halben Tag den Code hoch und runter studiert und frustriert festgestellt das bis auf einige programmier Patzer und anderer Kram der unschön ist das ganze in sich problemlos laufen sollte.... Was es ja auch macht, nur eben nicht bei jedem.... Argh!

Jaja, das kenn ich. Ist mir bei meinem GW60-Superrollo-Projekt (mittlerweile auch von Gitlab hierhin gezogen) auch so gegangen. Und da hatte ich alle vier GW60 hier. Die haben Toleranzen wie Scheunentore. Bis ich das alles soweit hatte, dass auch wirklich alle GW's damit klar kamen... 

Antwort zur "no sleep"-Version:
>Also zuerst das ADUM. Es ist "besser" geworden. Alle 30-50 Sekunden kommt der "sleep". Der dauert aber nur etwa 5 Sekunden an, dann ist das Teil wieder da.

Transistor-Version folgt... (trag ich dann hier mit rein)

softwarecrash, 2023-01-04:
> > Wurmt ist noch sehr milde ausgedrückt, hab heut den halben Tag den Code hoch und runter studiert und frustriert festgestellt das bis auf einige programmier Patzer und anderer Kram der unschön ist das ganze in sich problemlos laufen sollte.... Was es ja auch macht, nur eben nicht bei jedem.... Argh!
> 
> Jaja, das kenn ich. Ist mir bei meinem GW60-Superrollo-Projekt (mittlerweile auch von Gitlab hierhin gezogen) auch so gegangen. Und da hatte ich alle vier GW60 hier. Die haben Toleranzen wie Scheunentore. Bis ich das alles soweit hatte, dass auch wirklich alle GW's damit klar kamen... 
> 
> Antwort zur "no sleep"-Version:
> >Also zuerst das ADUM. Es ist "besser" geworden. Alle 30-50 Sekunden kommt der "sleep". Der dauert aber nur etwa 5 Sekunden an, dann ist das Teil wieder da.
> 
> > Bei Transistor sieht es besser aber ähnlich aus. Hier dauert aber eine "Runde" zwischen 1,5 und 2,5 Minuten. Der Aussetzer selbst liegt max. 5 Sekunden.
>  

Ok also in Summe alles ein wenig Murks außer debug ist aktiv dann rennt der Krempel sauber?

Gw60 Rollo? Ah Google erklärt, ein gurtwickler.
Hab leider keine Rollos, daher müsste ich mir mit Ikea verdunkelungs Rollos ein paar Schrittmotoren und ESP behelfen

softwarecrash, 2023-01-04:
Noch ein smart-ie hab irgendwie auch nichts anderes erwartet 😅

An fehm bin ich gepflegt vorbeigelaufen, node-red hab ich auch als "Plugin" komm aber eigentlich aus der iobroker Fraktion.

Na gut dann belassen wir das Mal dabei das debug aktiv bleibt auf Dauer, ich werd die Ausgaben noch ein wenig hübsch machen ggf etwas mehr.

Aber Werd mir das trotzdem nochmal ansehen, irgendwas kann da nicht ganz stimmen mit der Funktion die prüft ob Verbindung besteht, das müsste länger brauchen um es zu registrieren.

all-solutions, 2023-01-04:
> Noch ein smart-ie hab irgendwie auch nichts anderes erwartet 😅

Naja, wer auch an sowas "bastelt", kann doch nur ein Smart-ie sein...
> An fehm bin ich gepflegt vorbeigelaufen, node-red hab ich auch als "Plugin" komm aber eigentlich aus der iobroker Fraktion.

ioBroker hab ich mir mal angesehen und für meine "Anforderungen" als Oversized deklariert. Hier will keiner in der Familie z.B. irgendwelche GUI's. Hauptsache alles läuft. Ansonsten höre ich immer nur "Schaaaatz, das geht nicht" oder von den Stieftöchtern "Manuuu, die Alexa macht nicht, was ich ihr sage". Und so bin ich vor Jahren (da war es noch Version 0.x.x.) zu Node-Red gekommen. Läuft bei mir auf dem Synology-NAS im Docker.
> Na gut dann belassen wir das Mal dabei das debug aktiv bleibt auf Dauer, ich werd die Ausgaben noch ein wenig hübsch machen ggf etwas mehr.

Ja, für jetzt würde ich das so lassen. Ich kann es leider nicht selbst reproduzieren. Sonst wären wir vielleicht ein Stück weiter.  
> Aber Werd mir das trotzdem nochmal ansehen, irgendwas kann da nicht ganz stimmen mit der Funktion die prüft ob Verbindung besteht, das müsste länger brauchen um es zu registrieren.

Mach Dich da nicht wahnsinnig mit. Wenn es mit dem Debug läuft, lass es erstmal so. Frisst ja kein Brot. 
Und ja, ich weiß, es lässt einem keine Ruhe ;-).

softwarecrash, 2023-01-05:
> Naja, wer auch an sowas "bastelt", kann doch nur ein Smart-ie sein...
>
>
> ioBroker hab ich mir mal angesehen und für meine "Anforderungen" als Oversized deklariert. Hier will keiner in der Familie [...]

naja iobroker und diese VIS, brauch ich nicht, bei mir ist die einzige menschliche schnittstelle ein paar alexas und telegram. ich brauch kein virtuelle wohnung an der wand in der wohnung, da könnte ich mir den unfug auch sparen, das soll alles autark laufen.

> > Na gut dann belassen wir das Mal dabei das debug aktiv bleibt auf Dauer, ich werd die Ausgaben noch ein wenig hübsch machen ggf etwas mehr.
> 
> Ja, für jetzt würde ich das so lassen. Ich kann es leider nicht selbst reproduzieren. Sonst wären wir vielleicht ein Stück weiter.

so ohne anhaltspunkt ist es aussichtslos was zu finden.

> Mach Dich da nicht wahnsinnig mit. Wenn es mit dem Debug läuft, lass es erstmal so. Frisst ja kein Brot. Und ja, ich weiß, es lässt einem keine Ruhe ;-).

ich hab mir mein test bms mal nach hause geholt, ch werd wohl einfach das debug als festen bestandteil mit einbauen und bisl ausweiten. evt findet sich mit der zeit der fehler noch.

softwarecrash, 2023-01-05:
hab das ganze jetzt ne weile laufen mit und ohne debug, das einzige was ab und zu mal fehlschlägt ist die zellabfrage, aber das ist vielleicht alle 20 abfragen. das dürfte aber auch nicht dafür sorgen das deine bekannten garkeine daten bekommen.

softwarecrash, 2023-01-05:
so nach ein paar kleinigkeiten später und ein wenig stromspar masnahmen hab ich mit debug aktiv seit einer stunde keinen crc fehler.

nowaczuk, 2023-01-09:
Hallo,
leider verliere auch ich immerwieder die verbindung zum bms, so etwa 2-2,5h dann wird nurnoch "No connection or Sleeping
" angezeigt und mqtt aktualisierungen gibt es dann auch keine mehr :(
könnte man mir vll eine version mit aktiviertem debugging zukommen lassen, damit ich schauen kann ob das disconnecten bei mir dann auch weg ist?
wäre ich sehr dankbar drüber.

softwarecrash, 2023-01-09:
> Hallo,
> leider verliere auch ich immerwieder die verbindung zum bms, so etwa 2-2,5h dann wird nurnoch "No connection or Sleeping
> " angezeigt und mqtt aktualisierungen gibt es dann auch keine mehr :( [...]

Sicher das die Verbindung weg ist und das bms nicht einfach schlafen geht?
Das Problem mit dem Connection lost, ist eigentlich Instant.

nowaczuk, 2023-01-09:
hmmm, wodran erkenne ich denn, das das bms schlafen geht? warum/wie schläft das überhaupt?
weis nur das die verbindung immerwieder weg ist und stellenweise über stunden keine neuen daten etc rein kommen

softwarecrash, 2023-01-09:
> hmmm, wodran erkenne ich denn, das das bms schlafen geht? warum/wie schläft das überhaupt?
> weis nur das die verbindung immerwieder weg ist und stellenweise über stunden keine neuen daten etc rein kommen

Die Zeit bis das bms schlafen geht (Stromsparmodus) kannst du in der App oder am PC einstellen.
Wenn innerhalb dieser Zeit keine Ladung rein oder raus geht dann geht das Teil schlafen und schaltet Kommunikation und uart-Strom ab. Erst wenn eine Last gezogen wird oder Ladung kommt die über einem bestimmten Grenzwert geht wacht das bms wieder auf und sendet auch Daten.

nowaczuk, 2023-01-09:
hmm, ok, dann werde ich das mal beobachten bzw den wert für das "schlafen gehn" auf maximum setzen, ist wenn ich das nun richtig sehe 65535 also etwa 18 stunden, denke (hoffe) das sollte dann reichen
vielen dank für die info und hilfe :)
und auch danke für die tolle arbeit :)

softwarecrash, 2023-01-09:
> hmm, ok, dann werde ich das mal beobachten bzw den wert für das "schlafen gehn" auf maximum setzen, ist wenn ich das nun richtig sehe 65535 also etwa 18 stunden, denke (hoffe) das sollte dann reichen
> vielen dank für die info und hilfe :)
> und auch danke für die tolle arbeit :)

Genau, Versuch das erstmal.
Ansonsten gibt es noch die Möglichkeit den wake Pin auf gnd zu setzen dann bleibt das bms definitiv gezwungen an.

softwarecrash, 2023-01-09:
@all-solutions wenn du magst komm Mal auf den discord Server, ich hab da evt eine Idee wegen dem Connect Problem.
Muss ich Mal einbauen dann stell ich es dir zur verügung, ist nur vlöder Gedanke aber evt klappt das.

all-solutions, 2023-01-09:
Hi Rocky, ich komm Morgen rein. Können das dann mal testen.

nowaczuk, 2023-01-10:
@softwarecrash hattest scheinbar recht, bei mir liegt der disconnect wohl am sleep vom bms
welches wäre denn der wake pin den ich mit gnd verbinden müsste, damit es dauerhaft on bleibt?

all-solutions, 2023-01-10:
> welches wäre denn der wake pin den ich mit gnd verbinden müsste, damit es dauerhaft on bleibt?

Pin 4 ;-)

[image]

softwarecrash, 2023-01-10:
> > welches wäre denn der wake pin den ich mit gnd verbinden müsste, damit es dauerhaft on bleibt?
> 
> Pin 4 ;-)
> 
> [image]

darf ich das bild ins repo aufnehmen?

komisch das pin 3 12v sein soll. ich hab andere dokumente wo zb 15v angegeben ist und nur 50ma rauskommen

softwarecrash, 2023-01-10:
> > darf ich das bild ins repo aufnehmen?
> 
> Klar, ich kanns Dir auch ohne gelb markiertem Pin 4 schicken.
> 
> > komisch das pin 3 12v sein soll. ich hab andere dokumente wo zb 15v angegeben ist und nur 50ma rauskommen
> 
> Hab ich auch. Da streiten sich wohl die Geister bei DALY. Der Powerboard-Anschluss muss (zumindest bei neueren BMS) auch eine komplett andere Belegung haben. Ich habe da z.B. das Interface-Board von DALY dran. Das kommuniziert bestimmt nicht per "LED-out" mit dem BMS... [image]

gern.

ja das is chinesische dokumentation.... muss man echt aufpassen manchmal

all-solutions, 2023-01-10:
> gern.

Ich hab mal meine "gesammelten Werke" zu DALY in eine Repo hochgeladen.
https://github.com/all-solutions/DALY-docs-and-soft
Such Dir raus, was Du gebrauchen kannst ;-). Das 'docs/DALY BMS Pinout.jpg' ist das besagte Bild:
 
[image]

softwarecrash, 2023-03-08:
ich mach mal zu, glaube das ist erledigt

closed 2023-03-08 by softwarecrash