softwarecrash/Daly2MQTT #48: MQTT "Pack" Werte = NULL

wauzzi, 2023-01-18:
Es werden keine Pack Werte via MQTT übermittelt

[image]

derLoosi, 2023-01-19:
Servus,

das Topic Device_Control ist zum SETZEN von Werten, siehe hierzu die Readme bzw. das Wiki. Dieses wird beim Start auf null gesetzt. Trägst du jetzt einen Wert zum Beispiel bei Pack_SOC ein wird der SOC vom Daly damit überschrieben.

wauzzi, 2023-01-19:
Danke für Deine Rückmeldung,

wenn ich die Werte manuell eintrage funktioniert es tatsächlich. Vielen Dank. Allerdings wäre es schön, wenn der Zustandstyp nichts als Zeichenkette, sondern als Logikwert ("type": "boolean") initial angelegt wird. Ich habe es händisch geändert und es funktioniert als boolean. Also SOC könnte man ja initial den Wert aus"Pack_SOC" eintragen.
Vielleicht kannst Du das noch anpassen.

derLoosi, 2023-01-19:
Ich werde es mal mit in die Vorschlagliste übernehmen. Dies ist allerdings ein Verhalten welches vom IOBroker abhängig ist. Bei einem "normalen" Broker (nicht das was man über ioBroker als Instanz installiert) tauchen die Topics tatsächlich erst auf wenn man sie selbst beschreibt, das ist eigentlich auch der normalfall. Entscheiden muss das aber @softwarecrash .

softwarecrash, 2023-01-19:
das ist wirklich eine eigenheit von IOB, der esp subscribed lediglich die topics, iob erstellt diese dann (neuerdings) was gut ist sonst müsste man mit scripten sich erst punkte anlegen. beim subscribe gibt es allerdings keine möglichkeit zu sagen was der datenpunkt ist, wie auch, der esp aboniert ja lediglich den pfad, da iob nichts weiter darüber weis erstellt er ihn einfach als default mit string als wert. diese device control pfade haben garkein publish. man müsste entweder beim start bevor subscribed wird die publishen mit dem letzten datenwert true|false was dann nicht geht wenn kein bms dran ist oder grad keine daten. oder per default ein wert setzen den sich der esp dann wieder abholt und unvorhersehbar irgendwas schaltet. 
da müsste man wirklich mal schauen ob oder wie man das lösen kann.

closed 2023-01-19 by derLoosi