softwarecrash/Daly2MQTT #49: Discharge MOS State: ohne Funktion

wauzzi, 2023-01-18:
Wird der Schalter "Discharge MOS State:" betätigt, wird das MOS nicht angesteuert

derLoosi, 2023-01-19:
Servus,

hast du eventuell Popups blockiert? Eigentlich sollte eine Warnung kommen ob du den Discharge Mosfet wirklich abschalten möchtest, diese muss erst bestätigt werden. Kommt die bei dir?

wauzzi, 2023-01-19:
Danke für deine schnelle Rückmeldung. Über den Chrome kommt zwar die Abfrage, passieren tut aber nichts. Ich habe es eben über mein Android Chrome probiert. Dort tut es. Ich werde es heute Abend nochmals auf dem Rechner testen

derLoosi, 2023-01-19:
Kann es eventuell sein dass es nur einen Moment dauert bis er das übernimmt? Wenn der WeMos grad abfragen ans BMS sendet verzögert das das "setzen" und der schalter springt ggf. erst kurz auf inaktiv und einen moment später wieder aktiv.

derLoosi, 2023-01-20:
Hallo @wauzzi ,

konntest du schon weitere Tests durchführen?

wauzzi, 2023-01-21:
Sorry, ich kam erst heute wieder zum Testen. Es lag tatsächlich am AdBlocker. Des MOS schaltet jetzt war den Discharge zwar ab, wird jedoch nach Paar Minuten selbstständig wieder aktiviert

derLoosi, 2023-01-22:
Firmware Version:
- DALY BMS to MQTT V1.0.0

Verwendete Software:
- Firefox 109.0
- Mosquitto MQTT Server
- MQTT Explorer 0.4.0-beta1

19:48Uhr - Discharge über Webinterface deaktiviert
20:36Uhr - immernoch aus, dann hab ich ihn wieder über das Webinterface aktiviert.

MQTT Meldungen kamen die ganze zeit über rein. #50

wauzzi, 2023-01-22:
Ich glaube, ich habe den Fehler gefunden.

Sofern Discharge über MQTT auf true steht, jedoch im Webinterface der Schalter auf false steht, wird der Wert über MQTT aus dem IOBroker wieder übernommen und somit wieder aktiviert

softwarecrash, 2023-01-22:
> Ich glaube, ich habe den Fehler gefunden.
> 
> Sofern Discharge über MQTT auf true steht, jedoch im Webinterface der Schalter auf false steht, wird der Wert über MQTT aus dem IOBroker wieder übernommen und somit wieder aktiviert. Daher wäre es gut, wenn die tatsächlichen Zustände der MOS wieder via MQTT an den IOBroker zurückgegeben werden 

Das kann nur passieren wenn die WLAN Verbindung schlecht ist, dann macht er einen reconnect und subscribt die Werte neu.
Wie ist der WLAN Empfang laut deinem Router zu dem esp?

wauzzi, 2023-01-22:
> 
[image]

derLoosi, 2023-01-22:
> Daher wäre es gut, wenn die tatsächlichen Zustände der MOS wieder via MQTT an den IOBroker zurückgegeben werden

Das werden sie ja, allerdings nicht im Device_Control bereich. Sonst würde man ja eine "Schleife" bauen und den Wert immer wieder mit sich selbst überschreiben.

closed 2023-01-22 by wauzzi