softwarecrash/Daly2MQTT #50: MQTT keine Aktualisierung mehr der Daten

wauzzi, 2023-01-21:
Nach einer Erfolgreichen MQTT Verbindung werden die Daten korrekt an meinen IOBROKER MQTT Adapter übermittelt. Nach ca. 15 Minuten werden jedoch keine Daten mehr aktualisiert. Erst ein Reboot des WEMOS liefert wieder aktuelle Daten

softwarecrash, 2023-01-21:
welcher wert ist bei sleeptime im BMS hinterlegt?
wie ist der ESP verkabelt?
wo bekommt er seinen strom her?
was für ein BMS ist im einsatz?
bitte den mqtt adapter auf debug log stellen im iobroker und den teil an dem die verbindung abbricht hier posten.

wauzzi, 2023-01-21:
> welcher wert ist bei sleeptime im BMS hinterlegt? -->2999
wie ist der ESP verkabelt? --> Über den ADUM1201
wo bekommt er seinen strom her? --> Über eine separate externe USB Verbindung 
was für ein BMS ist im einsatz? Daly BMS-200A 16S (mit Ventialtor)
bitte den mqtt adapter auf debug log stellen im iobroker und den teil an dem die verbindung abbricht hier posten. -->Mach ich morgen

wauzzi, 2023-01-22:
So ich hab jetzt mal den MQTT Adapter quasseln lassen. Wenn die Verbindung abbricht, erscheint im Log vom Daly nur noch dieser Eintrag:
<html>
<body>

mqtt.0 | 23525 | 2023-01-22 19:07:24.496 | debug | Client [DALY-BMS-to-MQTT] pingreq
-- | -- | -- | -- | --

</body>
</html>

wauzzi, 2023-01-22:
Ich hab jetzt mal eine separate MQTT Instanz installiert und lass das Debug gerade laufen. Sobald der Fehler wieder auftritt, poste ich den LOG hier rein

wauzzi, 2023-01-23:
So ich hab jetzt den Daly--> MQTT auf eine separate MQTT Instanz laufen lassen. Auf dieser Instanz lässt sich der Fehler (leider) nicht mehr reproduzieren. Wer jetzt für den Ausfall daran "Schuld" ist, ist fraglich. Ich werde das Verhalten weiter beobachten.
Dennoch danke fürs supporten @softwarecrash

closed 2023-01-23 by wauzzi