softwarecrash/Daly2MQTT #56: What to do to convert you FW to ESP-01S

profra, 2023-02-07:
Hello,
I have a lot of old ESP-01S and 3 battery packs with BMS DALY "Lifepo4 16S 48V 200A" (I have my own photovoltaics on the roof of the house). I would like to transfer the BMS status information wirelessly using MQTT, not via a UART-USB converter. Could you provide the FW variant for this module with ESP8266 or advise what to adjust in your FW?
Thank you for answer.

softwarecrash, 2023-02-07:
hello,
i think it will work with ESP-01 to.
for future use, with the next release it will switched to other serial ports

profra, 2023-02-09:
ESP-01S only has 1MB flash, can you advise how to do it?

softwarecrash, 2023-02-09:
ah, in platformio.ini you must modify the partition settings and find the right config
for wemos d1 is board_build.ldscript = eagle.flash.4m2m.ld	; 4MB (FS:2MB OTA:~1019KB)
so try to comment out this line and flash directly. web update will not work.

profra, 2023-02-09:
I am sending a picture from a web browser... I have 16 cells in the battery pack and as you can see in the picture there are only 6... sometimes the number 16 appears, but mostly there is the wrong number of cells... please look at that... a formatting error?

[image]

softwarecrash, 2023-02-09:
thats correct, it displays the highest and lowest cell number with the voltage, not the whole cell pack. the descriptor left say Hi/Low/diff

profra, 2023-02-09:
Ah, "my oak head"...you're right...sorry

closed 2023-02-09 by softwarecrash