softwarecrash/Daly2MQTT #57: [GER] PortForward Support

robertdiers, 2023-02-13:
Hi,

Ich betreibe die Software hinter einen dedizierten Router mit eigenem Netzwerk:
-> MQTT klappt soweit mit einer Verbindung in ein anderes Netzwerk vor dem Router
-> Die Webseite wird innerhalb des Netz korrekt dargestellt

Ich habe auf dem Router ein PortForward eingerichtet, damit wird nur noch die Webseite ohne Daten angezeigt.
Im Browser sieht man auch gut das er versucht _ws://192.168.1.8/ws_ aufzurufen, das ist aber die Router IP und damit steht kein WebSocket zur Verfügung...

MQTT läuft natürlich, deswegen eher "nice to have".

Gruß Robert

softwarecrash, 2023-02-13:
hallo,
das ist etwas schwierig nachzustellen.
wenn er versucht bei dir die 1.8 aufzurufen per ws was steht als url im browser?

robertdiers, 2023-02-13:
Hi,

http://192.168.1.8:8084
Ich habe mehrere Daly hinter dem gleichen Router - alle PortForwards auf unterschiedlichen Ports (8084 bis 8086).
Für den WS Aufruf scheint die GUI einfach nur die IP zu nutzen, aber das routet dann natürlich nicht.

Gruß Robert

robertdiers, 2023-02-13:
[image]

robertdiers, 2023-02-13:
[image]

softwarecrash, 2023-02-13:
Ah okay, also müsste nur der angehängte Port falls vorhanden übernommen werden.
Das sollte möglich sein.
Ich würde dir ja ein Testfile zur Verfügung stellen, aber wir arbeiten gerade an v2 wo sich fast alle verdrahtungen ändern

robertdiers, 2023-02-13:
ist wie gesagt nicht dringend, da MQTT ja funktioniert - aber eventuell ist es einen Einbau in eine kommende Version wert ;-)

softwarecrash, 2023-02-14:
@robertdiers probier die firmware mal
[Daly v1.0.1 ws fix.zip](https://github.com/softwarecrash/DALY-BMS-to-MQTT/files/10730402/Daly.v1.0.1.ws.fix.zip)

robertdiers, 2023-02-14:
Das sieht gut aus - sehr vielen Dank! 
[image]

softwarecrash, 2023-02-14:
super, dann fliest das in die dev mit ein.
und vielen dank ;)

kann das dann zu?

robertdiers, 2023-02-14:
klar - Danke!

closed 2023-02-14 by softwarecrash