softwarecrash/Daly2MQTT #67: Keine Werte im iobroker

RomanWujnovic, 2023-03-01:
Hallo Softwarecrash,
Ich schreibe mal auf deutsch denn mein Englisch ist grauenvoll.
Eigentlich wollte ich mich hier nicht anmelden da ich für hier der falsche bin.
Aber irgendwie komme ich nicht weiter, und im Netz habe ich schon unzälige Stunden gesucht.
Aber nichts gefunden, vielleicht bin ich auch nicht soweit in der Lage das alles zu verstehen.
Zu meinem Problem:
Ich bin durch den youtube Kanal von "der Kanal" auf dich aufmerksam geworden.
Da hat er deine Anwendung vorgestellt um das BMS auszulesen.
Genau das was ich gebraucht habe.
Also ich wie nix die Bauteile bestellt und zusammen gelötet.
Und jetzt ds eigentliche Problem:
Ich bin zwar schon eine weile mit dem iobroker unterwegs (Smarthome, Poolsteuerung, kleine einfache Scripteusw.) aber irgendwie 
klappt da was nicht ganz.
Und zwar habe ich alles so gemacht wie auf deiner Seite beschrieben und über den browser bekomme ich auch Werte vom BMS.
Allerdings ändern sich die Werte erst nach dem gefühlten zehnten mal aktualisieren des browsers (ist das normal?) oder nach einem reboot.
Und das andere Problem ist der mqtt adapter, bisher habe ich den nämlich noch garnicht gebraucht. Und somit kenne ich mich mit dem mqtt garnicht aus. Ich habe zwar diesbezüglich schon viel im Netz gesucht aber nichts gefunden was mir half.
Ich habe es zwar hinbekommen das der adapter grün wurde und somit verbunden ist.
Und auch in den Objekten wurden auch die Ordner erstellt.
Aber leider etwas wenig und ohne Daten.
Chargefet                              Null
Dischargefet                          Null
Packsoc                                Null
Relais                                    Null
Daly-bms-to-mqtt               true
Device ip                               ip adresse vom wemos
Alive                                     true

Ich hatte auch alle Firmware Versionen mit den dazugehörigen Lötarbeiten durchgeführt aber nichts hat zum gewünschten 
Ergebnis geführt.
Heute habe ich deine letzte Version getestet.
Da kamen nur diese neuen Werte mit dazu:
Daly-bms-to-mqtt               true
Device ip                               ip adresse vom wemos
Alive                                     true
Aber im iobroker das selbe Ergebnis, keine Werte von soc oder Zellenspannung.
Habe ich vielleicht beim mqtt irgendwo verggessen etwas bestimmtes zu aktivieren oder einen Haken zu setzten.
Oder habe ich ein ganz anderes problem.
Ich habe wirklich schon alles Versucht und komme nicht weiter.
Sorry das ich dich damit belästige aber vielleicht kannst du mir helfen.

Gruß Roman

softwarecrash, 2023-03-01:
hi,
wenn du werte im webinterface bekommst scheint die verbindung zum bms ja zu stehen, die werte im webinterface sollten sich alle 3-5 sekunden aktualisieren.

das dir im mqtt nur so wenig werte angezeigt werden ist nur dann normal wenn keine verbindung zum bms besteht, dann sendet er nichts weiter.

es könnte sein das du eins der bms hast was sehr zickig ist, sind selten aber vorhanden.
probier mal diese bin datei für den wemos, da ist debug aktiv was aufgrund der sendezeit abhilfe schaffen kann.
[Daly-BMS-to-MQTT_d1_mini_2.2.8.zip](https://github.com/softwarecrash/DALY-BMS-to-MQTT/files/10863923/Daly-BMS-to-MQTT_d1_mini_2.2.8.zip)
damit hast du auch gleichzeitig die möglichkeit über usb und eine serielle konsole zu sehen was falsch laufen könnte

RomanWujnovic, 2023-03-01:
hi,
ich danke dir für deine schnelle Antwort und unterstützung.
Also von selber aktualisieren sich die Werte im Webinterface garnicht. Erst wenn ich den browser aktualisiere aber auch dann bekomme ich erst nach dem gefühlten zehnten mal aktualisieren andere werte. oder nach einem reboot.
Danke für die bin datei ich werde es mal damit versuchen.
Und war ja klar das ich wieder so ein zickiges ding bekomme. :-)

softwarecrash, 2023-03-01:
falls du die ganz aktuelle version hast, hast es aber schon nach der neuen anleitung verkabelt?

RomanWujnovic, 2023-03-01:
Ja ich hatte vorher die verkabelung mit dem adum 1201.
mit unterschiedlichen Softwareversionen.
Und jetzt mit der aktuellen Version nur mit dem Wemos und den Anschlüssen ground und d6,d7

RomanWujnovic, 2023-03-01:
also ich habe soeben die andere version drauf gezogen. Gleiches verhalten werte aktualisieren sich nicht von selbst im webinterface  und die ordner im iobroker sind auch wie ich vorher geschrieben habe.
muss im mqtt adapter was eingestellt werden was bei neuinstalation des adapters nicht gleich aktiv ist

softwarecrash, 2023-03-01:
eigentlich muss erstmal nichts weiter eingestellt werden.
wenn keine mqtt datenpunkte angelegt werden bzw nicht gesendet, dann hat das system keine gültigen informationen von deinem bms.
was sagt den beim webinterface das console log? das kannst du bei firefox/chrome mit F12 aufrufen, da müsste was stehen von connected oder so.

kannst du an dein esp ein usb kabel anschliesen und das serielle log abgreifen?

RomanWujnovic, 2023-03-01:
Hi,
In der Konsole stehen drei Sachen. 
connection closed
Trying to open a websocket connection…..
Connection opened 
Usb Kabel kann ich am esp anschließen wie das serielle abgreifen funzt weiß ich nicht. Dafür bin ich zu wenig in eurer Materie drin. Ich kann aber erst am Samstag wieder danach schauen. Die nächsten Tage habe ich leider andere Projekte am Start.

softwarecrash, 2023-03-01:
das ist soweit okay.

dann schau mal nach putty und comport bei google da gibt es einige anleitungen

RomanWujnovic, 2023-03-02:
Servus softwarecrash, also das Putty program sagt mir was. Das benützte ja auch für das Terminal beim raspi. Muss ich mich mal mit putty und wemos auseinander setzten was man da so machen kann. 

wobei Heute hatte ich mich des Öfteren wenn ich Zeit hatte per VPN auf das webinterface zu Hause geschaltet. Also ich glaube fast langsam bei mir liegt es definitiv am bms das zickt. Denn nach der letzte Software Version inklusive Anpassung am wemos ( Anschluss Anpassung) Bekomme ich gar keine aktuellen Werte mehr über das aktualisieren des Browsers. Sondern nur noch wenn ich den wemos Reboote. Aber ich baue mir momentan einen weiteren Speicher wo ich ein weiteres bms brauche. Den bestell ich das bms einfach jetzt schon.

softwarecrash, 2023-03-02:
du musst nur schauen welchen com port der wemos bekommt wenn du ihn ansteckst, bei putty dann auf im verbindungsfenster auf serial schalten und den comport reinschreiben z.b. COM4 und baud rate 9600 dann verbindet er sich und du bekommst die ganzen log infos was er grad so treibt

softwarecrash, 2023-03-03:
ich hab noch ein zwei kleine fehler gefunden und in einem neuen release beseitigt.
das dürfte dein problem vermutlich aber nicht entgültig beheben.
es gibt seit v2 aber auch blink codes mit der eingebauten LED des wemos, die dir grob auskunft geben was los ist. da kannst du auch mal schauen wie sie blinkt

LindemannKammerer, 2023-03-04:
Hallo Softwarecrash,
ich habe genau das gleiche Problem wie RomanWujnovic. Im Webinterface werden die Werte angezeigt, aber nicht aktualisiert.
Über MQTT kommt nur:
alive = true
Device_IP = xxx.xxx.xxx.xxx
Ich habe auch die Debug Version 2.2.10 installiert und gesehen, dass alle Zellspannungen ausgelesen werden. Aber nach der letzten Zelle (12S BMS 100A) kommt eine Fehlermeldung, als ob versucht wird weitere Zellen auszulesen.

<DALY-BMS DEBUG> CRC: Calc.: 44 Rec.: 44
                                        <DALY-BMS DEBUG> Frame No.: 1 Cell No: 1. 3291mV
<DALY-BMS DEBUG> Frame No.: 1 Cell No: 2. 3291mV
<DALY-BMS DEBUG> Frame No.: 1 Cell No: 3. 3292mV
<DALY-BMS DEBUG> CRC: Calc.: 46 Rec.: 46
                                        <DALY-BMS DEBUG> Frame No.: 2 Cell No: 4. 3292mV
<DALY-BMS DEBUG> Frame No.: 2 Cell No: 5. 3292mV
<DALY-BMS DEBUG> Frame No.: 2 Cell No: 6. 3291mV
<DALY-BMS DEBUG> CRC: Calc.: 45 Rec.: 45
                                        <DALY-BMS DEBUG> Frame No.: 3 Cell No: 7. 3291mV
<DALY-BMS DEBUG> Frame No.: 3 Cell No: 8. 3291mV
<DALY-BMS DEBUG> Frame No.: 3 Cell No: 9. 3291mV
<DALY-BMS DEBUG> CRC: Calc.: 47 Rec.: 47
                                        <DALY-BMS DEBUG> Frame No.: 4 Cell No: 10. 3292mV
<DALY-BMS DEBUG> Frame No.: 4 Cell No: 11. 3291mV
<DALY-BMS DEBUG> Frame No.: 4 Cell No: 12. 3291mV
<DALY-BMS DEBUG> Error: Received the wrong number of bytes! Expected 13, got 0
<DALY-BMS DEBUG> RX Buffer: [,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0]

Kannst Du damit etwas anfangen? Ich kann gerne noch weitere Daten liefern.
Anbei noch die gesamte Debug Ausgabe.

[BMS_Debug.txt](https://github.com/softwarecrash/DALY-BMS-to-MQTT/files/10889264/BMS_Debug.txt)

softwarecrash, 2023-03-04:
Hallo,
kannst du bitte das Log Mal etwas länger mitschreiben lassen?

Rufe Mal ip/livejson auf und kopiere die Werte hier her.

Wie sieht deine config im ESP aus?
Und wie schaut deine config vom iobroker mqtt  Adapter aus?

softwarecrash, 2023-03-04:
@RomanWujnovic hast du auch ein 16S ?

LindemannKammerer, 2023-03-04:
Hallo,
die livejson Werte sind:
{"Device":{"Name":"DALY-BMS-to-MQTT","IP":"192.168.178.109","ESP_VCC":3.031,"Relais_Active":false,"Relais_Manual":false,"Free_Heap":34200,"json_memory_usage":1539,"json_capacity":2048,"runtime":130,"ws_clients":1,"HEAP_Fragmentation":5,"free_blocksize":32552},"Pack":{"Voltage":39.4,"Current":-4.8,"Power":-189.12,"SOC":49.4,"Remaining_mAh":92378,"Cycles":1,"BMS_Temp":17,"Cell_Temp":0,"High_CellNr":12,"High_CellV":3.285,"Low_CellNr":9,"Low_CellV":3.283,"Cell_Diff":2,"DischargeFET":true,"ChargeFET":true,"Status":"Discharge","Cells":12,"Heartbeat":167,"Balance_Active":false},"CellV":{"CellV_1":3.284,"Balance_1":false,"CellV_2":3.284,"Balance_2":false,"CellV_3":3.285,"Balance_3":false,"CellV_4":3.285,"Balance_4":false,"CellV_5":3.285,"Balance_5":false,"CellV_6":3.284,"Balance_6":false,"CellV_7":3.285,"Balance_7":false,"CellV_8":3.284,"Balance_8":false,"CellV_9":3.285,"Balance_9":false,"CellV_10":3.284,"Balance_10":false,"CellV_11":3.285,"Balance_11":false,"CellV_12":3.284,"Balance_12":false},"CellTemp":{"Cell_Temp_1":0}}

Die ESP Config ist nur MQTT Server IP ohne Benutzer und Password. Angehängt die Config und eine länger Debugausgabe.
Bei der längeren Aufzeichnung kamen jetzt einmal MQTT Daten, aber ich verwende kein iobroker aktuell nur über mqtt explorer.

[BMS_Debug_long.txt](https://github.com/softwarecrash/DALY-BMS-to-MQTT/files/10889335/BMS_Debug_long.txt)
[image]

RomanWujnovic, 2023-03-04:
Hallo zusammen, ich kann heute nicht viel dazu beitragen da ich heute unterwegs bin. Ich kann nur sagen das ich jetzt mehr Daten im IoBroker habe. Aber nur die einzelzellen Spannung regelmäßig aktualisiert wird. Und ich habe ein 12 s System.

softwarecrash, 2023-03-04:
@LindemannKammerer probier diese mal bitte
[Daly-BMS-to-MQTT_d1_mini_2.2.11-a_debug.zip](https://github.com/softwarecrash/DALY-BMS-to-MQTT/files/10889364/Daly-BMS-to-MQTT_d1_mini_2.2.11-a_debug.zip)

softwarecrash, 2023-03-04:
@LindemannKammerer in der a war ein fehler.
hier die korrigierte b version, das sollte jetzt besser laufen
[Daly-BMS-to-MQTT_d1_mini_2.2.11-b_debug.zip](https://github.com/softwarecrash/DALY-BMS-to-MQTT/files/10889427/Daly-BMS-to-MQTT_d1_mini_2.2.11-b_debug.zip)

LindemannKammerer, 2023-03-04:
Das war's! Jetzt werden die Daten im Webinterface aktualisiert und ich erhalte den kompletten MQTT-Datensatz und keine Fehlermeldungen in der Debug-Ausgabe. Auch schon mit der a Version.
Vielen Dank softwarecrsh. Lag es an dem 12S System?

RomanWujnovic, 2023-03-04:
Wird die b Variante dann bei mir auch laufen.

RomanWujnovic, 2023-03-04:
Die Aktualisierung über das webinterface geht bei mir nämlich noch nicht. Ich kann die Version 2.2.11 a und b erst morgen Abend testen.

softwarecrash, 2023-03-04:
@LindemannKammerer ja, das hatte sich vor langer zeit mal ein rechenfehler eingeschlichen der bis jetzt unbemerkt blieb, das lag an den 12 zellen.

@RomanWujnovic die versionen sollten bei dir auch laufen.

kommt ins nächste bugfix release rein, bis dahin funktioniert es bei euch ja :)

RomanWujnovic, 2023-03-06:
Hallo Softwarecrash,
sorry das ich mich erst heute melde. 
Also bei mir sieht es auch soweit ganz gut aus. Die automatisch aktualiesierung im Webinterface funktioniert jetzt tadellos.
MEGA DANK schon mal dafür.
Allerdings fehlt mir im iobroker die Spannung von der ersten Zelle. Alle anderen sind da und aktualisierung sich auch brav. 
Und ich habe im log unzählige Meldungen mit denen ich leider nichts anfangen kann.

2023-03-06 18:44:08.758 - warn: mqtt.0 (3982263) Used invalid characters: mqtt.0.BMS01.Pack_Cells_0&BMS01.Pack changed to mqtt.0.BMS01.Pack_Cells_0&_BMS01.Pack

2023-03-06 18:44:32.760 - warn: mqtt.0 (3982263) Used invalid characters: mqtt.0.BMS01.Pack_Cells_Vo0&BMS01.Pa changed to mqtt.0.BMS01.Pack_Cells_Vo0&_BMS01.Pa

2023-03-06 18:45:14.760 - warn: mqtt.0 (3982263) Used invalid characters: mqtt.0.BMS01.Pack_Cells_Vo0&BMS01.Pa changed to mqtt.0.BMS01.Pack_Cells_Vo0&_BMS01.Pa

2023-03-06 18:45:57.015 - warn: mqtt.0 (3982263) Used invalid characters: mqtt.0.BMS01.Pack_Cells_Volt0&BMS01 changed to mqtt.0.BMS01.Pack_Cells_Volt0&_BMS01

2023-03-06 18:46:57.224 - warn: mqtt.0 (3982263) Used invalid characters: mqtt.0.BMS01.Pack_Cells_Vo0&BMS01.Pa changed to mqtt.0.BMS01.Pack_Cells_Vo0&_BMS01.Pa

2023-03-06 18:50:15.372 - warn: mqtt.0 (3982263) Used invalid characters: mqtt.0.BMS01.Pa0&BMS01.Pack_Cells_Ba changed to mqtt.0.BMS01.Pa0&_BMS01.Pack_Cells_Ba

2023-03-06 18:50:39.492 - warn: mqtt.0 (3982263) Used invalid characters: mqtt.0.BMS01.Pack0'_BMS01.Pack_Cells_ changed to mqtt.0.BMS01.Pack0__BMS01.Pack_Cells_

2023-03-06 18:50:39.544 - warn: mqtt.0 (3982263) Used invalid characters: mqtt.0.BMS01.Pack0'_BMS01.Pack_Cells_ changed to mqtt.0.BMS01.Pack0__BMS01.Pack_Cells_

2023-03-06 18:50:39.627 - warn: mqtt.0 (3982263) Used invalid characters: mqtt.0.BMS01.Pack0'_BMS01 changed to mqtt.0.BMS01.Pack0__BMS01

2023-03-06 18:51:09.369 - warn: mqtt.0 (3982263) Used invalid characters: mqtt.0.BMS01.Pa0&BMS01.Pack_Cells_Vo changed to mqtt.0.BMS01.Pa0&_BMS01.Pack_Cells_Vo

2023-03-06 18:51:45.371 - warn: mqtt.0 (3982263) Used invalid characters: mqtt.0.BMS01.Pack_Cells_Vo0&BMS01.Pa changed to mqtt.0.BMS01.Pack_Cells_Vo0&_BMS01.Pa

2023-03-06 18:52:51.442 - warn: mqtt.0 (3982263) Used invalid characters: mqtt.0.BM0BMS01.Pack_Cycles2 changed to mqtt.0.BM0_BMS01.Pack_Cycles2

und das geht immer so weiter. vielleicht kannst du damit was anfangen oder mich aufklären.

Gruss Roman

softwarecrash, 2023-03-06:
Grüße,
Das phänomen kenn ich ja, allerdings taucht das immer nur Mal auf, ich kann nicht sagen woher das kommt. Aber das steht noch auf der To-Do Liste.
Die erste Zelle fehlt? Auch mit Nummer oder geht die Liste direkt mit 2 los?

softwarecrash, 2023-03-06:
ah okay, das ist in der c schon behoben.
schau mal ob es damit nicht mehr zu diesen seltsamen zeichen kommt
[Daly-BMS-to-MQTT_d1_mini_2.2.11-c_debug.zip](https://github.com/softwarecrash/DALY-BMS-to-MQTT/files/10902001/Daly-BMS-to-MQTT_d1_mini_2.2.11-c_debug.zip)

RomanWujnovic, 2023-03-06:
Neue Version ist drauf. Ich mache gleich ein picture.

RomanWujnovic, 2023-03-06:
[image]
[image]

so sieht es jetzt aus.

softwarecrash, 2023-03-06:
lösch mal den ganzen bms baum und starte den mqtt server neu

RomanWujnovic, 2023-03-06:
Hatte ich nach dem Update gemacht. Taucht alles nacheinander wieder auf. Ich mach es aber nochmal.

RomanWujnovic, 2023-03-06:
[image]

so schaut es nach dem Neustart. Erstmal aus.

RomanWujnovic, 2023-03-06:
[image]

So dann ein paar Minuten später

RomanWujnovic, 2023-03-06:
[image]

Mqqt Einstellung

softwarecrash, 2023-03-06:
eigene states beim verbinden publizieren und states bei subscribe publizieren muss schon mal raus. aber das dürfte nicht den fehler beheben.
ruf mal bitte ip/lifejson auf und kopier den inhalt hier rein

RomanWujnovic, 2023-03-06:
Bei ip/livejson bräuchte ich bitte deine Unterstützung wo oder wie mache ich das. Sorry

softwarecrash, 2023-03-06:
> Bei ip/livejson bräuchte ich bitte deine Unterstützung wo oder wie mache ich das. Sorry

du rufst ja die weboberfläche vom esp auf, einfach hinter die ip noch ein /livejson schreiben, dann werden dir nen haufen daten angezeigt

RomanWujnovic, 2023-03-06:
{"Device":{"Name":"DALY-BMS-to-MQTT","IP":"192.168.10.69","ESP_VCC":3.017,"Relais_Active":false,"Relais_Manual":false,"Free_Heap":28792,"json_memory_usage":1537,"json_capacity":2048,"runtime":2091,"ws_clients":1,"HEAP_Fragmentation":7,"free_blocksize":26952},"Pack":{"Voltage":39.5,"Current":-9.6,"Power":-379.2,"SOC":81.2,"Remaining_mAh":72990,"Cycles":21,"BMS_Temp":15,"Cell_Temp":15,"High_CellNr":9,"High_CellV":3.308,"Low_CellNr":5,"Low_CellV":3.285,"Cell_Diff":23,"DischargeFET":true,"ChargeFET":true,"Status":"Discharge","Cells":12,"Heartbeat":3,"Balance_Active":false},"CellV":{"CellV_1":0,"Balance_1":false,"CellV_2":3.296,"Balance_2":false,"CellV_3":3.289,"Balance_3":false,"CellV_4":3.295,"Balance_4":false,"CellV_5":3.294,"Balance_5":false,"CellV_6":3.285,"Balance_6":false,"CellV_7":3.288,"Balance_7":false,"CellV_8":3.304,"Balance_8":false,"CellV_9":3.312,"Balance_9":false,"CellV_10":3.271,"Balance_10":false,"CellV_11":3.313,"Balance_11":false,"CellV_12":3.293,"Balance_12":false},"CellTemp":{"Cell_Temp_1":15}}

softwarecrash, 2023-03-06:
danke.
die wlan verbindung ist stabil?
ansonsten gehen mir aktuell die ideen aus.
da kann ich nur mal schauen ob ich irgendwas finde was den fehler verursachen könnte

RomanWujnovic, 2023-03-06:
Beziehst du das auf die viele warn Meldungen oder die vielen Ordnern oder auf die fehlende Spannung der zelle 1. 
An sich schon. Daneben hängt ein weiterer esp der meinen wechselrichter steuert über den shelly em3. Das läuft alles.

softwarecrash, 2023-03-06:
> Beziehst du das auf die viele warn Meldungen oder die vielen Ordnern oder auf die fehlende Spannung der zelle 1. An sich schon. Daneben hängt ein weiterer esp der meinen wechselrichter steuert über den shelly em3. Das läuft alles.

das bezieht sich auf die vielen ordner und deren namen, die warnungen hängen mit den ordnern zusammen.
mit der ersten zelle das ist gefixxt mit der neuen version die ich verlinkt habe. da sollte die zelle wieder daten haben.

RomanWujnovic, 2023-03-06:
ich habe gerade festgestellt das dass update glaube ich garnicht ausgeführt wird oder installiert wird.
der esp verhält sich nachdem man die Version ausgewählt hat und update drückt anders wie bisher.
und wenn der esp wieder erreichbar ist. steht im webinterface immer noch die b version

RomanWujnovic, 2023-03-06:
bisher kam beim update die meldung firmware update in arbeit oder so. aber jetzt kommt recht schnell die meldung.
fehler verbindung unterbrochen

RomanWujnovic, 2023-03-06:
hobe es jetzt viermal versucht ich bekomme die c version nicht auf den esp.

softwarecrash, 2023-03-06:
Dann lade sie Mal per USB drauf.
Evt schläft dein BMS gerade, dann kann es mir dem Update auch Mal nicht klappen

RomanWujnovic, 2023-03-06:
Das werde ich dann morgen versuchen und dir berichten.

softwarecrash, 2023-03-08:
probier mal die aktuelle version ohne debug. vermutlich kommen die zeichenfehler bzw warnungen von dem debug da wird ziehmlich datenschrott produziert

RomanWujnovic, 2023-03-08:
hi, ist die aktuelle die C version.

RomanWujnovic, 2023-03-08:
ich habe die seit gestern am laufen. ja ich habe noch komische ordner mit komischen buchstaben und zeichen.
die Warnungen unf Fehlermeldungen wurden aber weniger

softwarecrash, 2023-03-08:
dann lad dir mal das aktuelle release und flash das ohne debug version.

RomanWujnovic, 2023-03-08:
erledigt habe soeben die aktuelle drauf gemacht und alle unnötige ordner gelöscht

softwarecrash, 2023-03-10:
also, die komischen ordner und datenpunkte tauchen auschschlieslich bei iobroker und dem mqtt server auf. 
es scheint fast so als ob das nicht unbedingt ein problem vom esp oder der firmware ist. aber ich bleib weiter dran.
läuft es bei dir bis auf die seltsamheiten erstmal?

RomanWujnovic, 2023-03-10:
Servus, also die Fehlermeldung hält sich momentan im Rahmen. 
die Ordners Zahl ist halt ziemlich viel. 
[image]

ich bin einfach nur froh das es jetzt mal so soweit läuft. 
Dein Projekt ist einfach mega und ich bin froh das ich die wichtigsten Werte im IoBroker habe. Um andere Dinge damit auszulösen.

softwarecrash, 2023-03-10:
probier mal diese bitte, damit sind bei mir diese nervigen sonderzeichen probleme weg
[Daly-BMS-to-MQTT_d1_mini_2.2.12-c_debug.zip](https://github.com/softwarecrash/DALY-BMS-to-MQTT/files/10945173/Daly-BMS-to-MQTT_d1_mini_2.2.12-c_debug.zip)

closed 2023-03-14 by softwarecrash

RomanWujnovic, 2023-03-14:
Alles klar Softwarecrash ich werde es die Tage mal testen und berichten.
Auf jeden Fall schon mal dir einen Mega Dank für deine bemühungen und anstrengungen.
DANKE!