softwarecrash/Daly2MQTT #69: Keine Daten per MQTT

Steef-the-Beef, 2023-03-09:
Moin, 

erstmal vielen Dank für die Mühe in diesem Projekt!

Ich kann Daten aus dem BMS auslesen und diese werden auch auf dem Web Interface angezeigt.
Leider bekomme ich aber keine Daten in Node Red.

[image]
[image]
[image]

Was mache ich falsch? Bzw muss ich noch irgendwo das Senden der Daten per MQTT aktivieren?

Viele Grüße
Stefano

softwarecrash, 2023-03-09:
ich hol mal unseren node red spezialisten @all-solutions mit ins boot

Steef-the-Beef, 2023-03-09:
Der Lattenzaun fehlte :-D

Lattenzaun = #

Vielen Dank für die schnelle Hilfe!

closed 2023-03-09 by Steef-the-Beef