softwarecrash/Daly2MQTT #70: Webpage loading very long or not fully

crazy-ahmet, 2023-03-10:
After typing in the IP Adress of the device it´s often keep loading and loading. Sometimes it shows only a few parts of the site. I don´t know why this happens. Maybe the ESP is busy? During all this the MQTT values are refreshed frequently so the device itself is working.

softwarecrash, 2023-03-10:
the mqtt update interval is not a problem, testet with 3 seconds without issue.
you can try the 2.2.11 it wil run a little bit better, when that not help. pleasee try the atached firmware
[Daly-BMS-to-MQTT_d1_mini_2.2.12-b_debug.zip](https://github.com/softwarecrash/DALY-BMS-to-MQTT/files/10940871/Daly-BMS-to-MQTT_d1_mini_2.2.12-b_debug.zip)

softwarecrash, 2023-03-11:
working now?

crazy-ahmet, 2023-03-12:
I think it works now.

closed 2023-03-12 by softwarecrash