softwarecrash/Daly2MQTT #76: Show the status on the website and publish it via MQTT (if possible)

OliverWieland, 2023-03-14:
**Is your feature request related to a problem? Please describe.**
My ESP is placed on a site i cannot see the LED. So it woul'd be nice if I could see the status of the system on the webpage.
Also it would be very helpful if there would be a MQTT message about the connection status to the BMS.

softwarecrash, 2023-03-14:
Heartbeat is running 0-255 cycle so it is different from the last the bms is connected
second state is chargedischargestate it can have the "offline" state then the bms is sleeping or not conected

OliverWieland, 2023-03-14:
Okay, so I can detect if BMS is offline via MQTT right now, thank you for the info.
But what if MQTT is not alive? I know that there is a LWT, but i would be nice to see the MQTT connection status on the website as well

softwarecrash, 2023-03-14:
when mqtt not connected, the alive state goes to false, so you can script a routine to inform that it will be offline.

we have discuseed for a long time to add infos in the main page about that, and the result is that we dont need it.

a little future is the led, so when you dont see it, desolder it and add a cable