softwarecrash/Daly2MQTT #80: Lange Ladezeiten V2.2.13 auf ESP01

crazy-ahmet, 2023-03-19:
Wurde bei der Version für den ESP01 schon nachgebessert? Hatte das Problem ja vorher schon beim D1 Mini.
Die Webseite dauert lange zu laden, die einzelnen Elemente kommen teils mehrere Sekunden verzögert und der ganze Seitenaufbau dauert auch mal 10s und mehr.
So ziemlich bei allem was man anklickt lädt es 5-10s. Manchmal baut sich die Seite gar nicht vollständig auf und es bleibt hängen.

softwarecrash, 2023-03-19:
bitte bearbeite dein issue nochmal und füge folgende infromationen an.
- BMS | Zellenanzahl | aufbau
- ESP genaues model | Speicher | verkabelung | Stromvbersorgung
- Firmware Version
- Spiele die Debug version auf und kopiere den inhalt von ip/livejson als code tag hier her

crazy-ahmet, 2023-03-19:
Habe ein DALY 16S LIFEPO4 150A BMS
ESP01S (der kleine schwarze), Stromversorgung über BMS (Stecker mit 20cm Kabel, habe noch nen 100uF Elko parallel)
Firmware wie gesagt die aktuelle V2.2.13
Blinkcode gibts beim ESP01S ja nicht oder? (er flackert im MQTT refresh Intervall)

Vielleicht bekommt er doch zu wenig Spannung, messe 2,84V, allerdings nach jedem senden geht es kurz auf über 3,2V hoch. 

Trotzdem tut er nach Upload der Debug Version komischerweise laufen und rennt schnell. Erstmal schlug das Update fehl, trotzdem rennt er ab dann. Dann nochmal eingeleitet und rennt immer noch, trotz der 2,84V die ich nochmal nachgemessen habe. Bei früheren Version viel mir so eine Unterspannungswarnung im Webinterface mal auf. Die hab ich hier nie bekommen.

{"Device":{"Name":"ESP BMS","IP":"192.168.0.57","ESP_VCC":2.952,"Wifi_RSSI":-69,"Relais_Active":false,"Relais_Manual":false,"CPU_Frequency":160,"Real_Flash_Size":1048576,"Flash_Size":1048576,"Sketch_Size":490720,"Free_Sketch_Space":536576,"Free_Heap":31376,"HEAP_Fragmentation":6,"Free_BlockSize":29656,"json_memory_usage":1912,"json_capacity":2048,"runtime":17,"ws_clients":1},"Pack":{"Voltage":53.6,"Current":0,"Power":0,"SOC":99.5,"Remaining_mAh":238800,"Cycles":0,"BMS_Temp":19,"Cell_Temp":19,"High_CellNr":4,"High_CellV":3.358,"Low_CellNr":16,"Low_CellV":3.334,"Cell_Diff":24,"DischargeFET":true,"ChargeFET":true,"Status":"Stationary","Cells":16,"Heartbeat":15,"Balance_Active":false},"CellV":{"CellV_1":3.35,"Balance_1":false,"CellV_2":3.357,"Balance_2":false,"CellV_3":3.356,"Balance_3":false,"CellV_4":3.358,"Balance_4":false,"CellV_5":3.356,"Balance_5":false,"CellV_6":3.352,"Balance_6":false,"CellV_7":3.348,"Balance_7":false,"CellV_8":3.356,"Balance_8":false,"CellV_9":3.35,"Balance_9":false,"CellV_10":3.354,"Balance_10":false,"CellV_11":3.351,"Balance_11":false,"CellV_12":3.356,"Balance_12":false,"CellV_13":3.354,"Balance_13":false,"CellV_14":3.346,"Balance_14":false,"CellV_15":3.34,"Balance_15":false,"CellV_16":3.334,"Balance_16":false},"CellTemp":{"Cell_Temp_1":19}}

softwarecrash, 2023-03-19:
dann rufe das livejson nochmal ab wenn der esp hängt oder langsam ist. im gut laufenden zustand kann man da nichts erkennen.

crazy-ahmet, 2023-03-19:
Irgendwie kann ich den Fehler mit der Debug Version nicht reproduzieren, wenn ich die ohne aufspiele läuft es teils, teils zickt er aber auch. Habe die Debug nun 3x neu gebootet und auch das Teil 2x an und abgesteckt. Fehler taucht hier aber nicht auf, zumindest bisher.

softwarecrash, 2023-03-19:
das sieht eher nach einem restart aus oder der esp bekommt keine validen daten vom bms.

crazy-ahmet, 2023-03-19:
Ja der Hänger war nach eine Reboot, vermutlich gabs dann noch keine Daten vom BMS. Werde weiter beobachten ob der Fehler wieder auftritt.

crazy-ahmet, 2023-03-21:
Mit der Debug Version nicht aber habe 2x zurück upgedated auf die Version ohne Debug und da war der Fehler da. Werde es morgen nochmal an einem anderen BMS probieren.

crazy-ahmet, 2023-03-21:
Also es hat schon deutlich gehakelt wenn die Seite sich mit mehreren Sekunden Pause blockweise aufbaut. 
Vielleicht liegts wirklich an den 2,84V die das BMS nur rausgibt aber warum das bei der Debug Version unerheblich ist?
Der ESP war ein billiges Teil aus Ali hier: https://www.aliexpress.com/item/32739060009.html

Habe nun nochmal die normale 2.2.13 drauf gemacht. 
Update zurück hat funktioniert. Beim ersten Laden der Seite kamen aber nur ein paar Zeilenblöcke ohne Überschrift.
Habe ihn dann mal gelassen. Es scheint als wenn er die Verbindung zum BMS aufgebaut hat läuft es wieder flüssig.

Kann man hier keine PN schicken oder gibts nen Discord channel oder sowas?

crazy-ahmet, 2023-03-21:
Habe den EPS01S wie oben schon geschrieben. Hab das Teil jetzt leider schon abgeklemmt und eingepackt für ein anderes BMS bei einem Kollegen morgen.

crazy-ahmet, 2023-03-23:
Also ich habe das ganze gestern ewig probiert. Natürlich war die Wifi SSID und Passwort beim Kollegen anders. Es hat Ewigkeiten nicht funktioniert sich mit dem BMS per Wifi zu verbinden. Es hat zwar den DALY AP aufgemacht aber immer wenn ich ihn mit dem Smartphone angeklickt habe um mich per Wifi zu verbinden ist er wieder verschwunden, evt. hat er kurz versucht zu verbinden, ich weiß nicht, der AP war nach 1s wieder weg.

Habe es dann ewig probiert und mit dem Laptop auch. Ich dachte das BMS hat zu wenig Spannung, habe den ESP dann nur mit Strom vom USB des Laptops aus dem Programmieradapter versorgt und es irgendwann geschafft das neue WLAN einzutragen. Im Router wurde die IP des Teils aber nicht angezeigt, habe dann in der Hauptfritzbox das Device gefunden, im Router mit dem es sich verbunden hat war aber nix.

Naja es hat dann schließlich funktioniert und läuft auch am BMS. Lt. ESP hat das Teil 3,3V Spannung (ist ein 150A Daly BMS was zusätzlich noch RS485/CAN Ausgang hat). Bei meinem anderen BMS (das gleiche 150A Daly aber ohne RS485/CAN) werden lt. ESP nur 3V angezeigt und 2,84V habe ich gemessen. Merkwürdig, dass die Ausgangsspannungen wohl so unterschiedlich sein können.

Manchmal hat das Teil auch tatsächlich 3 oder 4x geblinkt wie die Version des D1 Mini. Wenn er mit dem BMS kommuniziert blinkt er aber nicht mehr, nur flackern bei der WLAN Übertragung. Die Funktion der Blinkcodes ist schon schön, dachte die geht beim ESP01 gar nicht aber scheinbar hat es wohl 4x geblinkt als es noch nicht im WLAN war.

Hier noch das Livejson
{"Device":{"Name":"BMS","IP":"192.168.0.39","ESP_VCC":3.316,"Wifi_RSSI":-70,"Relais_Active":false,"Relais_Manual":false},"Pack":{"Voltage":51.3,"Current":12.9,"Power":661.77,"SOC":13.3,"Remaining_mAh":31920,"Cycles":7,"BMS_Temp":19,"Cell_Temp":19,"High_CellNr":9,"High_CellV":3.245,"Low_CellNr":5,"Low_CellV":3.169,"Cell_Diff":76,"DischargeFET":true,"ChargeFET":true,"Status":"Charge","Cells":16,"Heartbeat":111,"Balance_Active":false},"CellV":{"CellV_1":3.173,"Balance_1":false,"CellV_2":3.237,"Balance_2":false,"CellV_3":3.241,"Balance_3":false,"CellV_4":3.219,"Balance_4":false,"CellV_5":3.169,"Balance_5":false,"CellV_6":3.171,"Balance_6":false,"CellV_7":3.239,"Balance_7":false,"CellV_8":3.187,"Balance_8":false,"CellV_9":3.245,"Balance_9":false,"CellV_10":3.243,"Balance_10":false,"CellV_11":3.175,"Balance_11":false,"CellV_12":3.195,"Balance_12":false,"CellV_13":3.229,"Balance_13":false,"CellV_14":3.195,"Balance_14":false,"CellV_15":3.225,"Balance_15":false,"CellV_16":3.183,"Balance_16":false},"CellTemp":{"Cell_Temp_1":19,"Cell_Temp_2":19}}

Wundere mich warum das WLAN mit -70dB so schlecht ist, der Router ist 3m Sichtlinie davon entfernt, naja vielleicht war ich davor gestanden, es läuft ja jetzt. 

Hatte dann aber keine Lust mehr auszuprobieren ob es wieder klemmt wenn man das Teil frisch ansteckt oder Firmware Versionen ändert.

softwarecrash, 2023-03-23:
> Also ich habe das ganze gestern ewig probiert. Natürlich war die Wifi SSID und Passwort beim Kollegen anders. Es hat Ewigkeiten nicht funktioniert sich mit dem BMS per Wifi zu verbinden. Es hat zwar [...]

die software bzw librarys haben mit solchen aktionnen manchmal probleme, dafür ist der reset da, wenn man sowas umzieht am besten vorher den esp zurücksetzen.