softwarecrash/Daly2MQTT #82: DALY BMS to MQTT 2.2.13 Keine Daten mehr von BMS

AudipowerS4, 2023-03-19:
Hallo, ich habe ein Problem das nach einigen Stunden ich keine Daten mehr bekomme über ESP als ob der ESP keine Daten mehr kriegt von BMS...

softwarecrash, 2023-03-19:
Auf welche Zeit ist der sleeptimer des BMS gestellt?

AudipowerS4, 2023-03-19:
BMS steht auf Standard wert 3000s . Aber ich hab dahinter noch Connector Board mit Ausgängen was man separat kaufen kann für den Daly. Dieses Board läuft von BMS TX und RX leitung + noch die 12V Versorung zu Connector Board. Was ich nicht verstehe das dieses Board läuft und die 12V Versorungsleitung ist von BMS nicht unterbrochen aber der ESP steht still und sendet mir keine Daten zu den Iobroker.

AudipowerS4, 2023-03-19:
Nach den Trennen von den BMS Stecker  und wieder reinstecken läuft wieder alles ... Das sieht nach Sleep Problem aus...

softwarecrash, 2023-03-19:
ist das webinterface noch erreichbar wenn er keine daten mehr sendet, und steht dort bei status offline?

AudipowerS4, 2023-03-19:
Ja Webinterface wahr errechbar aber die Daten wahren eingefroren. Ich werde das noch weiter Beobachten...

softwarecrash, 2023-03-19:
wenn er keine daten bekommt sendet er auch keine neuen. interessant ist lediglich der status ob da offline stand, denn dann reagiert das bms nicht mehr.

AudipowerS4, 2023-03-20:
Fehler bis jetzt nicht aufgetaucht...

closed 2023-03-20 by AudipowerS4