softwarecrash/Daly2MQTT #83: /livejson schließt die TCP connection nicht.

gzi01, 2023-03-20:
**Describe the bug**
Bei auslesen des JSON via PHP entstehen Verzögerungen von 1 Minute bzw. bricht das Script mit timeout ab. 

**To Reproduce**
Steps to reproduce the behavior in PHP :

       php -r ' $jsondata = file_get_contents("http://10.0.0.25/livejson"); echo $jsondata; '

(takes 60 seconds)

Work around (responds promptly):

       php -r ' $ch = curl_init(); curl_setopt($ch, CURLOPT_URL, "http://10.0.0.25/livejson"); curl_setopt($ch, CURLOPT_TIMEOUT, 5 ); curl_setopt($ch, CURLOPT_RETURNTRANSFER, 1); $jsondata = curl_exec($ch); echo $jsondata;'

Ich denke, es ist das Problem, das in https://stackoverflow.com/questions/3629504/php-file-get-contents-very-slow-when-using-full-url beschriebn ist. Zitat:" that the remote web server DIDN'T CLOSE THE TCP CONNECTION"

**Expected behavior**
auch die erste Variante sollte rasch die Antwort liefern.

**Screenshots**
n/a

**Desktop (please complete the following information):**
Raspi commandline

**Additional context**
DALY BMS to MQTT 2.2.11

softwarecrash, 2023-03-20:
thanks,
the primary problem comes from the webserver lib, so i cant change this. but open a pull request to fix the problem, now waiting for response

softwarecrash, 2023-03-20:
i have made a temp fix with own lib.
you can download the main source and compile itself.
the test with the php command will work.

softwarecrash, 2023-03-22:
so sorry to say that, but we cant fix this finaly. its a bug in the ESPAsyncWebserver lib

closed 2023-03-22 by softwarecrash