softwarecrash/Daly2MQTT #86: Editable values over MQTT

mobilon0815, 2023-03-22:
It doesn't matter which DataPoint I use, whether true, false or without a payload, /Device_Control or just / is also sufficient. 
As soon as I publish something to the topic (default BMS01), the **"ChargeFET" goes to false.**

checked with:
/Device_Control/Pack_DischargeFET
/Device_Control/Pack_DischargeFET:true
/Device_Control/Pack_DischargeFET:false
/Device_Control/Pack_ChargeFET
/Device_Control/Pack_ChargeFET:true
/Device_Control/Pack_ChargeFET:false
/Device_Control/Relay:true
/Device_Control
/

There is no way to change this again with any publish. 

****used Version: V2.2.11**
**used tool: MQTT-Explorer on WIN11**
**reproducible****

double checked with:
 **MyMQTT V 2.2.3 on Android same behavior - reproducible**

before clicking "Publish"

[image]

after clicking "Publish"

[image]

softwarecrash, 2023-03-22:
thanks,
didt you use the json style or the standart mqtt style? (in settings)
please update first to the latest firmware. (not solve the problem, but some changes)

mobilon0815, 2023-03-22:
before:

[image]

after:

[image]

softwarecrash, 2023-03-22:
> before:
> 
> [image]
> 
> after:
> 
> [image]

double check your topic string, you have BMS01//Dev...... but you need BMS01/Dev.....

mobilon0815, 2023-03-22:
i use json style

[image]

I will update now to 2.2.13 an check again...

mobilon0815, 2023-03-22:
checked again  with BMS**01/D**evice_Control/Pack_DischargeFET  -> same

mobilon0815, 2023-03-22:
updated to 2.2.13 

[image]

**nothing happens**

[image]

nothing happens ???? "ChargeFET" is still true but "DischargeFET" too

softwarecrash, 2023-03-22:
in the style you try to manipulate the bms it will not work.
the device controll path only active in classic mqtt.
i will rework this, so please wait the next days for a fix

mobilon0815, 2023-03-22:
downgraded to 2.2.11 and checked like in the brandnew wiki...(JSON...)

[image]

ChargeFET **AND** DischargeFET have gone from true to false.

{"Pack": {"DischargeFET": true}} brings DischargeFET back to true (ChargeFET still false)

If you publish without the outer brackets, only the ChargeFET changes (once to false)  "Pack": {"DischargeFET": true}

softwarecrash, 2023-03-22:
let me repeat

wait for a fix

softwarecrash, 2023-03-22:
you can switch off the json in settings to use the device_control path

softwarecrash, 2023-03-22:
now with .14 fixed, use the topic paths descripted in the wiki

closed 2023-03-22 by softwarecrash