softwarecrash/Daly2MQTT #93: Keine Daten vom BMS

1990marco1990, 2023-04-15:
Hallo,

Ich nutze ein NODEMCU und habe:

TX----->RX
RX----->TX
GND---->GND    

verbunden.

DALY BMS läuft und kann mich auch auf den Webserver anmelden. Leider kommen keine Daten vom BMS. Sobald ich das Display wieder anschließe zeigt das Display alle Daten an.

Habe ich etwas vergessen?
Vielleicht könnt ihr mir ja weiterhelfen.

softwarecrash, 2023-04-15:
Schau Mal ins Wiki da steht wie man es richtig verkabelt.

1990marco1990, 2023-04-16:
Ja nur leider ist da das NODEMCU nicht extra aufgeführt oder ich finde es nicht.

derLoosi, 2023-04-16:
wir unterstützen nur ESP8266 und ESP01

softwarecrash, 2023-04-16:
Der nodemcu ist nicht mit aufgeführt weil es nicht nötig ist, der hat genauso Beschriftungen mit D6 und D7, die sind identisch zum wemos.

softwarecrash, 2023-04-19:
no response

closed 2023-04-19 by softwarecrash