softwarecrash/Daly2MQTT #94: FET-Umschaltung mit MQTT

soso005, 2023-04-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Vielen Dank für das tolle Tool. Das macht viele Anwendungsfälle möglich.

In der Version 2.2.16 habe ich bei der FET-Umschaltung über MQTT folgendes Verhalten:

Ausgangslage: beide FET false . Das Web-UI ist nicht in einem Browser geöffnet.

Pack_ChargeFET wird allein auf true geschaltet: Das DALY-BMS-to-MQTT wiederholt die Nachricht beliebig oft, schaltet Pack_ChargeFET aber nicht um. Dies kann auch nicht durch BMS01/Device_Control/Pack_ChargeFET = false unterbrochen werden. Erst durch BMS01/Device_Control/Pack_DischargeFET =true werden die Nachrichten unterbrochen und beide FET schalten auf true.

Bei anschließender Deaktivierung von Pack_ChargeFET schaltet Pack_DishargeFET automatisch mit auf false.  Bei erneuter Aktivierung von Pack_ChargeFET wird dann Pack_DishargeFET wieder automatisch true.

Ist das so gewünscht? Mir fehlt, nur Pack_ChargeFET unabhängig schalten zu können, wenn Pack_DishargeFET false ist.

### Screenshots / Fotos

[image]

### Version

2.2.14 and above

### Relevant livejson output

```json
{"Device":{"Name":"Solar_BMSAkku01","IP":"192.168.178.119","ESP_VCC":3.002,"Wifi_RSSI":-65,"Relais_Active":false,"Relais_Manual":false,"sw_version":"2.2.16"},"Pack":{"Voltage":26.3,"Current":0,"Power":0,"SOC":45.7,"Remaining_mAh":47985,"Cycles":2,"BMS_Temp":10,"Cell_Temp":10,"High_CellNr":7,"High_CellV":3.29,"Low_CellNr":1,"Low_CellV":3.286,"Cell_Diff":4,"DischargeFET":false,"ChargeFET":false,"Status":"Stationary","Cells":8,"Heartbeat":206,"Balance_Active":false},"CellV":{"CellV_1":3.286,"Balance_1":false,"CellV_2":3.29,"Balance_2":false,"CellV_3":3.29,"Balance_3":false,"CellV_4":3.29,"Balance_4":false,"CellV_5":3.289,"Balance_5":false,"CellV_6":3.289,"Balance_6":false,"CellV_7":3.29,"Balance_7":false,"CellV_8":3.288,"Balance_8":false},"CellTemp":{"Cell_Temp_1":10}}
```

### What browsers are you seeing the problem on?

Firefox

softwarecrash, 2023-04-18:
hallo,
ich hab das grad eben mal durchgespielt. und kann es leider nicht wirklich nachvollziehen.
wenn einer der FETs automatisch mitschaltet dann kommt das vom BMS selbst, da hab ich leider keinen einfluss drauf, seites der software gibt es dahingehend keine logik oder dergleichen, die befehle werden durchgereicht, und andersrum der status abgefragt.

das der esp auf device control etwas schreibt ist soweit erstmal richtig, er schreibt aber eigentlich nur einmalig eine antwort zur bestätigung des wertes. ggf kann man ihn durcheinander bringen wenn man zu schnell zu viele befehle sendet (die müssen abgearbeitet werden können).
was hast du den für einen mqtt server dahinter? falls es der iobroker "eigene" ist der macht in den standart einstellungen probleme da er alles was reinkommt direkt wieder publisht und eine schleife erzeugt, ggf ist das bei dir der fall, dann wird der esp mit schaltbefehlen überschwemmt.

soso005, 2023-04-19:
vielen Dank, als mqtt-Server läuft bei mir in einem Docker-Container 

Auch im Falle der problemlosen Umschaltungen vom DischargeFET gibt es mindestens fünf Wiederholungen der Nachrichten, ggf. bis das BMS tatsächlich umschaltet.
Wenn die Wiederholungen vom mqtt-Server kommen, ist mir noch nicht klar, wie er erkennt, wer der Empfänger (hier DALY-BMS-to-MQTT) ist, um dann die Wiederholung vom Empfänger als Quittung zu interpretieren, um anschließend die eigenen Wiederholungen zu beenden. Da kenne ich mich so gar nicht aus.

softwarecrash, 2023-04-19:
der empfänger ist einfach das jeweilige topic, und umgekehrt was er subscribed hat.
was für ein mqtt server läuft den in dem docker oder welches image ist es?
ggf kann @derLoosi da noch was beitragen, er ist eher der mqtt spezi

derLoosi, 2023-04-19:
Da müsste ich wenn ich Zeit hab mal rumbasteln und schauen wie du das im Code gelöst hast dass er sich nicht selbst in einer Endlosschleife auf das subscribte topic überschreibt :) Bisher waren nur Probleme mit dem internen ioBroker Adapter bekannt weil der nen paar komische Sachen macht, mit nem externen Server hatten wir da aber bisher keine. Hast du eine Möglichkeit auf die Logdateien im Docker Container zuzugreifen und mal den Abschnitt hier zu Posten wo er die komischen Sachen macht?
Vielleicht kann man dann schonmal zumindest eingrenzen wer da was in welches topic schreibt und wo wir suchen müssen.

soso005, 2023-04-19:
Sorry, den mqtt-Server hage ich vergessen:
mosquitto version 2.0.14

Den logfile kann ich öffnen. Hat fast 78.000 Zeilen und 5,8MB.
Wie finde ich den relevanten Abschnitt?

Der Anfang sieht wie folt aus:
1652909351: mosquitto version 2.0.14 starting
1652909351: Config loaded from /mosquitto/config/mosquitto.conf.
1652909351: Opening ipv4 listen socket on port 1883.
1652909351: mosquitto version 2.0.14 running
1652909570: New connection from 192.168.178.11:44728 on port 1883.
1652909570: New client connected from 192.168.178.11:44728 as 1on3qY9TSRqXHryyLJwnk8 (p1, c1, k60, u'user').
1652909570: Client 1on3qY9TSRqXHryyLJwnk8 disconnected.
1652909570: New connection from 192.168.178.11:40002 on port 1883.
1652909570: New client connected from 192.168.178.11:40002 as 7sgUl5Epv4wbmXHjNrbvkZ (p2, c1, k60, u'user').

Das Ende:
1681904097: New connection from 172.17.0.1:60826 on port 1883.
1681904117: New client connected from 172.17.0.1:60826 as DVES_54828F (p2, c1, k30).
1681904117: Client DVES_54828F closed its connection.
1681904121: New connection from 172.17.0.1:60974 on port 1883.
1681904121: New client connected from 172.17.0.1:60974 as DVES_54828F (p2, c1, k30).
1681905329: Saving in-memory database to /mosquitto/data//mosquitto.db.
1681907130: Saving in-memory database to /mosquitto/data//mosquitto.db.

derLoosi, 2023-04-19:
Ich glaube wir konnten den Fehler finden, wir sind dran :)
Du meinst dass er den Datenpunkt unter Device_Control sehr oft innerhalb kurzer Zeit beschreibt oder?

soso005, 2023-04-19:
genau. Die Wiederholungszeit ist sehr kurz, 0,05-00,12 Sekunden.

Wenn bei der Ausgangslage, dass beide FET auf flase stehen
- nur auf /Device_Control/Pack_ChargeFET mit true geschrieben wird,
  schaltet der Pack_ChargeFET gar nicht um und es wird dauerhaft geschrieben,
- nur auf  /Device_Control/Pack_DischargeFET mit true geschrieben wird,
 schaltet DischargeFET nach 5-10 Wiederholungen um bzw. nach dem Umschalten endet das Schreiben.

D.h. das eigentliche Problem ist womöglich, dass Pack_ChargeFET nicht umschaltet, wenn Pack_DischargeFET auf false steht. Die Wiederholungen an sich scheinen mir unproblematisch zu sein.

softwarecrash, 2023-04-19:
diese rückmeldung auf den DP innerhalb von pack control ist eine eigenheit die ich mir eingebaut hatte als bestätigung. normal wechseln die anderen datenpunkte dann auch den status.
das er einen befehl aber garnicht übernimmt ist etwas seltsam, das geht hier im test aktuell problemlos.
bedenke bitte das es eine einfach kommunikation ist, der esp und bms brauchen etwas zeit um einen befehl zu verarbeiten bevor der nächste kommen darf. es gibt leider keinen befehlspuffer

soso005, 2023-04-19:
Das beschriebene Verhalten ist auch dann, wenn ich die Nachricht manuell nur einmalig sende. Dann kommt es auch zu den automatischen, sehr schnelle Wiederholungen und Charge schaltet auch nach Minuten nicht um, wenn Discharge deaktiv ist.
Evtl. liegt es dann tatsächlich am BMS selbst, dass Charge nicht eingeschaltet werden kann, wenn Discharge deaktiv ist. Ich teste das mit dem zum BMS gehörigen Bluetooth-Sender und der App.

softwarecrash, 2023-04-19:
Irgendeinen Zusammenhang gab es, das könnte evt das von dir beschriebene scenario sein. Hab mich damals auch sehr gewundert.
In der App geht es seltsamerweise unabhängig, aber in der PC Software nicht.
Da ich aber das Protokoll der PC Software umgesetzt hab ist man dahingehend ein wenig angeschmiert

softwarecrash, 2023-04-19:
Dann ist das eine Bedingung vom BMS.
Elektronisch nicht ganz unsinnig, da wenn du den Ladekanal auf machst eine unbekannte Spannung ohne Ausgleich Anliegen kann, wenn erst der entladefet an ist kann die angelegte Spannung sich angleichen.
Die Thematik ist noch deutlich komplexer, aber in diese Richtung wird die Bedingung des BMS gehen.
Ich dächte mein produktiv bms hat auch dieses Verhalten, wohingegen mein deutlich älteres Test bms auf dem Tisch dieses Verhalten nicht zeigt.

softwarecrash, 2023-05-10:
kann das hier dann weg?

closed 2023-05-12 by softwarecrash