softwarecrash/Daly2MQTT #106: [Bug]:

stoc05, 2023-06-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Hallo zusammen!
Mir ist aufgefallen das die mqtt Daten im Pack  High_CellNr und High_CellV mit den Daten im CellV nicht nicht identisch sind.
Pack.High_CellNr: 3
Pack.High_CellV: 3.268
Pack.Low_CellNr: 8
Pack.Low_CellV: 3.263

CellV.CellV_3: 3,266
CellV.CellV_4 : 3.268
CellV.CellV_8 : 3.266

### Screenshots / Fotos

[image]

### Version

2.x.x and above

### Relevant livejson output

```json
object
topic: "BMS01/Pack_Data"
payload: object
Device: object
Pack: object
Voltage: 26.1
Current: -24.9
Power: -649.89
SOC: 51.9
Remaining_mAh: 145320
Cycles: 62
BMS_Temp: 24
Cell_Temp: 24
High_CellNr: 3
High_CellV: 3.268
Low_CellNr: 8
Low_CellV: 3.263
Cell_Diff: 5
DischargeFET: true
ChargeFET: true
Status: "Discharge"
Cells: 8
Heartbeat: 82
Balance_Active: false
CellV: object
CellV_1: 3.265
Balance_1: false
CellV_2: 3.267
Balance_2: false
CellV_3: 3.266
Balance_3: false
CellV_4: 3.268
Balance_4: false
CellV_5: 3.266
Balance_5: false
CellV_6: 3.268
Balance_6: false
CellV_7: 3.266
Balance_7: false
CellV_8: 3.266
Balance_8: false
CellTemp: object
qos: 0
retain: false
_topic: "BMS01/Pack_Data"
_msgid: "951cbd08.1ffef"
```

### Which BMS is connected?

8S 24V 200A

### What browsers are you seeing the problem on?

Microsoft Edge

softwarecrash, 2023-06-23:
Hallo,
hab das gerade mal geprüft, kann das bei meinem 7S /8S nicht feststellen.
die Daten von hi/low Zellen Differenz und Zellspannungen werden auch nicht gleichzeitig aktualisiert, ggf ergibt sich daraus eine gewisse Verschiebung.
die Daten der Zellspannungen manchmal wegen hash Fehlern nicht verwertet werden können, auch da kann sich eine gewisse Differenz bilden bis die Daten wieder aktuell sind
eine andere Möglichkeit ist eine gewisse Ungenauigkeit beim runden, float (insbesondere double werte) sind nicht exakt, daran kann ich aber nichts ändern.
ein "verrutschen" der Daten würde ich erst mal ausschließen, aber du kannst es gern mal beobachten, ich konnte gerade bei meinen BMS´s so etwas nicht sehen.

stoc05, 2023-06-25:
Hallo softwarecrash!
Mir ist auch aufgefallen dass das Auseinanderdriften nur beim Entladen passiert, beim Ladevorgang scheint alles normal zu funktionieren. Im Prinzip ist es auch nicht so wichtig, weil ich mit diesen Daten eh nichts mache. Die Daten wie SOC, Temperatur, Status funktionieren zuverlässig :-)

softwarecrash, 2023-06-25:
Okay, wie gesagt es kann gut sein das du auch eins von den daly hast was zickig ist, da schlägt das Lesen der Zellspannung oft fehl daraus kann das resultieren, lässt sich aber weder vermeiden noch beheben.
Kann das dann zu?

closed 2023-06-27 by softwarecrash