softwarecrash/Daly2MQTT #108: [Bug]: Can't upgrade to 2.2.18

murfy76, 2023-06-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Hello,

I can't upgrade from 2.2.17 to 2.2.18. I tried OTA upgrade over web interface and online flash tool over USB without erase option. Web interface still showing 2.2.17 version. Thank you.

Peter 

### Screenshots / Fotos

[image]

### Version

2.x.x and above

### Relevant livejson output

```json
{"Device":{"Name":"DALY-BMS-to-MQTT","IP":"192.168.9.162","ESP_VCC":2.606,"Wifi_RSSI":-75,"Relais_Active":false,"Relais_Manual":false,"sw_version":"2.2.17","Flash_Size":4194304,"Sketch_Size":420768,"Free_Sketch_Space":3751936},"Pack":{"Voltage":25.9,"Current":-2.1,"Power":-54.39,"SOC":76.8,"Remaining_mAh":46080,"Cycles":10,"BMS_Temp":19,"Cell_Temp":19,"High_CellNr":4,"High_CellV":3.249,"Low_CellNr":1,"Low_CellV":3.224,"Cell_Diff":25,"DischargeFET":true,"ChargeFET":true,"Status":"Discharge","Cells":8,"Heartbeat":165,"Balance_Active":false},"CellV":{"CellV_1":3.233,"Balance_1":false,"CellV_2":3.23,"Balance_2":false,"CellV_3":3.244,"Balance_3":false,"CellV_4":3.239,"Balance_4":false,"CellV_5":3.238,"Balance_5":false,"CellV_6":3.21,"Balance_6":false,"CellV_7":3.242,"Balance_7":false,"CellV_8":3.232,"Balance_8":false},"CellTemp":{"Cell_Temp_1":19}}
```

### Which BMS is connected?

Daly Smart, 60A Common Port, LiFePO4 8S 24V, UART

### What browsers are you seeing the problem on?

Firefox, Chrome

softwarecrash, 2023-06-29:
so one problem is your power, 2,6V is to low for most flashing.
try flash from pc with usb clabe and tasmotizer tool, without connected bms or something

softwarecrash, 2023-07-02:
Closed by no replay since 3 days

closed 2023-07-02 by softwarecrash