softwarecrash/Daly2MQTT #110: [Bug]: No BMS Data via MQTT

sideshowbob100, 2023-07-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Daly2MQTT in general works. The Web Interface is showing the different values of the BMS.
Then I connected the Wemos via Wifi with my iobroker but in iobroker only one object is available (alive = true). There are no more objects.
What can I do?

### Screenshots / Fotos

[image]
[image]
[image]
[image]

### Steps To Reproduce

-

### Version

2.x.x and above

### Relevant livejson output

```json
418 I'm a teapot
```

### Which BMS is connected?

BMS 16s 80A

### What browsers are you seeing the problem on?

Safari, Microsoft Edge

softwarecrash, 2023-07-02:
Hello,
from the screenshots i see you use the sonoff adapter, you need the mqtt broker/client adapter, set it as broker, and change the port to like 1885 (in esp too) to avoid collision with the sonoff adapter

sideshowbob100, 2023-07-02:
THX. Now it works.

closed 2023-07-02 by softwarecrash