softwarecrash/Daly2MQTT #113: [FEATURE] Cell Gauge in webUI - numbering/coloring

mt-mrx, 2023-07-22:
### Is your feature request related to a problem? Please describe

no

### Describe the solution you'd like

It would be really nice to have a numbering of the individual cell bars in the the Cell Gauge in the webUI. Basically an x-axis label.
I believe if you only have a 4S BMS it's not really needed but in my case I have a 15S and it would be really nice to see which cell correlates to which bar.

It would also be nice to color the highest and lowest cell bar. Maybe with the same colors high=red, low=light blue as the Daly app uses? 

### Describe alternatives you've considered

don't change anything ;-)

softwarecrash, 2023-07-23:
whatever you wish
[image]

closed 2023-07-23 by softwarecrash

mt-mrx, 2023-07-23:
wohoo, great, thanks

[image]

softwarecrash, 2023-07-23:
there a third color for cells with active balance, please report if it correct numbering, i cant test it actual

mt-mrx, 2023-07-23:
Something is not quite right, I see the color when the BMS is balancing but it showed first one cell (#5) being balanced and then highlighted #13 as well.

Btw. we can continue in English or switch to German, whatever is preferred.

[image]

[image]

softwarecrash, 2023-07-23:
i think that is correct, compare it with the mqtt values oder the esp-ip/livejson, it can have more than one cell at the same time that is in balance state

mt-mrx, 2023-07-23:
You are correct. I checked the livejson, sometimes two cells have a balance value of "true".

It's also a bit hard to check what is exactly happening because I have the Daly Active Balancer running in parallel, but I'm not monitoring it, as a result the BMS balancing changes quite fast.

```
{
  "Device": {
    "Name": "DALY-BMS-to-MQTT",
    "IP": "192.168.199.34",
    "ESP_VCC": 3.002,
    "Wifi_RSSI": -58,
    "Relais_Active": false,
    "Relais_Manual": false,
    "sw_version": "2.4.1",
    "Flash_Size": 4194304,
    "Sketch_Size": 424032,
    "Free_Sketch_Space": 3747840
  },
  "Pack": {
    "Voltage": 52.2,
    "Current": 0,
    "Power": 0,
    "SOC": 99.6,
    "Remaining_mAh": 1095,
    "Cycles": 4,
    "BMS_Temp": 24,
    "Cell_Temp": 24,
    "cell_hVt": 3.65,
    "cell_lVt": 2.3,
    "High_CellNr": 13,
    "High_CellV": 3.522,
    "Low_CellNr": 1,
    "Low_CellV": 3.426,
    "Cell_Diff": 96,
    "DischargeFET": true,
    "ChargeFET": true,
    "Status": "Stationary",
    "Cells": 15,
    "Heartbeat": 251,
    "Balance_Active": true
  },
  "CellV": {
    "CellV_1": 3.427,
    "Balance_1": false,
    "CellV_2": 3.483,
    "Balance_2": false,
    "CellV_3": 3.479,
    "Balance_3": false,
    "CellV_4": 3.523,
    "Balance_4": true,
    "CellV_5": 3.493,
    "Balance_5": false,
    "CellV_6": 3.504,
    "Balance_6": false,
    "CellV_7": 3.487,
    "Balance_7": false,
    "CellV_8": 3.453,
    "Balance_8": false,
    "CellV_9": 3.433,
    "Balance_9": false,
    "CellV_10": 3.493,
    "Balance_10": false,
    "CellV_11": 3.504,
    "Balance_11": false,
    "CellV_12": 3.527,
    "Balance_12": true,
    "CellV_13": 3.515,
    "Balance_13": false,
    "CellV_14": 3.464,
    "Balance_14": false,
    "CellV_15": 3.449,
    "Balance_15": false
  },
  "CellTemp": {
    "Cell_Temp_1": 24
  }
}
```

softwarecrash, 2023-07-23:
yes with the active balancer its tricky.
so now all correct?

mt-mrx, 2023-07-23:
yes, looks good, thanks