softwarecrash/Daly2MQTT #120: [Bug]: 2.8.2 settings page not working

robertdiers, 2023-08-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Hi,

After clicking on settings, only the upload option is available.
I tried to rollback to old version, but get this error:

settings:1 Uncaught ReferenceError: postFile is not defined
    at HTMLInputElement.onclick (settings:1:2225)

regards Robert

### Screenshots / Fotos

settings:1 Uncaught ReferenceError: postFile is not defined
    at HTMLInputElement.onclick (settings:1:2225)

### Steps To Reproduce

open Settings and try to upload an old release

### Version

2.x.x and above

### Relevant livejson output

```json
n/a
```

### Which BMS is connected?

16s

### What browsers are you seeing the problem on?

Chrome

softwarecrash, 2023-08-13:
thanks,
is fixed, please re-download the relase file and flash it again
(fix is includet in the actual release)

robertdiers, 2023-08-13:
thanks, will try

robertdiers, 2023-08-13:
it works, thanks

closed 2023-08-13 by robertdiers