softwarecrash/Daly2MQTT #124: [Bug]: Settings not available anymore

toltemp, 2023-08-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

ESP-01S (1MB Flash)

### What happened?

I upgraded to 2.8.2. I think I have had 2.5.1 before, not quite sure.
After upgrading the software functions as expected in general, communicates by MQTT etc.
But the settings page shows only the upper line, where I can upload a new release and update. All other settings that used to show up below this line (configure, reboot, reset, back) are not there anymore.

### Screenshots / Fotos

[image]

### Steps To Reproduce

1. Press "Settings" on main page
2. Look ;-)

### Version

2.x.x and above

### Relevant livejson output

```json
{"Device":{"Name":"DALY-BMS","IP":"xxx","ESP_VCC":2.737,"Wifi_RSSI":-37,"Relais_Active":false,"Relais_Manual":false,"sw_version":"2.8.2","Flash_Size":4194304,"Sketch_Size":427136,"Free_Sketch_Space":3743744},"Pack":{"Voltage":26.2,"Current":0,"Power":0,"SOC":54,"Remaining_mAh":151200,"Cycles":17,"BMS_Temp":24,"Cell_Temp":24,"cell_hVt":3.65,"cell_lVt":2.3,"High_CellNr":6,"High_CellV":3.283,"Low_CellNr":8,"Low_CellV":3.275,"Cell_Diff":8,"DischargeFET":true,"ChargeFET":true,"Status":"Stationary","Cells":8,"Heartbeat":187,"Balance_Active":false,"Fail_Codes":""},"CellV":{"CellV_1":3.276,"Balance_1":false,"CellV_2":3.282,"Balance_2":false,"CellV_3":3.283,"Balance_3":false,"CellV_4":3.282,"Balance_4":false,"CellV_5":3.282,"Balance_5":false,"CellV_6":3.283,"Balance_6":false,"CellV_7":3.281,"Balance_7":false,"CellV_8":3.275,"Balance_8":false},"CellTemp":{"Cell_Temp_1":24}}
```

### Which BMS is connected?

Daly BMS 200A

### What browsers are you seeing the problem on?

Firefox, Chrome, Safari

softwarecrash, 2023-08-19:
Hallo,
anhand deines sceenshots nehme ich mal an das deutsch bevorzugt wird.
bitte einmal folgende schritte abarbeiten:
- bin file erneut von der release seite herunterladen und flashen ( am besten über adapter, der ESP01 hat eig. nicht genug flash übrig um ein OTA update hinzubekommen, bei ESP01 sollte das upload feld garnicht angezeigt werden.
- wenn das problem weiterhin besteht, bitte die die debug version für esp01 flashen (auch neu herunterladen)
- ist auch bei der debug das problem dann bitte auf der seite (bsp firefox) rechtsklick und auf "seitenquelltext anzeigen" und die ausgabe hier posten.

toltemp, 2023-08-20:
Hallo,
besten Dank für die schnelle Antwort. Ich habe in dem Formular einen Fehler gemacht, es handelt sich nicht um einen ESP01 sondern um einen D1 mit 4MB.

Dein erster Punkt hat das Problem sofort gelöst. Neu geflasht und die Einstellungsmöglichkeiten sind wieder da. Meine bisherigen Einstellungen sind auch erhalten geblieben, super.

Eine kleine Spende für deine großartige Arbeit sollte angekommen sein. Vielen Dank nochmal.

closed 2023-08-20 by toltemp