softwarecrash/Daly2MQTT #126: [Bug]: WebSerial not working

YuriyKu, 2023-08-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Hello. Seem like webserial page stop working for debug version after latest release.

### Screenshots / Fotos

[image]
[image]

### Steps To Reproduce

1. Go to main page;
2. Press Settings;
3. Press WebSerial;
4. I'm teapot displayed

### Version

2.x.x and above

### Relevant livejson output

```json
{"Device":{"Name":"Daly2MQTT","IP":"192.168.1.34","ESP_VCC":3.337,"Wifi_RSSI":-79,"Relais_Active":false,"Relais_Manual":false,"sw_version":"2.8.2_debug d1_mini Aug 17 2023 23:56:43","Flash_Size":4194304,"Sketch_Size":501424,"Free_Sketch_Space":3670016,"CPU_Frequency":80,"Real_Flash_Size":4194304,"Free_Heap":29808,"HEAP_Fragmentation":11,"Free_BlockSize":26624,"json_memory_usage":2045,"json_capacity":2048,"runtime":694,"ws_clients":1,"MQTT_Json":false},"Pack":{"Voltage":53.2,"Current":0,"Power":0,"SOC":98.8,"Remaining_mAh":158080,"Cycles":31,"BMS_Temp":24,"Cell_Temp":24,"cell_hVt":3.65,"cell_lVt":2.5,"High_CellNr":10,"High_CellV":3.332,"Low_CellNr":1,"Low_CellV":3.329,"Cell_Diff":3,"DischargeFET":true,"ChargeFET":true,"Status":"Stationary","Cells":16,"Heartbeat":179,"Balance_Active":false,"Fail_Codes":""},"CellV":{"CellV_1":3.329,"Balance_1":false,"CellV_2":3.331,"Balance_2":false,"CellV_3":3.331,"Balance_3":false,"CellV_4":3.331,"Balance_4":false,"CellV_5":3.331,"Balance_5":false,"CellV_6":3.331,"Balance_6":false,"CellV_7":3.331,"Balance_7":false,"CellV_8":3.331,"Balance_8":false,"CellV_9":3.331,"Balance_9":false,"CellV_10":3.332,"Balance_10":false,"CellV_11":3.33,"Balance_11":false,"CellV_12":3.33,"Balance_12":false,"CellV_13":3.331,"Balance_13":false,"CellV_14":3.33,"Balance_14":false,"CellV_15":3.331,"Balance_15":false,"CellV_16":3.328,"Balance_16":false},"CellTemp":{"Cell_Temp_1":24}}
```

### Which BMS is connected?

16S 48V Daly

### What browsers are you seeing the problem on?

Firefox

softwarecrash, 2023-08-20:
Hello,
The web serial function is actual disabled, it Makes Problems with ota.
it will come Back with the next Bugfix release

softwarecrash, 2023-08-20:
fixed with V2.8.3

closed 2023-08-20 by softwarecrash