softwarecrash/Daly2MQTT #127: [Bug]: Settings page only contains update. But even that no longer works

dj9717, 2023-08-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

I use your fully stocked and tested PCB.
Since Release 2.8.2 i dont see the full settings menu anymore.
Only the update funktion are available. But even that no longer works.
Tested on newest available release of firefox and chrome both under android and linux.
But i can, for example, call up and use the settingsedit page directly.

### Screenshots / Fotos

[image]

### Steps To Reproduce

1. go to main Page
2. choose Settings

### Version

2.x.x and above

### Relevant livejson output

```json
{"Device":{"Name":"Daly2MQTT","IP":"xxx.xxx.xxx.xxx","ESP_VCC":2.71,"Wifi_RSSI":-46,"Relais_Active":false,"Relais_Manual":false,"sw_version":"2.8.2","Flash_Size":4194304,"Sketch_Size":427136,"Free_Sketch_Space":3743744},"Pack":{"Voltage":53.8,"Current":11.8,"Power":634.84,"SOC":90.7,"Remaining_mAh":253960,"Cycles":14,"BMS_Temp":24,"Cell_Temp":24,"cell_hVt":3.65,"cell_lVt":2.3,"High_CellNr":14,"High_CellV":3.369,"Low_CellNr":9,"Low_CellV":3.365,"Cell_Diff":4,"DischargeFET":true,"ChargeFET":true,"Status":"Charge","Cells":16,"Heartbeat":132,"Balance_Active":false,"Fail_Codes":""},"CellV":{"CellV_1":3.366,"Balance_1":false,"CellV_2":3.368,"Balance_2":false,"CellV_3":3.369,"Balance_3":false,"CellV_4":3.369,"Balance_4":false,"CellV_5":3.369,"Balance_5":false,"CellV_6":3.369,"Balance_6":false,"CellV_7":3.368,"Balance_7":false,"CellV_8":3.369,"Balance_8":false,"CellV_9":3.365,"Balance_9":false,"CellV_10":3.369,"Balance_10":false,"CellV_11":3.365,"Balance_11":false,"CellV_12":3.368,"Balance_12":false,"CellV_13":3.369,"Balance_13":false,"CellV_14":3.369,"Balance_14":false,"CellV_15":3.367,"Balance_15":false,"CellV_16":3.365,"Balance_16":false},"CellTemp":{"Cell_Temp_1":24}}
```

### Which BMS is connected?

BMS Type Daly 16S 250 A UART.

### What browsers are you seeing the problem on?

Firefox, Chrome

softwarecrash, 2023-08-21:
please upgrade to v2.8.3

dj9717, 2023-08-21:
Ok, update via esptool because i don't have a tasmotizer. The Bug is fixed.
thank you

closed 2023-08-21 by softwarecrash