softwarecrash/Daly2MQTT #132: [Bug]: Stability of communication between ESP and BMS

mt-mrx, 2023-09-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

I'm seeing a lot of communication instability between the ESP (Wemos D1 mini) and the BMS.

My main question is, is that level of "CRC fail" and "NO DATA" normal or not?

And if it's not normal, any ideas what I could check? Any experience with sort of issue? I can hook it up to the PC software. I could hook up an oscilloscope to the signal lines and verify that the signal flanks are sharp and the voltage levels fine.

Here and excerpt from the debug log.
```
[8:47:49 PM] <MQTT> Data sent to MQTT Server... [8:47:49 PM] Done[8:47:49 PM] 
[8:48:07 PM] <UART> NO DATA[8:48:08 PM] 
[8:48:08 PM] <DALY-BMS DEBUG> Receive failed, Temp values won't be modified!
[8:48:20 PM] <MQTT> Data sent to MQTT Server... [8:48:20 PM] Done[8:48:20 PM] 
[8:48:51 PM] <MQTT> Data sent to MQTT Server... [8:48:51 PM] Done[8:48:51 PM] 
[8:48:56 PM] <UART> CRC FAIL[8:48:56 PM] 
[8:49:14 PM] <UART> CRC FAIL[8:49:14 PM] 
[8:49:22 PM] <MQTT> Data sent to MQTT Server... [8:49:22 PM] Done[8:49:22 PM] 
[8:49:39 PM] <UART> NO DATA[8:49:39 PM] 
[8:49:40 PM] <UART> NO DATA[8:49:41 PM] 
[8:49:41 PM] <UART> NO DATA[8:49:41 PM] 
[8:49:45 PM] <UART> NO DATA[8:49:45 PM] 
[8:49:46 PM] <UART> NO DATA[8:49:47 PM] 
[8:49:48 PM] <UART> NO DATA[8:49:48 PM] 
[8:49:48 PM] <DALY-BMS DEBUG> Receive failed, Temp values won't be modified!
[8:49:49 PM] <UART> NO DATA[8:49:49 PM] 
[8:49:53 PM] <UART> CRC FAIL[8:49:53 PM] 
[8:49:55 PM] <UART> CRC FAIL[8:49:55 PM] 
[8:49:55 PM] <UART> NO DATA[8:49:55 PM] 
[8:49:55 PM] <DALY-BMS DEBUG> Receive failed, Temp values won't be modified!
[8:49:57 PM] <MQTT> Data sent to MQTT Server... [8:49:57 PM] Done[8:49:57 PM] 
[8:49:59 PM] <UART> NO DATA[8:49:59 PM] 
[8:50:00 PM] <UART> NO DATA[8:50:00 PM] 
[8:50:03 PM] <UART> NO DATA[8:50:03 PM] 
[8:50:04 PM] <UART> NO DATA[8:50:04 PM] 
[8:50:05 PM] <UART> NO DATA[8:50:05 PM] 
[8:50:06 PM] <UART> NO DATA[8:50:06 PM] 
[8:50:09 PM] <UART> CRC FAIL[8:50:09 PM] 
[8:50:10 PM] <UART> NO DATA[8:50:10 PM] 
[8:50:11 PM] <UART> NO DATA[8:50:11 PM] 
[8:50:12 PM] <UART> NO DATA[8:50:12 PM] 
[8:50:15 PM] <UART> NO DATA[8:50:15 PM] 
[8:50:16 PM] <UART> NO DATA[8:50:16 PM] 
[8:50:16 PM] <DALY-BMS DEBUG> Receive failed, Temp values won't be modified!
[8:50:16 PM] <UART> NO DATA[8:50:16 PM] 
[8:50:17 PM] <UART> NO DATA[8:50:17 PM] 
[8:50:18 PM] <UART> NO DATA[8:50:18 PM] 
[8:50:18 PM] <DALY-BMS DEBUG> Receive failed, Temp values won't be modified!
[8:50:18 PM] <UART> NO DATA[8:50:18 PM] 
[8:50:18 PM] <DALY-BMS DEBUG> Receive failed, Temp values won't be modified!
[8:50:19 PM] <UART> NO DATA[8:50:19 PM] 
[8:50:20 PM] <UART> NO DATA[8:50:20 PM] 
[8:50:25 PM] <UART> NO DATA[8:50:26 PM] 
[8:50:28 PM] <MQTT> Data sent to MQTT Server... [8:50:28 PM] Done[8:50:28 PM] 
[8:50:44 PM] <UART> CRC FAIL[8:50:44 PM] 
[8:50:45 PM] <UART> NO DATA[8:50:45 PM] 
[8:50:45 PM] <DALY-BMS DEBUG> Receive failed, Temp values won't be modified!
[8:50:54 PM] <UART> NO DATA[8:50:54 PM] 
[8:50:54 PM] <DALY-BMS DEBUG> Receive failed, Temp values won't be modified!
[8:50:59 PM] <MQTT> Data sent to MQTT Server... [8:50:59 PM] Done[8:50:59 PM] 
[8:51:00 PM] <UART> NO DATA[8:51:00 PM] 
[8:51:00 PM] <DALY-BMS DEBUG> Receive failed, Temp values won't be modified!
[... 127 lines cut ...]
    "Low_CellNr": 9,
    "Low_CellV": 3.268,
    "Cell_Diff": 3,
    "DischargeFET": true,
    "ChargeFET": true,
    "Status": "Discharge",
    "Cells": 15,
    "Heartbeat": 120,
    "Balance_Active": false,
    "Fail_Codes": ""
  },
...
}
```

### Which BMS is connected?

Lifepo4 15S 48V 60A UART / Bluetooth

It has a connector where I've seen the CAN/485 label on other BMS pictures, but it has no printing above the connector.
So it has only printing for "NTC" and "UART" on the case.

### What browsers are you seeing the problem on?

no Issue with the Browser or WebUI

softwarecrash, 2023-09-28:
Hello,
some crc fails are ok, but your log looks not so good, no data means that the bms is not connected or in sleep mode and reflect the query commands.
a crc fail quote 5/100 is normal.
can you provide pictures of your setup.
whats your wire length?
whats for a model code have your bms?
whats your power source for esp and for the battery/bms? some power sources like inverters made a heavy Em field that disturb the communication
about the can/485, this ports are only active when it have a label

mt-mrx, 2023-09-28:
Hi,

Power source of the ESP: MeanWell HDR-15-5 / 5v / 2.4A connected to AC
Model Code BMS: R25T -IC16

Cable length from ESP to BMS is ~20cm. Basically I had first ~10cm but then cut off the ends and attached the ADUM1201 and added another ~10cm.

Chargers: Meanwell HLG-100H-54AB + HLG-480H-54AB
Discharging via inverter Hoymiles HM-300.
But I see the same communication issues when the chargers are AC disconnected (relais) + DC disconnected (MOSFET) and also the inverter is diconnected AC (relais) + DC (MOSFET) so I can exclude them as error sources.

While charging or discharging I'm sending an MQTT wake messages every 4 minutes, the BMS is set to sleep after 5 minutes. ESP is set for 30s MQTT update.

I have attached some pictures, but while taking them I disconnected the ESP from the BMS and reconnected it. Afterwards the communication looks much better, so maybe only a bad seated connector. 
If that was it, sorry for bothering you. :(

thanks

Matthias

Some more log output after reconnecting it.
```
[6:42:45 PM] <WEBS> Data sent to WebSocket... [6:42:45 PM] Done[6:42:45 PM] 
[6:42:46 PM] <WEBS> Data sent to WebSocket... [6:42:47 PM] Done[6:42:47 PM] 
[6:42:57 PM] <MQTT> Data sent to MQTT Server... [6:42:58 PM] Done[6:42:58 PM] 
[6:43:20 PM] <UART> CRC FAIL[6:43:20 PM] 
[6:43:20 PM] <MQTT> MQTT Callback: message recived: true[6:43:20 PM] 
[6:43:20 PM] <MQTT> MQTT Callback: wakeup manual from Web[6:43:20 PM] 
[6:43:20 PM] <SYS > Wakeup acivated[6:43:20 PM] 
[6:43:20 PM] <SYS > Wakeup deacivated[6:43:20 PM] 
[6:43:22 PM] <UART> CRC FAIL[6:43:22 PM] 
[6:43:23 PM] <MQTT> Data sent to MQTT Server... [6:43:24 PM] Done[6:43:24 PM] 
[6:43:32 PM] <UART> CRC FAIL[6:43:32 PM] 
[6:43:54 PM] <MQTT> Data sent to MQTT Server... [6:43:55 PM] Done[6:43:56 PM] 
[6:44:26 PM] <MQTT> Data sent to MQTT Server... [6:44:26 PM] Done[6:44:26 PM] 
[6:44:42 PM] <UART> CRC FAIL[6:44:42 PM] 
[6:44:56 PM] <MQTT> Data sent to MQTT Server... [6:44:56 PM] Done[6:44:56 PM] 
[6:45:07 PM] <UART> CRC FAIL[6:45:07 PM] 
[6:45:22 PM] <UART> NO DATA[6:45:22 PM] 
[6:45:22 PM] <DALY-BMS DEBUG> Receive failed, Charge / discharge mos Status won't be modified!
[6:45:27 PM] <MQTT> Data sent to MQTT Server... [6:45:27 PM] Done[6:45:27 PM] 
[6:45:58 PM] <MQTT> Data sent to MQTT Server... [6:45:58 PM] Done[6:45:58 PM] 
[6:46:11 PM] <UART> CRC FAIL[6:46:11 PM] 
[6:46:28 PM] <MQTT> Data sent to MQTT Server... [6:46:29 PM] Done[6:46:29 PM] 
[6:47:00 PM] <MQTT> Data sent to MQTT Server... [6:47:00 PM] Done[6:47:00 PM] 
[6:47:32 PM] <MQTT> Data sent to MQTT Server... [6:47:32 PM] Done[6:47:32 PM] 
[6:47:53 PM] <UART> CRC FAIL[6:47:53 PM] 
[6:48:02 PM] <MQTT> Data sent to MQTT Server... [6:48:02 PM] Done[6:48:02 PM] 
[6:48:10 PM] <UART> CRC FAIL[6:48:10 PM] 
[6:48:13 PM] <UART> CRC FAIL[6:48:13 PM] 
[6:48:33 PM] <MQTT> Data sent to MQTT Server... [6:48:33 PM] Done[6:48:33 PM] 
[6:49:04 PM] <MQTT> Data sent to MQTT Server... [6:49:04 PM] Done[6:49:05 PM] 
[6:49:36 PM] <MQTT> Data sent to MQTT Server... [6:49:36 PM] Done[6:49:36 PM] 
[6:49:38 PM] <UART> NO DATA[6:49:38 PM] 
[6:49:38 PM] <DALY-BMS DEBUG> Receive failed, Charge / discharge mos Status won't be modified!
[6:50:06 PM] <MQTT> Data sent to MQTT Server... [6:50:06 PM] Done[6:50:07 PM] 
[6:50:12 PM] <UART> CRC FAIL[6:50:12 PM] 
[6:50:37 PM] <MQTT> Data sent to MQTT Server... [6:50:37 PM] Done[6:50:37 PM] 
[6:50:53 PM] <UART> CRC FAIL[6:50:53 PM] 
[6:50:55 PM] <UART> CRC FAIL[6:50:56 PM] 
[6:51:07 PM] <MQTT> Data sent to MQTT Server... [6:51:08 PM] Done[6:51:08 PM] 
[6:51:33 PM] <UART> CRC FAIL[6:51:33 PM] 
[6:51:35 PM] <UART> CRC FAIL[6:51:35 PM] 
[6:51:38 PM] <MQTT> Data sent to MQTT Server... [6:51:38 PM] Done[6:51:38 PM] 
[6:52:10 PM] <MQTT> Data sent to MQTT Server... [6:52:10 PM] Done[6:52:10 PM] 
[6:52:41 PM] <MQTT> Data sent to MQTT Server... [6:52:41 PM] Done[6:52:41 PM] 
[6:53:13 PM] <MQTT> Data sent to MQTT Server... [6:53:13 PM] Done[6:53:13 PM] 
[6:53:32 PM] <UART> CRC FAIL[6:53:32 PM] 
[6:53:34 PM] <UART> CRC FAIL[6:53:34 PM] 
[6:53:44 PM] <MQTT> Data sent to MQTT Server... [6:53:44 PM] Done[6:53:44 PM] 
[... 29 lines cut ...]
[6:59:07 PM] <UART> CRC FAIL[6:59:07 PM] 
[6:59:16 PM] <MQTT> MQTT Callback: message recived: true[6:59:16 PM] 
[6:59:16 PM] <MQTT> MQTT Callback: wakeup manual from Web[6:59:16 PM] 
[6:59:16 PM] <SYS > Wakeup acivated[6:59:16 PM] 
[6:59:16 PM] <MQTT> Data sent to MQTT Server... [6:59:16 PM] Done[6:59:16 PM] 
[6:59:16 PM] <SYS > Wakeup deacivated[6:59:48 PM] <MQTT> Data sent to MQTT Server... [6:59:48 PM] Done[6:59:48 PM] 
[7:00:18 PM] <MQTT> Data sent to MQTT Server... [7:00:18 PM] Done[7:00:18 PM] 
[7:00:40 PM] <UART> CRC FAIL[7:00:40 PM] 
[7:00:48 PM] <MQTT> Data sent to MQTT Server... [7:00:48 PM] Done[7:00:48 PM] 
[7:00:58 PM] <UART> CRC FAIL[7:00:58 PM] 
[7:01:00 PM] <UART> CRC FAIL[7:01:00 PM] 
[7:01:02 PM] <UART> CRC FAIL[7:01:02 PM] 
[7:01:19 PM] <MQTT> Data sent to MQTT Server... [7:01:19 PM] Done[7:01:19 PM] 
[7:01:51 PM] <MQTT> Data sent to MQTT Server... [7:01:51 PM] Done[7:01:51 PM] 
[7:02:22 PM] <MQTT> Data sent to MQTT Server... [7:02:23 PM] Done[7:02:23 PM] 
[7:02:43 PM] <UART> CRC FAIL[7:02:43 PM] 
[7:02:53 PM] <MQTT> Data sent to MQTT Server... [7:02:53 PM] Done[7:02:53 PM] 
```

[image]
[image]
[image]
[image]
[image]
[image]

softwarecrash, 2023-09-28:
thanks for that much input.
yeah some bms are sensitive, i noticed one that doenst work, but after dissconnect and reconnect all, it start working, 
so i think you have one of them, the uart signal from this bms´s are verry weak, on the logik and oszi it looks okay, but the voltage level sometimes is to low and the bms have not enugh power to pull down the rx line.

all other from you looks verry good. a little mistake is the antenna from the wemos point to the cables, but this is not the problem.
the crc erros are not nice, but not so bad.
i hope it will stay working

mt-mrx, 2023-09-28:
No need for thanking me for providing debug data, I'm looking for support, so thank you for looking into it. ;-)

Yeah, I had the Wemos antenna point that way because I first wanted to supply it by using a standard micro usb cable but in the end decided against it.

Hmm, maybe it gets worse when the inverter is running. I'm using the inverter with a setting of constantly producing "16%" of it's output power, which is around ~50W.

I start it at sunset and have it run until the lowest cell voltage is 3.0V.

So the following log is from when the inverter started at 7:13:45 PM till I turned it off at 7:46:00 PM.

Need to test it a bit more by turning the inverter on/off and checking how everything behaves.

Any idea if something like two ferrit clamp on cores around the inverter DC + and - cables would help? 

```
[7:14:04 PM] <MQTT> Data sent to MQTT Server... [7:14:05 PM] Done[7:14:05 PM] 
[7:14:05 PM] <MQTT> MQTT Callback: message recived: true[7:14:05 PM] 
[7:14:05 PM] <MQTT> MQTT Callback: wakeup manual from Web[7:14:05 PM] 
[7:14:05 PM] <SYS > Wakeup acivated[7:14:05 PM] 
[7:14:05 PM] <SYS > Wakeup deacivated[7:14:05 PM] 
[7:14:06 PM] <UART> CRC FAIL[7:14:06 PM] 
[7:14:08 PM] <UART> CRC FAIL[7:14:08 PM] 
[7:14:10 PM] <MQTT> Data sent to MQTT Server... [7:14:10 PM] Done[7:14:10 PM] 
[7:14:18 PM] <UART> CRC FAIL[7:14:18 PM] 
[7:14:20 PM] <UART> CRC FAIL[7:14:21 PM] 
[7:14:22 PM] <UART> CRC FAIL[7:14:23 PM] 
[7:14:39 PM] <UART> NO DATA[7:14:40 PM] 
[7:14:41 PM] <MQTT> Data sent to MQTT Server... [7:14:41 PM] Done[7:14:41 PM] 
[7:14:46 PM] <UART> CRC FAIL[7:14:46 PM] 
[7:14:50 PM] <UART> NO DATA[7:14:50 PM] 
[7:14:54 PM] <UART> NO DATA[7:14:54 PM] 
[7:15:11 PM] <MQTT> Data sent to MQTT Server... [7:15:12 PM] Done[7:15:12 PM] 
[7:15:43 PM] <MQTT> Data sent to MQTT Server... [7:15:43 PM] Done[7:15:43 PM] 
[7:16:14 PM] <MQTT> Data sent to MQTT Server... [7:16:14 PM] Done[7:16:15 PM] 
[7:16:45 PM] <UART> CRC FAIL[7:16:45 PM] 
[7:16:46 PM] <MQTT> Data sent to MQTT Server... [7:16:47 PM] Done[7:16:47 PM] 
[7:17:16 PM] <MQTT> Data sent to MQTT Server... [7:17:16 PM] Done[7:17:16 PM] 
[7:17:18 PM] <UART> CRC FAIL[7:17:19 PM] 
[7:17:20 PM] <UART> CRC FAIL[7:17:20 PM] 
[7:17:27 PM] <UART> NO DATA[7:17:28 PM] 
[7:17:28 PM] <DALY-BMS DEBUG> Receive failed, Charge / discharge mos Status won't be modified!
[7:17:32 PM] <UART> CRC FAIL[7:17:33 PM] 
[7:17:39 PM] <UART> NO DATA[7:17:39 PM] 
[7:17:39 PM] <DALY-BMS DEBUG> Receive failed, Charge / discharge mos Status won't be modified!
[7:17:47 PM] <MQTT> Data sent to MQTT Server... [7:17:47 PM] Done[7:17:47 PM] 
[7:18:05 PM] <MQTT> MQTT Callback: message recived: true[7:18:05 PM] 
[7:18:05 PM] <MQTT> MQTT Callback: wakeup manual from Web[7:18:05 PM] 
[7:18:05 PM] <SYS > Wakeup acivated[7:18:05 PM] 
[7:18:05 PM] <SYS > Wakeup deacivated[7:18:05 PM] 
[7:18:06 PM] <MQTT> Data sent to MQTT Server... [7:18:06 PM] Done[7:18:06 PM] 
[7:18:21 PM] <UART> CRC FAIL[7:18:21 PM] 
[7:18:36 PM] <MQTT> Data sent to MQTT Server... [7:18:36 PM] Done[7:18:36 PM] 
[7:19:08 PM] <MQTT> Data sent to MQTT Server... [7:19:08 PM] Done[7:19:08 PM] 
[7:19:15 PM] <UART> CRC FAIL[7:19:15 PM] 
[7:19:17 PM] <UART> CRC FAIL[7:19:17 PM] 
[7:19:18 PM] <UART> CRC FAIL[7:19:19 PM] 
[7:19:39 PM] <MQTT> Data sent to MQTT Server... [7:19:39 PM] Done[7:19:39 PM] 
[7:20:05 PM] <UART> CRC FAIL[7:20:05 PM] 
[7:20:09 PM] <MQTT> Data sent to MQTT Server... [7:20:10 PM] Done[7:20:10 PM] 
[7:20:15 PM] <UART> CRC FAIL[7:20:15 PM] 
[7:20:24 PM] <UART> CRC FAIL[7:20:24 PM] 
[7:20:33 PM] <UART> NO DATA[7:20:33 PM] 
[7:20:33 PM] <DALY-BMS DEBUG> Receive failed, min/max cell values won't be modified!
[7:20:33 PM] <UART> NO DATA[7:20:34 PM] 
[7:20:34 PM] <DALY-BMS DEBUG> Receive failed, min/max cell values won't be modified!
[7:20:34 PM] <UART> NO DATA[7:20:34 PM] 
[7:20:34 PM] <DALY-BMS DEBUG> Receive failed, min/max cell values won't be modified!
[7:20:40 PM] <MQTT> Data sent to MQTT Server... [7:20:40 PM] Done[7:20:40 PM] 
[7:20:40 PM] <UART> NO DATA[7:20:40 PM] 
[7:20:40 PM] <DALY-BMS DEBUG> Receive failed, min/max cell values won't be modified!
[... 152 lines cut ...]
[7:44:10 PM] <MQTT> Data sent to MQTT Server... [7:44:11 PM] Done[7:44:11 PM] 
[7:44:39 PM] <UART> CRC FAIL[7:44:39 PM] 
[7:44:41 PM] <MQTT> Data sent to MQTT Server... [7:44:41 PM] Done[7:44:41 PM] 
[7:45:09 PM] <UART> CRC FAIL[7:45:10 PM] 
[7:45:11 PM] <MQTT> Data sent to MQTT Server... [7:45:11 PM] Done[7:45:11 PM] 
[7:45:18 PM] <UART> NO DATA[7:45:18 PM] 
[7:45:18 PM] <DALY-BMS DEBUG> Receive failed, Charge / discharge mos Status won't be modified!
[7:45:22 PM] <UART> CRC FAIL[7:45:22 PM] 
[7:45:42 PM] <MQTT> Data sent to MQTT Server... [7:45:42 PM] Done[7:45:42 PM] 
[7:45:47 PM] <UART> CRC FAIL[7:45:47 PM] 
[7:46:05 PM] <MQTT> MQTT Callback: message recived: true[7:46:05 PM] 
[7:46:05 PM] <MQTT> MQTT Callback: wakeup manual from Web[7:46:05 PM] 
[7:46:05 PM] <SYS > Wakeup acivated[7:46:05 PM] 
[7:46:05 PM] <SYS > Wakeup deacivated[7:46:06 PM] 
[7:46:06 PM] <MQTT> Data sent to MQTT Server... [7:46:06 PM] Done[7:46:06 PM] 
[7:46:38 PM] <MQTT> Data sent to MQTT Server... [7:46:38 PM] Done[7:46:38 PM] 
[7:46:41 PM] <UART> CRC FAIL[7:46:42 PM] 
[7:46:43 PM] <UART> CRC FAIL[7:46:43 PM] 
[7:47:08 PM] <MQTT> Data sent to MQTT Server... [7:47:08 PM] Done[7:47:09 PM] 
[7:47:40 PM] <MQTT> Data sent to MQTT Server... [7:47:40 PM] Done[7:47:40 PM] 
```
Inverter turned off
```

```

softwarecrash, 2023-09-28:
Inverter Makes a Lot of EMV so every ferit or shielded cable are good idea, on the other hand, it make it Not worse.
So when you get the needed Data near Realtime you can ignore some crc Errors.
I think your BMS its a Bit "faulty" in comunication, so we cant make Things better.
Remember you See only the fails, Not the succsessfull requests. So Long the fails Not more then the good requests all is okay.

mt-mrx, 2023-09-28:
> PS: i have to thank for the great information, so many User ASK for Help Like "its Not working, do Something" ;)

Occupational hazard :-D , I'm working as a IT administrator in data center, so I know what it is like to get the "it doesn't work and I didn't do anything" requests.

Anyway, I'll try my luck with fiddeling around a bit more, maybe I can improve something. If I do, I'll report back.

closed 2023-09-28 by mt-mrx