softwarecrash/Daly2MQTT #134: [FEATURE] <title>

FloMaetschke, 2023-10-19:
### Is your feature request related to a problem? Please describe

Ich habe eben nochmal seit längerem die Weboberfläche aufgemacht, gesehen dass es ein Update gab.
Soweit, so gut. Herunter geladen, und den Update Dialog entdeckt. Update ausgeführt. Bumm, kaputt.
 Ich Esel habe die esp01 Variante auf den D1 Mini geflasht. Da war auch nen ESP drauf.
Also Gehäuse aufschrauben, abstecken, an den PC nehmen, micro-usb Kabel suchen, neu drauf flashen, geht wieder.
Nicht tragisch eigentlich.

### Describe the solution you'd like

Könnte man nicht leicht auf den Dateinamen (esp01, d1_mini)) oder einer identifizierenden Daten-Stelle im .bin schauen, um welches Image es sich handelt und eine entsprechende Warnung anzeigen wie:
"Achtung, das von ihnen gewählte Image ist vermutlich inkompatibel Wirklich fortsetzen? OK/Abbrechen"

### Describe alternatives you've considered

.

softwarecrash, 2023-10-19:
hallo,
also ohne weiteres in das bin file schauen was darin ist klappt nicht.
über den dateinamen evt möglich, aber auch da könnte es zu problemen kommen wenn man statt daly ausversehen solar oder epever erwischt.

FloMaetschke, 2023-10-19:
Naja, ich hab zuerst auf den Hinweis unten im Footer geklickt: Version 2.9 --> da gabs nur 4 Dateien für 2 Boards jeweils mit oder ohne Debug. Ich glaube das zu unterscheiden würden den meisten unbedarften Usern schon helfen. Alle anderen werden sicher eh wissen, was sie tun.

softwarecrash, 2023-10-19:
Das zu unterscheiden oder auszublenden ist unmöglich.

FloMaetschke, 2023-10-21:
Oh, das wusste ich nicht. Ich bin nicht so tief in der Materie. Ich hatte angenommen, man könnte eine Variable beim Build je nach target machen und dort den string reinspeichern, und beim Upload den Dateinamen prüfen, ob der String im Dateinamen enthalten ist? Wenn nicht: "Achtung! Sind Sie sicher, dass die Firmware für den %TARGET_NAME% ist ? Ja/Abbrechen."  kA. Irgendwie so.

softwarecrash, 2023-10-21:
wir haben das gerade mal besprochen, das wäre natürlich über den dateinamen möglich, aber sobald jemand die datei umbenennt oder selber compiliert passt das wieder nicht mehr und würde den upload sperren.
daher werden wir diese funktion nicht umsetzen.

closed 2023-10-21 by softwarecrash