softwarecrash/Daly2MQTT #136: HA Integration doesn´t work anymore

Emmpunkt, 2023-11-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

After updating to 2.10 the ha integration doesn´t work anymore.
The switches are shown, but no cellvoltages etc.

The integration did not work (at least for me) properly before, evertime i had to restart homeassistant, i had to do a "/set?ha" again.
After  a restart/reboot everthing was gone.
But now it shows no values, only the switches. Values = "Nicht verfügbar / Not available"

### Screenshots / Fotos

[image]

### Version

2.x.x and above

### Relevant livejson output

```json
// 20231105142453
// http://192.168.2.152/livejson

{
  "Device": {
    "Name": "Daly2MQTT",
    "IP": "192.168.2.152",
    "ESP_VCC": 3.297,
    "Wifi_RSSI": -81,
    "Relais_Active": false,
    "Relais_Manual": false,
    "sw_version": "2.10.0",
    "Flash_Size": 4194304,
    "Sketch_Size": 440192,
    "Free_Sketch_Space": 3731456
  },
  "Pack": {
    "Voltage": 13.1,
    "Current": 0,
    "Power": 0,
    "SOC": 0.1,
    "Remaining_mAh": 280,
    "Cycles": 12,
    "BMS_Temp": 13,
    "Cell_Temp": 13,
    "cell_hVt": 3.65,
    "cell_lVt": 2.3,
    "High_CellNr": 3,
    "High_CellV": 3.3,
    "Low_CellNr": 1,
    "Low_CellV": 3.29,
    "Cell_Diff": 10,
    "DischargeFET": true,
    "ChargeFET": true,
    "Status": "Stationary",
    "Cells": 4,
    "Heartbeat": 12,
    "Balance_Active": false,
    "Fail_Codes": "SOC Low level 2"
  },
  "CellV": {
    "CellV_1": 3.29,
    "Balance_1": false,
    "CellV_2": 3.294,
    "Balance_2": false,
    "CellV_3": 3.3,
    "Balance_3": false,
    "CellV_4": 3.297,
    "Balance_4": false
  },
  "CellTemp": {
    "Cell_Temp_1": 13,
    "Cell_Temp_2": 13
  }
}
```

### Which BMS is connected?

4s 200A
UART

### What browsers are you seeing the problem on?

Chrome

Emmpunkt, 2023-11-05:
I think i found the main problem.
In the naming-scheme for ha is now the device name. Changed from "Cell_Voltage_1" to "xxxxxxxx Cell_Voltage_1". 
Now there is still the problem that HA "forgets" the device after restart. Is it only in my installation?(Im using a docker container)

softwarecrash, 2023-11-05:
with the 2.10 update nothing has changed on the HA function or mqtt. only the basic auth is added and some other smal fixes.
so the configuration is changed and you must re enter all your data on the esp, when you enter different data like name or something, HA cannot regonize the new pad and you must configure it again. or rename all to the old things

softwarecrash, 2023-11-06:
is see german in your picture, is this language better for you?

Emmpunkt, 2023-11-06:
Might be easier for me ;)
Also nochmal auf Deutsch:
Eigentlich sollten alle Einstellungen wieder so sein wie sie waren, die waren bei mir noch noch im Chrome-Autofill vorhanden.
Das ich die Einstellungen neu machen musste, war klar, stand ja da.
An den Mqtt Sachen hat sich auch nichts geändert. Alles was ich über Grafana und Node-Red gemacht habe, läuft immer noch.
Nur bei HA hat er neue Entitäten angelegt (hab ich erst nicht gesehn). Und auch nur bei den Messwerten, nicht bei den Schaltern.
Die Entität hieß vorher z.B: Cell_Voltage_1, nachher dann Daly2MQTT_Cell_Voltage_1.
Daly2MQTT ist der Device-Name.

Im Prinzip geht es ja jetzt und wenn mans neu macht fällt es noch nicht einmal auf.
Bleibt aber immer noch die Frage warum ich die HA Konfiguration immer wieder mit /set?ha anstossen muss, wenn ich HA neu gestartet habe. Mach ich das nicht weil ichs vergessen habe, sind keine Werte da. Liegt das an meinen Docker-Container? Bei allem anderen was ich im HA habe, ist das aber nicht der Fall...

softwarecrash, 2023-11-06:
okay, also der name sollte da eigentlich nicht sein wenn er vorher da nicht war, ich schau mir das grad noch mal an.
ggf nimmt HA aus der mqtt struktur alte config einträge und vermasselt da was.

Emmpunkt, 2023-11-06:
Da steht eigendlich nicht viel drin

[image]

softwarecrash, 2023-11-06:
schau doch mal bitte mit mqttexplorer wie die struktur da aussieht bei dir und mach mal ein paar bilder

softwarecrash, 2023-11-06:
das sieht erstmal völlig normal aus, kann da grad kein fehler sehen

softwarecrash, 2023-11-06:
das einzige was scheinbar anders ist bei dir jetzt ist das es vorher wohl daly-bms-tomqtt hies und jetzt daly2mqtt.
ggf hilft es wenn du alle DP im mqtt unter homeassistant/sensor und switch löschst, die entitäten dazu auch und dann noch mal neu generieren lässt vom esp

messageexplorer, 2023-11-06:
As for me it is still working.

Emmpunkt, 2023-11-06:
Ich guck später nochmal...
Ist mir gerade noch aufgefallen Pack_Cell_Difference ist in mA anstatt mV
[image]

softwarecrash, 2023-11-06:
> Ich guck später nochmal... Ist mir gerade noch aufgefallen Pack_Cell_Difference ist in mA anstatt mV [image]

das hat messageexplorer bereits angemerkt und ist korrigiert, bugfix release dazu folgt später

messageexplorer, 2023-11-06:
@Emmpunkt,
I can see issue in configuration. Seems you forgot fill MQTT Topic:
[image]

Emmpunkt, 2023-11-06:
So ich habs gerade nochmal getestet. In HA alles gelöscht. Im MQTT Explorer gelöscht und im ESP die Config neu schreiben lassen. Oder muss ich ihn komplett neu flashen?
Hat jedenfalls nichts gebracht. Nach HA Neustart keine Werte mehr.
 ̶W̶a̶s̶ ̶i̶s̶t̶ ̶M̶Q̶T̶T̶ ̶D̶a̶t̶a̶ ̶T̶r̶i̶g̶g̶e̶r̶ ̶P̶a̶t̶h̶?̶

Das "daly-bms-to-mqtt" kommt übrigens von meinem Router (Netgear), da heisst er wohl so...

Jarnsen, 2023-11-06:
Ich schaue mir das nochmal in der HA MQTT Doku an.

Emmpunkt, 2023-11-06:
> Ich schaue mir das nochmal in der HA MQTT Doku an.

Da sich niemand anders beschwert, finde ich das schon merkwürdig. 
Ich habe allerdings nur 2 Geräte über MQTT automatisch konfiguriert. Daly2MQTT und Espresence, daher kann ich nicht wirklich eine Aussage treffen ob meine Installation auch wirklich in Ordnung ist. Alles andere ist Tasmota,Tapo,Shelly,Tuya, Mitsubishi, Emoncms ... die jeweils ne eigene Integration haben
Nach /set?ha ist auch alles wieder da.
Da ich aber über die Batteriewerte Verbraucher auf 230V umschalten möchte bevor die Batterie leer ist, wäre es schon gut wenn das auch zuverlässig funktionieren würde.

Emmpunkt, 2023-11-06:
> Schreib mir mal auf Discord… es reicht wenn dann hier die Lösung ist…

Wenn ich wüsste wie man das macht, hab ich noch nie benutzt....;D

Jarnsen, 2023-11-06:
Email
Diybyjarnsen@gmail.com

Jarnsen, 2023-11-06:
Discord ist einfach ne App. Ähnlich wie WhatsApp nur das du verschiedene Server und Channel hast

softwarecrash, 2023-11-06:
@Emmpunkt discord gehts [hier lang ](https://discord.gg/7gTJk22JDE) alles andere wird dir dann erklärt, einfach in dem solar2mqtt kanal melden, dann ziehen wir dich in ein unterkanal

Emmpunkt, 2023-11-08:
Are you closing this Issue?

messageexplorer, 2023-11-08:
Hi, I rebooted today Home assistant and Daly2MQTT device was out, did
/set?ha  and it returned back. Seems in new version messages send without
retained flag. As result HA lost configuration after reboot.

I created script to send this messages with retained flag and after HA
reboot everithing works fine.

Emmpunkt, 2023-11-08:
> Hi, I rebooted today Home assistant and Daly2MQTT device was out, did /set?ha and it returned back. Seems in new version messages send without retained flag. As result HA lost configuration after [...]

You are talking about Ver.2.10.1?

softwarecrash, 2023-11-08:
@messageexplorer pleased try the Last Release, it fixes it with retain flag

softwarecrash, 2023-11-10:
fix with last release

closed 2023-11-10 by softwarecrash