softwarecrash/Daly2MQTT #138: [Bug]: In Home assistant Pack_Cell_Difference in mA instead of mV

messageexplorer, 2023-11-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

In Home assistant Pack_Cell_Difference in mA instead of mV

{
  "state_topic": "BMS01/Pack_Cell_Difference",
  "unique_id": "sensor.Daly2MQTT_Pack_Cell_Difference",
  "name": "Pack_Cell_Difference",
  "icon": "mdi:scale-balance:",
  "unit_of_measurement": "mA",
  "device_class": "voltage",
  "device": {
    "identifiers": [
      "578146"
    ],
    "configuration_url": "http://192.168.0.21",
    "name": "Daly2MQTT",
    "model": "Daly2MQTT",
    "manufacturer": "SoftWareCrash",
    "sw_version": "Solar2MQTT 2.10.0"
  }
}

### Screenshots / Fotos

[image]

### Steps To Reproduce

1.  Go to /set?ha URL
2.  Open Home Assistant and you will see wrong UoM.

### Version

2.x.x and above

### Relevant livejson output

```json
{
  "state_topic": "BMS01/Pack_Cell_Difference",
  "unique_id": "sensor.Daly2MQTT_Pack_Cell_Difference",
  "name": "Pack_Cell_Difference",
  "icon": "mdi:scale-balance:",
  "unit_of_measurement": "mA",
  "device_class": "voltage",
  "device": {
    "identifiers": [
      "578146"
    ],
    "configuration_url": "http://192.168.0.21",
    "name": "Daly2MQTT",
    "model": "Daly2MQTT",
    "manufacturer": "SoftWareCrash",
    "sw_version": "Solar2MQTT 2.10.0"
  }
}
```

### Which BMS is connected?

Daly 4s 60A

### What browsers are you seeing the problem on?

no Issue with the Browser or WebUI

messageexplorer, 2023-11-06:
I fixed it for myself just by resending json with correct UoM. 
But it is not correct UoM,  then it can be low priority issue.

softwarecrash, 2023-11-06:
Thank you, i have corrected it, it will comes with the next bugfix release

closed 2023-11-06 by softwarecrash