softwarecrash/Daly2MQTT #144: [Bug]: Home Assistant MQTT - Entities not representing a total amount

marcellofiore, 2023-11-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Data are not available for the HomeAssistent energy dashboard. Entities not already to representing a total amount!
More info: [https://community.home-assistant.io/t/how-to-enable-power-consumption-statistics/327102/16](https://community.home-assistant.io/t/how-to-enable-power-consumption-statistics/327102/16)
HomeAssistent Docs: [https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics](https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics)

### Screenshots / Fotos

[image]

### Steps To Reproduce

1) Open HomeAssistant
2) Go to Settings
3) into Plances => Energy
4) Try adding a Sensor for **Energy storage battery**

### Version

2.x.x and above

### Relevant livejson output

```json
{"Device":{"Name":"BMS01","IP":"192.168.178.94","ESP_VCC":2.848,"Wifi_RSSI":-43,"Relais_Active":false,"Relais_Manual":false,"sw_version":"2.10.1","Flash_Size":4194304,"Sketch_Size":440288,"Free_Sketch_Space":3731456},"Pack":{"Voltage":50.8,"Current":0,"Power":0,"SOC":3.4,"Remaining_mAh":3400,"Cycles":3,"BMS_Temp":19,"Cell_Temp":19,"cell_hVt":3.65,"cell_lVt":2.3,"High_CellNr":8,"High_CellV":3.181,"Low_CellNr":9,"Low_CellV":3.173,"Cell_Diff":8,"DischargeFET":true,"ChargeFET":true,"Status":"Stationary","Cells":16,"Heartbeat":141,"Balance_Active":false,"Fail_Codes":"SOC Low level 2"},"CellV":{"CellV_1":3.176,"Balance_1":false,"CellV_2":3.178,"Balance_2":false,"CellV_3":3.178,"Balance_3":false,"CellV_4":3.179,"Balance_4":false,"CellV_5":3.178,"Balance_5":false,"CellV_6":3.179,"Balance_6":false,"CellV_7":3.177,"Balance_7":false,"CellV_8":3.181,"Balance_8":false,"CellV_9":3.173,"Balance_9":false,"CellV_10":3.175,"Balance_10":false,"CellV_11":3.176,"Balance_11":false,"CellV_12":3.175,"Balance_12":false,"CellV_13":3.176,"Balance_13":false,"CellV_14":3.179,"Balance_14":false,"CellV_15":3.174,"Balance_15":false,"CellV_16":3.176,"Balance_16":false},"CellTemp":{"Cell_Temp_1":19}}
```

### Which BMS is connected?

Daily 200A

### What browsers are you seeing the problem on?

Safari

Jarnsen, 2023-11-17:
Its possible to write german? I Know this issue, If you like to fix it, can help you…

marcellofiore, 2023-11-17:
> Its possible to write german? I Know this issue, If you like to fix it, can help you…

Ja klar, kannst du gerne auf Deutsch schreiben 😃

Jarnsen, 2023-11-18:
Also, ich kenne den Fehler warum es nicht angezeigt wird. Das liegt an der state_class. Ich frage mich nur wo du die Daten hernimmst? Daly gibt diese Daten gar nicht aus. Oder möchtest du diesen Sensor erstellen?

marcellofiore, 2023-11-18:
Hallo, genau, Problem ist state_class, muss man die neue Namen nutzen… Daten wären doch in Daily… Current * Voltage und dann die kWH rechnen… das kann in wemos direkt machen und an MQTT senden mit state_class Uhr Zeit etc… dann wird in die plance Gut angezeigt! Wie meinst du das sensor erstellen?

Jarnsen, 2023-11-18:
Das ist aber kein Bug, das ist ne neue Funktion die du möchtest, die aus der aktuellen power berechnet werden muss.

marcellofiore, 2023-11-18:
Der Bug ist dass: die MQTT nutzt die alter state_class und nicht die neue. Es reicht wenn die Current und Voltage die neue State_class bekommen, dann wird von Home Assistant gerechnet… Ich habe nur vorgeschlagen was super wäre… also Bug ist doch da 😉

Jarnsen, 2023-11-18:
KWh ist nicht Current (A) * Voltage (V), das wäre Scheinleistung (VA). KWh ist Power (W) x Zeit (h) x 1000. Aber egal. Ich zeige dir wie du den Sensor ganz einfach selber erstellen kannst.

`#sensor.daly2mqtt_pack_power - > den musst du an deine Gegebenheit anpassen, das ist der Sensor der die aktuellen W anzeigt

###########
# Berechnung AKKULADUNG in W
###########
      
    - name: "akku_import"
      unique_id: akku_import
      unit_of_measurement: 'W'
      device_class: power
      state: >-
          {% if (states('sensor.daly2mqtt_pack_power'))|float(0) > 0 %}
            {{ states('sensor.daly2mqtt_pack_power')|float(0) }}
          {% else %}
            {{ 0 }}
          {% endif %}
      availability: "{{
            [ states('sensor.daly2mqtt_pack_power')
            ] | map('is_number') | min
          }}"

   
###########
# Berechnung AKKUENTLADUNG in W
###########
      
    - name: "akku_export"
      unique_id: akku_expoert
      unit_of_measurement: 'W'
      device_class: power
      state: >-
          {% if (states('sensor.daly2mqtt_pack_power'))|float(0) < 0 %}
            {{ (states('sensor.daly2mqtt_pack_power'))|float(0) * -1 }}
          {% else %}
            {{ 0 }}
          {% endif %}
      availability: "{{
            [ states('sensor.daly2mqtt_pack_power')
            ] | map('is_number') | min
          }}"
`

Im Anschluß den Helfer Riemann Summenintegralsensor für Import und Export anlegen, so wird aus W kWh.

Dies ist nur das Beispiel für Import, Export solltest du selber hinbekommen. Es werden erst kWh angezeigt wenn auch in die jeweilige Richtung Strom geflossen ist.
[image]

So sollte es aussehen wenn fertig

Sensoren in W (aktuelle Leistung)
[image]

Sensoren in kWh (Stromverbrauch)
[image]

Wenn du die Remaining Capacity meinst, das gehört nicht ins Energydashboard. Für nen Akku verlangt HA das was rein geht und das was raus geht in wh oder kWh und das machst du wie oben beschrieben.

Außerdem geht das nach dem Mond nach einiger zeit wenn es nicht immer wieder auf 100% gesetzt wird. Anzeigen kannst du es dir lassen indem du im MQTT Explorer in homeassistant/sensor/wiedeindeviceheist den payload von Remaining Capacity anzeigen lassen diesen kopieren und unter publish einfügen. Im Anschluß device_class löschen (HA kann nichts mit Ah anfangen), wenn du willst könntest du auch device_class: " " durch state_class: "measurement" ersetzen. Aber wie gesagt, selbst wenn du es jetzt in HA siehst kann HA nix damit anfangen. Mach es wie oben beschrieben, das ist die sauberste Lösung.

marcellofiore, 2023-11-18:
> KWh ist nicht Current (A) * Voltage (V), das wäre Scheinleistung (VA). KWh ist Power (W) x Zeit (h) x 1000. Aber egal. Ich zeige dir wie du den Sensor ganz einfach selber erstellen kannst.
> 
> #sensor.daly2mqtt_pack_power - > den musst du an deine Gegebenheit anpassen, das ist der Sensor der die aktuellen W anzeigt
> 
> ```
> ###########
> # Berechnung AKKULADUNG in W
> ###########
> template:
>   - sensor:      
>     - name: "akku_import"
>       unique_id: akku_import
>       unit_of_measurement: 'W'
>       device_class: power
>       state: >-
>           {% if (states('sensor.daly2mqtt_pack_power'))|float(0) > 0 %}
>             {{ states('sensor.daly2mqtt_pack_power')|float(0) }}
>           {% else %}
>             {{ 0 }}
>           {% endif %}
>       availability: "{{
>             [ states('sensor.daly2mqtt_pack_power')
>             ] | map('is_number') | min
>           }}"
> 
> 
>  
>   
> ###########
> # Berechnung AKKUENTLADUNG in W
> ###########
>       
>     - name: "akku_export"
>       unique_id: akku_expoert
>       unit_of_measurement: 'W'
>       device_class: power
>       state: >-
>           {% if (states('sensor.daly2mqtt_pack_power'))|float(0) < 0 %}
>             {{ (states('sensor.daly2mqtt_pack_power'))|float(0) * -1 }}
>           {% else %}
>             {{ 0 }}
>           {% endif %}
>       availability: "{{
>             [ states('sensor.daly2mqtt_pack_power')
>             ] | map('is_number') | min
>           }}"
> ```
> 
> Im Anschluß den Helfer Riemann Summenintegralsensor für Import und Export anlegen, so wird aus W kWh.
> 
> Dies ist nur das Beispiel für Import, Export solltest du selber hinbekommen. Es werden erst kWh angezeigt wenn auch in die jeweilige Richtung Strom geflossen ist. [image]
> 
> So sollte es aussehen wenn fertig
> 
> Sensoren in W (aktuelle Leistung) [image]
> 
> Sensoren in kWh (Stromverbrauch) [image]
> 
> Wenn du die Remaining Capacity meinst, das gehört nicht ins Energydashboard. Für nen Akku verlangt HA das was rein geht und das was raus geht in wh oder kWh und das machst du wie oben beschrieben.
> 
> Außerdem geht das nach dem Mond nach einiger zeit wenn es nicht immer wieder auf 100% gesetzt wird. Anzeigen kannst du es dir lassen indem du im MQTT Explorer in homeassistant/sensor/wiedeindeviceheist den payload von Remaining Capacity anzeigen lassen diesen kopieren und unter publish einfügen. Im Anschluß device_class löschen (HA kann nichts mit Ah anfangen), wenn du willst könntest du auch device_class: " " durch state_class: "measurement" ersetzen. Aber wie gesagt, selbst wenn du es jetzt in HA siehst kann HA nix damit anfangen. Mach es wie oben beschrieben, das ist die sauberste Lösung.

Ich hatte geschrieben, Current * Volt und dann die kWH rechnen…und nicht gesagt das dass die kWH sind, ich weiß dass A*V = Watt 😅

Danke für deine Einleitung, ich werde das heute Abend versuchen, sage dir Bescheid wie es wird 👍🏻

Jarnsen, 2023-11-18:
Ok, ja teste mal.

softwarecrash, 2023-11-18:
Auf dem ESP werden keinerlei berechnungen dahingehend durchgeführt, alle daten die gelifert werden kommen so vom BMS.
es gibt auch keine pläne dahingehend etwas zu bauen

closed 2023-11-18 by softwarecrash