softwarecrash/Daly2MQTT #151: Update vom 1.0.0 auf aktuelle Version - Keine Werte mehr

heckmic, 2024-01-17:
### A clear and concise description of what the problem is.

Wenn ich auf die aktuelle Version aktualisieren will (u.A. wegen MQTT Problemen - in IOBROKER werden immer neue Objekte angelegt) dann bekomme ich keine Daten mehr vom BMS (Offline).
Downgrade ich wieder, ist alles ok.

heckmic, 2024-01-17:
Habs rausgefunden! - Ohne den Transistor

softwarecrash, 2024-01-17:
Die MQTT "probleme" kommen von instabilen WLAN verbindungen

heckmic, 2024-01-17:
> Die MQTT "probleme" kommen von instabilen WLAN verbindungen

Das wäre aber komisch, da ich hier nur gutes unify habe und gut 100 andere mqtt Geräte. Mit der neuen Version ist es bis jetzt nicht mehr passiert. Abwarten.

closed 2024-03-03 by github-actions[bot]