softwarecrash/Daly2MQTT #152: [Bug]: V2.12.0 now reports "Pack_Remaining_mAh" in Ah

mt-mrx, 2024-01-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Breaking change in 2.12.0 now reports to the same "old" topic which has mAh as unit in the topic name the remaining capacity now in Ah. Before we got 13800 mAh, now we get 13.8 Ah which is very misleading.
Please change MQTT topic to "Pack_Remaining_Charge", "Pack_Remaining_Capacity" or "Pack_Remaining_Ah" or something like that. 

### Screenshots / Fotos

issue is code not hardware related

### Steps To Reproduce

Upgrade to 2.12.0

### Version

2.x.x and above

### Relevant livejson output

```json
not relevant
```

### Which BMS is connected?

not relevant

### What browsers are you seeing the problem on?

no Issue with the Browser or WebUI

softwarecrash, 2024-01-23:
Hello,
thank you, thats a missleading typo. i have corrected it and patched to the V2.12.0 release, please redownload anf flash it.

mt-mrx, 2024-01-23:
thanks, fixed

closed 2024-01-23 by mt-mrx