softwarecrash/Daly2MQTT #153: Cell amount problem

darkazar, 2024-01-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

After updateing from 2.10 to 2.12 over ota i see different ampunt of cells over mqqt and web interface. I have 16s cells and over web inteface is only 8 visible. 

### Screenshots / Fotos

[image]
[image]

### Version

2.x.x and above

### Relevant livejson output

```json
{"Device":{"Name":"Daly2MQTT","IP":"192.168.1.179","ESP_VCC":3.789,"Wifi_RSSI":-61,"Relais_Active":false,"Relais_Manual":false,"sw_version":"2.12.0","Flash_Size":4194304,"Sketch_Size":448512,"Free_Sketch_Space":3723264},"Pack":{"Voltage":64.4,"Current":0,"Power":0,"SOC":97.7,"Remaining_Ah":99.6,"Cycles":16,"BMS_Temp":22,"Cell_Temp":22,"cell_hVt":4.15,"cell_lVt":2.8,"cell_hVt2":4.2,"cell_lVt2":3.4,"pack_hVt":66.4,"pack_lVt":44.8,"pack_hVt2":67.2,"pack_lVt2":55,"High_CellNr":11,"High_CellV":4.033,"Low_CellNr":1,"Low_CellV":4.02,"Cell_Diff":13,"DischargeFET":true,"ChargeFET":true,"Status":"Stationary","Cells":16,"Heartbeat":182,"Balance_Active":false,"Fail_Codes":""},"CellV":{"CellV_1":4.02,"CellV_2":4.03,"CellV_3":4.032,"CellV_4":4.03,"CellV_5":4.03,"CellV_6":4.031,"CellV_7":4.03,"CellV_8":4.031,"CellV_9":4.032,"CellV_10":4.03,"CellV_11":4.033,"CellV_12":4.033,"CellV_13":4.033,"CellV_14":4.033,"CellV_15":4.032,"CellV_16":4.031},"CellTemp":{"Cell_Temp_1":22}}
```

### Which BMS is connected?

Daly 16S 60V 150A
[image]

### What browsers are you seeing the problem on?

Chrome

softwarecrash, 2024-01-23:
@darkazar @mt-mrx 
please try this version
[Daly2MQTT_d1_mini_2.12.1.zip](https://github.com/softwarecrash/Daly2MQTT/files/14027547/Daly2MQTT_d1_mini_2.12.1.zip)

darkazar, 2024-01-23:
Now is working perfectly.
[image]

softwarecrash, 2024-01-23:
please re-download the V2.12.0 release, the smal patches are in this

closed 2024-01-23 by softwarecrash