softwarecrash/Daly2MQTT #157: [Bug]: Daly2MQTT installer does not flash the latest version

gzi01, 2024-01-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

I used the tool at https://softwarecrash.github.io/Daly2MQTT/espflashtool/ to flash a Wemos D1 mini and got release 2.8.3 instead of the latest 2.12.0 .

I would have expected the latter or at least options to select a release.

### Screenshots / Fotos

[image]

### Steps To Reproduce

described above

### Version

2.x.x and above

### Relevant livejson output

```json
n/a
```

### Which BMS is connected?

not relevant

### What browsers are you seeing the problem on?

Microsoft Edge

softwarecrash, 2024-01-29:
the online flash tool is actualy not maintained

softwarecrash, 2024-01-30:
Removed, no capacity to maintain it.

closed 2024-01-30 by softwarecrash

gzi01, 2024-01-31:
Ist es möglich in die Doku aufzunehmen, wie der empfohlene Weg aussieht, die .BIN auf den Wemos D1 mini zu übertragen?

softwarecrash, 2024-01-31:
ist doch beschrieben, einfach mit tasmotizer
https://github.com/softwarecrash/Daly2MQTT?tab=readme-ov-file#how-to-use