softwarecrash/Daly2MQTT #158: Status if cells will be balanced are no longer published

Bavarialex, 2024-01-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Since update to the current release 2.12.0 the topics which cells get active balance are no longer updated.
Balance active state works...

### Screenshots / Fotos

no Fotos

### Version

2.x.x and above

### Relevant livejson output

```json
{"Device":{"Name":"Daly2MQTT","IP":"10.11.12.165","ESP_VCC":3.396,"Wifi_RSSI":-53,"Relais_Active":false,"Relais_Manual":false,"sw_version":"2.12.0","Flash_Size":4194304,"Sketch_Size":448464,"Free_Sketch_Space":3723264},"Pack":{"Voltage":13,"Current":10,"Power":130,"SOC":68.3,"Remaining_Ah":102.4,"Cycles":1,"BMS_Temp":21,"Cell_Temp":21,"cell_hVt":3.65,"cell_lVt":2.3,"cell_hVt2":3.75,"cell_lVt2":2.5,"pack_hVt":14.6,"pack_lVt":9.2,"pack_hVt2":14.7,"pack_lVt2":10,"High_CellNr":2,"High_CellV":3.269,"Low_CellNr":3,"Low_CellV":3.244,"Cell_Diff":25,"DischargeFET":true,"ChargeFET":true,"Status":"Charge","Cells":4,"Heartbeat":41,"Balance_Active":false,"Fail_Codes":""},"CellV":{"CellV_1":3.26,"CellV_2":3.269,"CellV_3":3.244,"CellV_4":3.25},"CellTemp":{"Cell_Temp_1":21}}
```

### Which BMS is connected?

4S 100A daly UART

### What browsers are you seeing the problem on?

Chrome

softwarecrash, 2024-01-30:
Have you read the release notes for the V2.12.0 ?

[release notes](https://github.com/softwarecrash/Daly2MQTT/releases/tag/V2.12.0)
>⚠️ Removed cell balance states for single cells due to compliance to Daly previous and current protocol communication.
>⚠️ Changed remaining Capacity from mAh to Ah due to compliance to Daly previous and current protocol communication.

Bavarialex, 2024-01-30:
Oh shit, it's not a bug but a feature... ;-)

Apologize for this - can be closed...

closed 2024-01-30 by Bavarialex