softwarecrash/Daly2MQTT #159: Daly2MQTT creates strange datapoints / folders in ioBroker object tree

Tesla-Birne, 2024-02-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Daly2MQTT is working fine. After 2 days of using it, I get some strange looking datapoints added by Daly2MQTT:

### Screenshots / Fotos

[image]

### Steps To Reproduce

I deleted all of the new useless datapoints, the next day there were new ones created. They become more and more the longer the software is active.

### Version

2.x.x and above

### Relevant livejson output

```json
{"Device":{"Name":"Daly2MQTT","IP":"192.168.2.152","ESP_VCC":3.378,"Wifi_RSSI":-58,"Relais_Active":false,"Relais_Manual":false,"sw_version":"2.12.0","Flash_Size":4194304,"Sketch_Size":448464,"Free_Sketch_Space":3723264},"Pack":{"Voltage":27,"Current":0.8,"Power":21.6,"SOC":56.8,"Remaining_Ah":130.6,"Cycles":0,"BMS_Temp":17,"Cell_Temp":17,"cell_hVt":3.65,"cell_lVt":2.3,"cell_hVt2":3.65,"cell_lVt2":2.65,"pack_hVt":29.2,"pack_lVt":18.4,"pack_hVt2":29.2,"pack_lVt2":22,"High_CellNr":6,"High_CellV":3.384,"Low_CellNr":3,"Low_CellV":3.38,"Cell_Diff":4,"DischargeFET":true,"ChargeFET":true,"Status":"Stationary","Cells":8,"Heartbeat":131,"Balance_Active":false,"Fail_Codes":""},"CellV":{"CellV_1":3.382,"CellV_2":3.383,"CellV_3":3.38,"CellV_4":3.382,"CellV_5":3.382,"CellV_6":3.384,"CellV_7":3.383,"CellV_8":3.383},"CellTemp":{"Cell_Temp_1":17}}
```

### Which BMS is connected?

Daly 150A, UART

### What browsers are you seeing the problem on?

Safari

softwarecrash, 2024-02-02:
please check your wifi strength, this behavior comes when while transimitting data the connection abort or esp restart

Tesla-Birne, 2024-02-02:
OK, I will try to find a better place for the D1.
Thanks so far

closed 2024-03-19 by github-actions[bot]