softwarecrash/Daly2MQTT #160: [FEATURE] SOC Einstellbar über MQTT oder Webinterface

flofresser, 2024-02-11:
### Is your feature request related to a problem? Please describe

Das DALY BMS bietet die Möglichkeit den SOC über Bluetooth einzustellen.
Wenn man die definierte maximale Ladespannung erreicht hat kann man den SOC auf 100% stellen.
Wäre es möglich dieses Feature auch bei Daly2MQTT hinzuzufügen?

Beim ESP-Home Sensor wurde es folgendermaßen umgesetzt:
Hier ist es möglich über den ESP-32 des SOC auf 0/25/50/75 und 100% zu stellen.

select:
  - platform: template
    name: "Daly Battery Level setup"
    optimistic: True
    options:
      - 100%
      - 75%
      - 50%
      - 25%
      - 0%
    initial_option: 100%
    set_action:
      then:
        - if:
            condition:
              lambda: 'return x == "100%";'
            then:
              - uart.write:
                  data: [0xA5, 0x40, 0x21, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x03, 0xE8, 0xF9]
              - logger.log:
                  format: "Send cmd to Daly: Set SOC to 100%"
            else:
              - if:
                  condition:
                    lambda: 'return x == "75%";'
                  then:
                    - uart.write:
                        data: [0xA5, 0x40, 0x21, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x02, 0xEE, 0xFE]
                    - logger.log:
                        format: "Send cmd to Daly: Set SOC to 75%"
                  else:
                    - if:
                        condition:
                          lambda: 'return x == "50%";'
                        then:
                          - uart.write:
                              data: [0xA5, 0x40, 0x21, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xF4, 0x03]
                          - logger.log:
                              format: "Send cmd to Daly: Set SOC to 50%"
                        else:
                          - if:
                              condition:
                                lambda: 'return x == "25%";'
                              then:
                                - uart.write:
                                    data: [0xA5, 0x40, 0x21, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xFA, 0x08]
                                - logger.log:
                                    format: "Send cmd to Daly: Set SOC to 25%"
                              else:
                                - if:
                                    condition:
                                      lambda: 'return x == "0%";'
                                    then:
                                      - uart.write:
                                          data: [0xA5, 0x40, 0x21, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x0E]
                                      - logger.log:
                                          format: "Send cmd to Daly: Set SOC to 0%"

### Describe the solution you'd like

-

### Describe alternatives you've considered

-

flofresser, 2024-02-11:
Ok Sorry ich habe gerade gesehen dass das e schon Funktioniert:

Ohne Logindaten:
http://ipadress:/set?soc=1

mit Logindaten:
http://user:pass@ipadress:/set?soc=1

wäre trotzdem cool wenn es übers Webinterface einstellbar wäre.

Danke für das tolle Projekt, funktioniert wahnsinnig gut.

softwarecrash, 2024-02-11:
Hallo,
das der SOC nicht über webinterface einstellbar ist, ist gewollt.
zb wenn man mit dem smartphone drauf schaut, ist zu schnell mal auf irgendeinen schieberegler getippt und es ist verstellt, die fet switches sind schon kritisch.

den SOC zu korrigieren machen die meisten über scripte die den passenden wert dann über mqtt oder eben webhook setzen.

softwarecrash, 2024-02-11:
@flofresser wenn man so darüber nachdenkt lässt es sich evt doch mit einbauen mit einer zusätzlichen abfrage.
ich würde es nur gern irgendwo verstecken, genauso wie die zell einzelansicht, evt als klick trigger auf "Package:" oder was würde da sinn ergeben?

flofresser, 2024-02-11:
Ich denke ich würde unter Settings ein neues Feld zum ausklappen machen z.B. " Batterie & SOC".
Und dort würde ich den SOC Wert eintragen lassen und mit einem Button "Send to BMS" abschicken lassen.
Da kannst du dann auch die einzelnen Zell-Ansichten darstellen.
[image]

softwarecrash, 2024-02-11:
@flofresser 
probier die mal
[Daly2MQTT_d1_mini_2.12.1A3.zip](https://github.com/softwarecrash/Daly2MQTT/files/14232666/Daly2MQTT_d1_mini_2.12.1A3.zip)
mit doppelklick auf die SOC bar

flofresser, 2024-02-11:
Ich finde es Hammer!
Es funktioniert auf dem PC und Iphone 8 einwandfrei.
Nur beim Iphone SE 2020 funktioniert das irgendwie nicht.
Ich denke das Problem liegt am IOS 17.
Ist aber für mich soweit wirklich in Ordnung.
Danke vielmals

softwarecrash, 2024-02-11:
alle browser mit allen funktionen abzudecken ist sehr schwer, zumal es ja scheinbar unterschiede innerhalb der IOS browser gibt.
da kann ich leider nicht auf alle rücksicht nehmen

flofresser, 2024-02-11:
Nach erneutem neu laden des Webinterfaces hats nun doch mit allen Geräten funktioniert.

Super Danke

softwarecrash, 2024-02-11:
super.
Release dazu ist gerade raus.

closed 2024-02-11 by softwarecrash