softwarecrash/Daly2MQTT #161: [Bug]: DALY BMS to MQTT 2.2.11 works fine but all new firmware no communication

AudipowerS4, 2024-02-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

DALY BMS to MQTT 2.2.11 Firmware works fine ... but all firmware later i have problem with Communication.

### Screenshots / Fotos

[image]
[image]

### Version

2.x.x and above

### Relevant livejson output

```json
{"Device":{"Name":"Daly2MQTT","IP":"192.168.188.63","ESP_VCC":3.284,"Wifi_RSSI":-71,"Relais_Active":false,"Relais_Manual":false,"sw_version":"2.13.0","Flash_Size":4194304,"Sketch_Size":448544,"Free_Sketch_Space":3723264},"Pack":{"Voltage":0,"Current":0,"Power":0,"SOC":0,"Remaining_Ah":0,"Cycles":0,"BMS_Temp":0,"Cell_Temp":0,"cell_hVt":0,"cell_lVt":0,"cell_hVt2":0,"cell_lVt2":0,"pack_hVt":0,"pack_lVt":0,"pack_hVt2":0,"pack_lVt2":0,"High_CellNr":0,"High_CellV":0,"Low_CellNr":0,"Low_CellV":0,"Cell_Diff":0,"DischargeFET":false,"ChargeFET":false,"Status":"Offline","Cells":0,"Heartbeat":0,"Balance_Active":false,"Fail_Codes":""},"CellV":{},"CellTemp":{}}
```

### Which BMS is connected?

DALY BMS 200A UART

### What browsers are you seeing the problem on?

Firefox

AudipowerS4, 2024-02-12:
UPDATE: Daly2MQTT 2.8.4 Works

softwarecrash, 2024-02-12:
can you flash the debug version and grab the usb-serial data or webserial data?

AudipowerS4, 2024-02-14:
{"Device":{"Name":"Batterie1","IP":"192.168.188.63","ESP_VCC":3.284,"Wifi_RSSI":-75,"Relais_Active":false,"Relais_Manual":false,"sw_version":"2.13.0_debug d1_mini Feb 11 2024 15:36:40","Flash_Size":4194304,"Sketch_Size":519904,"Free_Sketch_Space":3653632,"CPU_Frequency":80,"Real_Flash_Size":4194304,"Free_Heap":30344,"HEAP_Fragmentation":5,"Free_BlockSize":29040,"json_memory_usage":1358,"json_capacity":2048,"runtime":110,"ws_clients":1,"MQTT_Json":false},"Pack":{"Voltage":0,"Current":0,"Power":0,"SOC":0,"Remaining_Ah":0,"Cycles":0,"BMS_Temp":0,"Cell_Temp":0,"cell_hVt":0,"cell_lVt":0,"cell_hVt2":0,"cell_lVt2":0,"pack_hVt":0,"pack_lVt":0,"pack_hVt2":0,"pack_lVt2":0,"High_CellNr":0,"High_CellV":0,"Low_CellNr":0,"Low_CellV":0,"Cell_Diff":0,"DischargeFET":false,"ChargeFET":false,"Status":"Offline","Cells":0,"Heartbeat":0,"Balance_Active":false,"Fail_Codes":""},"CellV":{},"CellTemp":{}}

AudipowerS4, 2024-02-14:
Maybe was the Port change ? What about difference 2.11 and 2.13 about Ports ?

AudipowerS4, 2024-02-14:
[image]

softwarecrash, 2024-02-15:
> Maybe was the Port change ? What about difference 2.11 and 2.13 about Ports ?

https://github.com/softwarecrash/Daly2MQTT/wiki/Wiring
No Pin changes, whats your wireing?

> ![grafik](https://private-user-images.githubusercontent.com/84537667/304849166-4a92b743-0995-469f-a046-9fb7931521a8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc5ODQ2NjAsIm5iZiI6MTcwNzk4NDM2MCwicGF0aCI6Ii84NDUzNzY2Ny8zMDQ4NDkxNjYtNGE5MmI3NDMtMDk5NS00NjlmLWEwNDYtOWZiNzkzMTUyMWE4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjE1VDA4MDYwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVjYmYwNjY2NDcxOGExZmY2YWEzYjZiMmVjNDMzY2E1ODI0NzFhMzg1NDgxZjJkMDY4OWZhY2U0YzRlNTRhYTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.VEzeOthNG0zEyfQ8iGZqqghoR_klrZ2mOSYVS3mrF9I)

no answer from the BMS, have you a transistor or something in the connection ESP-BMS?

closed 2024-04-01 by github-actions[bot]