softwarecrash/Daly2MQTT #162: [Bug]: MQTT Payload of Pack_Current and Pack_Power contain a space before the number when under the value 10.0 like " 6.3"

daebikelinz, 2024-02-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

The payload of the current and power contain a space before the number if they are one digit before the decimal point.
The payload for 1.0 A is " 1.0" instead of "1.0"
The postprocessing OpenHAB item can't resolve the " 1.0" to a number:
"2024-02-15 21:37:00.350 [WARN ] [ab.binding.mqtt.generic.ChannelState] - Incoming payload ' 0.0' not supported by type 'NumberValue'"

### Screenshots / Fotos

[image]

Here the power is frozen to 2 digits before the comma 10.6 because the payload "10.6" can be converted.
Values under 10.0 W can't be processed so OpenHAB item still shows 10.6 W besides being dark outside and should be 0.0 W.
[image]

### Version

2.x.x and above

### Relevant livejson output

```json
{"Device":{"Name":"Daly2MQTT","IP":"192.168.178.117","ESP_VCC":3.293,"Wifi_RSSI":-78,"Relais_Active":false,"Relais_Manual":false,"sw_version":"2.13.0","Flash_Size":4194304,"Sketch_Size":448544,"Free_Sketch_Space":3723264},"Pack":{"Voltage":52.9,"Current":0,"Power":0,"SOC":50,"Remaining_Ah":52.5,"Cycles":0,"BMS_Temp":8,"Cell_Temp":8,"cell_hVt":3.65,"cell_lVt":2.3,"cell_hVt2":3.7,"cell_lVt2":2.7,"pack_hVt":58.4,"pack_lVt":36.8,"pack_hVt2":58.4,"pack_lVt2":40,"High_CellNr":6,"High_CellV":3.312,"Low_CellNr":1,"Low_CellV":3.31,"Cell_Diff":2,"DischargeFET":true,"ChargeFET":true,"Status":"Stationary","Cells":16,"Heartbeat":163,"Balance_Active":false,"Fail_Codes":""},"CellV":{"CellV_1":3.31,"CellV_2":3.31,"CellV_3":3.311,"CellV_4":3.311,"CellV_5":3.31,"CellV_6":3.312,"CellV_7":3.312,"CellV_8":3.312,"CellV_9":3.312,"CellV_10":3.312,"CellV_11":3.312,"CellV_12":3.312,"CellV_13":3.312,"CellV_14":3.312,"CellV_15":3.312,"CellV_16":3.312},"CellTemp":{"Cell_Temp_1":8}}
```

### Which BMS is connected?

Daly BMS 16s 100A 

### What browsers are you seeing the problem on?

no Issue with the Browser or WebUI

felsenputzer, 2024-02-15:
I stumbled on the same issue. I susspect it is due to the ussage of
`dtostrf(bms.get.packCurrent, 4, 1, msgBuffer));`
which will generate 4 charachter long strings minimum, however small numbers (0.0 ... 9,9) only have three characters, therfore a leading space will be added.

softwarecrash, 2024-02-15:
@felsenputzer you are right, thats my fail.
@felsenputzer @daebikelinz try this version
[Daly2MQTT_d1_mini_2.13.0A1_debug.zip](https://github.com/softwarecrash/Daly2MQTT/files/14302501/Daly2MQTT_d1_mini_2.13.0A1_debug.zip)

felsenputzer, 2024-02-15:
@softwarecrash Thanks for the quick fix, just uploaded it and it works fine for me now.

daebikelinz, 2024-02-15:
> @felsenputzer you are right, thats my fail. @felsenputzer @daebikelinz try this version [Daly2MQTT_d1_mini_2.13.0A1_debug.zip](https://github.com/softwarecrash/Daly2MQTT/files/14302501/Daly2MQTT_d1_mini_2.13.0A1_debug.zip)

Problem solved. That was quick. Thank you!

closed 2024-02-15 by daebikelinz

softwarecrash, 2024-02-15:
Great,
The fix comes Up with the next Bugfix Release.
Tank you both for the Report and Help.