softwarecrash/Daly2MQTT #164: [Bug]: Since FW 2.12.0

mt-mrx, 2024-02-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Since the update to 2.12.0 (debug) I'm seeing unexpected "Pack_DischargeFET" = false messages, "Pack_Current" values of exactly "-3000", "Pack_SOC" = 0, "Pack_Voltage" = 0 and "Pack_Remaining_Ah" = 0 . (Not sure if there are more parameters reporting (wrong) readings)

During the day, when I have surplus PV energy, I keep the BMS awake, to charge it with the surplus energy, by sending every 30s a "Device_Control/Wake_BMS" message.

As already discussed in #132 my BMS sometimes isn't responsive enough and I see the "Pack_Status" changing from "Stationary" to "Offline" and during the next Wake_BMS back to "Stationary".
According to my data logging (influxdb) I see the following timestamps:

I'm collecting all MQTT messages coming in in a 2s window and write it with the same timestamp into the DB.
I'm converting "Offline" to -1 and "Stationary" to 0.
Whenever the "Pack_Status" is "Stationary", "Charge" or "Discharge" I'm writing the data to the DB, otherwise I'm writing ONLY the "Pack_Status" and "Alive".

```
Pack_Status     0        2024-02-27T12:40:46.006Z
Pack_ChargeFET  1        2024-02-27T12:40:46.006Z
Pack_Current    0        2024-02-27T12:40:46.006Z
Pack_SOC        62.3     2024-02-27T12:40:46.006Z

Pack_Status     -1       2024-02-27T12:41:27.125Z

Pack_Status     0        2024-02-27T12:41:47.520Z
Pack_ChargeFET  0        2024-02-27T12:41:47.520Z
Pack_Current    -3000    2024-02-27T12:41:47.520Z
Pack_SOC        0        2024-02-27T12:41:47.520Z

Pack_Status     0        2024-02-27T12:42:15.045Z
Pack_ChargeFET  1        2024-02-27T12:42:15.045Z
Pack_Current    0        2024-02-27T12:42:15.045Z
Pack_SOC        62.3     2024-02-27T12:42:15.045Z
```

So to me it looks like the "wrong" data is sent when the pack comes back online or better that's when that data is definitely sent, it could be that it already appears when the "Pack_Status" changes to "Offline" but I'm not writing "Offline" data to the DB, see explanation above.

I have not seen this behavior with firmware version 2.11.0 (and earlier).

If I remember correctly when there are BMS read/communication errors the previous values were reused and not reset.
Did anything change in that regard?

### Screenshots / Fotos

I'm still using the exact same setup as in #132

### Steps To Reproduce

Cannot reproduce it, looks like it happens when the communication between the BMS and the Wemos is unreliable and the BMS is not responding to the Wemos for a while.

### Version

2.x.x and above

### Relevant livejson output

```json
Have no method of collecting the livejson output when the issue appears.
```

### Which BMS is connected?

Lifepo4 15S 48V 60A UART / Bluetooth

### What browsers are you seeing the problem on?

no Issue with the Browser or WebUI

mt-mrx, 2024-02-27:
Ah, I just figured out how to recreate the issue I'm seeing.
It has nothing to do with bad communication between the BMS and the Wemos.

I have set my BMS to got to sleep after 5 minutes. I'm only waking it up during the day when there is even a chance to charge it, so if I have a certain amount of power being exported to the grid.
If there is a cloudy day and the power changes quickly it's being woken up goes to sleep and is being woken up again.

So, I can recreate the issue, even at night, I have to wait for the BMS to go to sleep and then wake it.
When I wake it, the first data that's being sent reports the wrong values. 

While the BMS is sleeping I see the "correct"/"old" data in the livejson and MQTT output, so the state and values from when it went to sleep.

Below you see a log and can and see the timing of the wrong messages, I used MQTT to send the Wake_BMS signal:

```
$ mosquitto_sub -i "clientidsub" -u user -P pw -h brokerip -p 1883 -t 'dalybms/#' -v | ts "%H:%M:%.S" | grep -E "Pack_Status|Pack_Current|Pack_Voltage|Wake_BMS"
22:31:21.113858 dalybms/Pack_Voltage 46.2
22:31:21.114120 dalybms/Pack_Current  0.0
22:31:21.117623 dalybms/Pack_Status Offline
22:32:01.702101 dalybms/Pack_Voltage 46.2
22:32:01.702578 dalybms/Pack_Current  0.0
22:32:01.705811 dalybms/Pack_Status Offline
22:32:07.351963 dalybms/Device_Control/Wake_BMS true
22:32:08.673267 dalybms/Device_Control/Wake_BMS false
22:32:11.963706 dalybms/Pack_Voltage  0.0
22:32:11.964110 dalybms/Pack_Current -3000.0
22:32:11.970878 dalybms/Pack_Status Stationary
22:32:43.908161 dalybms/Pack_Voltage 46.2
22:32:43.908523 dalybms/Pack_Current  0.0
22:32:43.911404 dalybms/Pack_Status Stationary
22:33:15.907124 dalybms/Pack_Voltage 46.2
22:33:15.907873 dalybms/Pack_Current  0.0
22:33:15.916410 dalybms/Pack_Status Stationary
...
22:36:27.662153 dalybms/Pack_Voltage 46.2
22:36:27.662498 dalybms/Pack_Current  0.0
22:36:27.668592 dalybms/Pack_Status Stationary
22:36:59.645788 dalybms/Pack_Voltage 46.2
22:36:59.646747 dalybms/Pack_Current  0.0
22:36:59.658206 dalybms/Pack_Status Stationary
22:37:41.615937 dalybms/Pack_Voltage 46.2
22:37:41.616742 dalybms/Pack_Current  0.0
22:37:41.631586 dalybms/Pack_Status Offline
22:38:22.204224 dalybms/Pack_Voltage 46.2
22:38:22.204480 dalybms/Pack_Current  0.0
22:38:22.208980 dalybms/Pack_Status Offline
22:39:02.797606 dalybms/Pack_Voltage 46.2
22:39:02.798488 dalybms/Pack_Current  0.0
22:39:02.803226 dalybms/Pack_Status Offline
22:39:10.636293 dalybms/Device_Control/Wake_BMS true
22:39:11.038449 dalybms/Device_Control/Wake_BMS false
22:39:14.219997 dalybms/Pack_Voltage  0.0
22:39:14.220526 dalybms/Pack_Current -3000.0
22:39:14.223820 dalybms/Pack_Status Stationary
22:39:46.193061 dalybms/Pack_Voltage 46.1
22:39:46.193258 dalybms/Pack_Current  0.0
22:39:46.195899 dalybms/Pack_Status Stationary
```

softwarecrash, 2024-02-28:
Hello,
this "-3000 bug" it not realy a bug, its definitly your BMS was send this data, in old versions i have set a condition to filter the whole data when this value comes up.
in the new comunication changes (due to support the old BMS) this filter not exsist.
give me a bit time, i will look that i reimplement this filter function again to "fix" the stupidy of this BMS

mt-mrx, 2024-02-28:
Thanks, sounds really stupid, I'm almost at the point where I consider buying a JK BMS instead :-( .
What is especially annoying is that I also see "Pack_DischargeFET = false" and I trigger on that to turn my attached HM-300 inverter off to protect its input capacitors even though that FET is definitely not off since I see no voltage drop on the HM-300 DC side.

Do you know if there is a way to get "new/better" firmware for the Daly BMS?
I can't find any information on the https://www.dalybms.com/ website if there even is a method to update the firmware.

softwarecrash, 2024-02-28:
you dont need to wake your BMS, it will wake when charge or discharge.

i dont know much about the firmwares for the BMS, just ask daly support.

mt-mrx, 2024-02-28:
The problem with my setup is, that it's very small, small capacity (below 1kWh), small currents involved.
The BMS doesn't detect that it's charging or discharging when you do so with less than ~35-40 Watt.
I even tried to change that minimum current setting via the PC app but there is a lower limit below which the BMS doesn't detect charging/discharging current and so it goes to sleep even though it's being charged.
Since I want to monitor the cell voltages I need to keep it awake manually.

softwarecrash, 2024-02-28:
try this, when values comes up with the -300 it will throw all data and recollect
[Daly2MQTT_d1_mini_2.13.0A1.zip](https://github.com/softwarecrash/Daly2MQTT/files/14435286/Daly2MQTT_d1_mini_2.13.0A1.zip)

closed 2024-02-29 by mt-mrx